probe-rs/probe-rs #1: Open issues for planned features?

chocol4te, 2019-03-12:
I’d like to contribute but don’t really know what needs doing, I don’t suppose you could open some issues for missing features/bug fixes?

Yatekii, 2019-03-13:
Cool that you want to contribute!

At the moment we are basically reading and implementing the ADI v5.2 spec.
We are experimenting with what works best and what path we want to follow.
There is a lot of things (API related) that are not optimal yet and need improvements.
Basically what works is basic reading of memory locations (ROM and RAM).
Writing RAM works; not so much for ROM tho. ROM is a bit tricky, as it is different for each and every chip.
Also, we have not ventured into debug capabilities like setting breakpoints, halting, stepping, etc.
If you have something specific you would like to implement or you have general suggestions on the code, you are very welcome to just drop a PR. If you wanna make sure it's OK beforehand, just ask real quick on that specific thing. But we basically accept any reasonable contribution as we are still at the very beginning.

@therealprof  has a an also very basic implementation of a REPL, where he tries out the ergonomics of the `probe-rs` API and we reiterate if it sucks. You can have a look here: https://github.com/therealprof/probemeister

You can also find us on mozilla#rust-embedded if you want to chat. Hope to find you there soon; otherwise GH issues is ok too ;)

Yatekii, 2019-05-21:
Ok, so we made huge progress and can now fully read ROM tables with a DAPlink.
We also have preliminary support for DebugPort features.
Furthermore I am close to properly flashing a ROM image into the flash =)

So if you still got too much time and interest, feel free to pick one of the v 0.1.0 milestone issues :)

As usual, questions can be asked on GH or IRC.

closed 2019-10-07 by Yatekii