probe-rs/probe-rs #21: Error detecting ROM-Table for FRDM-KL26Z

Tiwalun, 2019-10-07:
Probe-rs fails to find the ROM-Table for a FRDM-KL26Z chip. Based on the logs, it seems that
a read of the DRW register does not work, and then the program aborts reading the ROM table.

See line 333 in the following log for the incomplete read:

https://gist.github.com/therealprof/3d6f13518fa2d9975af5b7361b36f991#file-probe-rs-cli-info-debug-output-for-frdm-kl26z-L333

Compared to other entries, a second line showing the read value is missing.

Yatekii, 2019-10-15:
Is this resolved?

therealprof, 2019-10-15:
I can check later today.

Yatekii, 2020-03-13:
@therealprof does this offer still stand? =)

Yatekii, 2020-05-31:
Does this issue still persist?
I think it could be resolved with the new ROM-table parsing additions.

therealprof, 2020-05-31:
Let me check...

therealprof, 2020-05-31:
Fixed now. Thanks.

closed 2020-05-31 by therealprof

Yatekii, 2020-05-31:
Thanks for testing!