probe-rs/probe-rs #42: `cargo flash` does not work inside members of a workspace

jonas-schievink, 2019-10-18:
It looks for `target` in the current directory, but in a workspace `target` is in the workspace root, resulting in a panic like this:

```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', src/libcore/result.rs:1084:5
```

Yatekii, 2019-10-18:
This is an error in cargo-project: https://github.com/japaric/cargo-project/issues/5 .
Maybe someone finds time to fix it. I think I focus on other things for now.

jonas-schievink, 2019-10-19:
Hmm, I ran this from the subcrate when it failed (Rubble's demos/nrf52810-demo)

Yatekii, 2019-10-19:
That is odd. Guess we'll need to investigate some more.

Yatekii, 2019-12-06:
Wasn't this fixed?

brainstorm, 2020-01-01:
Saw this when working with project https://github.com/chris-zen/mdp-link-rs, the symptom when running `cargo flash` is the message seen on https://github.com/probe-rs/probe-rs/issues/102.

Yatekii, 2020-01-01:
Well, the problem is that cargo-project (the crate we use to get the metainfo) does always look for a valid cargo manifest before marching up the directory tree and looking for a project workspace.

This: https://github.com/japaric/cargo-project/blob/master/src/lib.rs#L74 is where it fails to parse the workspace manifest, because it expects a project manifest. Maybe you could try and go from there :)

brainstorm, 2020-01-09:
Had a quick look at it and I'm not sure what would be the best course of action since `cargo flash` from the top level workspace directory is not specific enough: which sub-cargo-project should be flashed if there are multiple that generate the corresponding targets?