probe-rs/probe-rs #52: Wrong PART on STM32F042 and nRF52832

Yatekii, 2019-10-23:
the STM32F042 and the nRF52832 have a wrong PART. Should be fixed by an owner of those chips/devkits asap.

Yatekii, 2019-11-13:
The `nRF52832` PART is fixed.

Yatekii, 2019-11-13:
Anyone with that chip would be very welcome to evaluate the PART and add it.

therealprof, 2019-11-14:
@Yatekii Isn't that burried in one of the gists I've done a while ago?

Yatekii, 2019-11-14:
Maybe. I am not sure, there was a lot of them ;)

marikka, 2019-11-17:
@Yatekii How can I evaluate the PART?

I have a STM32F042K6T6 connected via the STM32F3DISCOVERY board. Here's what I get by running `info`. Hope this helps. :)

```
~/.cargo/bin/probe-rs-cli info
DP info: 0xbb11477

Available Access Ports:
IDR {
    REVISION: 0x0,
    DESIGNER: 0x23b,
    CLASS: MEMAP,
    _RES0: 0x0,
    VARIANT: 0x2,
    TYPE: AMBA_AHB3,
}
Class1RomTable(
    CSComponentId {
        base_address: 0xe00ff000,
        class: RomTable,
        peripheral_id: PeripheralID {
            REVAND: 0x000000,
            CMOD: No,
            REVISION: 0x000000,
            JEP106: Some(
                JEP106Code({ cc: 0x00, id: 0x20 } => Some("STMicroelectronics")),
            ),
            PART: 0x000440,
            SIZE: 0x000001,
        },
    },
    RomTable {
        entries: [
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component_data: GenericIPComponent(
                    CSComponentId {
                        base_address: 0xe000e000,
                        class: GenericIPComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000000,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x000008,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component_data: GenericIPComponent(
                    CSComponentId {
                        base_address: 0xe0001000,
                        class: GenericIPComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000000,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x00000a,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component_data: GenericIPComponent(
                    CSComponentId {
                        base_address: 0xe0002000,
                        class: GenericIPComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000000,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x00000b,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
        ],
    },
)
```

Yatekii, 2019-11-17:
Hey thanks for the paste! The PART should be the one from the first romtable entry. So, in this example 0x440.

Yatekii, 2020-03-13:
This is not up to date anymore. See #108 #113 and #115 for more info on this.

closed 2020-03-13 by Yatekii