probe-rs/probe-rs #76: ST-Link v2-1 is not supported

rnestler, 2019-11-12:
**Describe the bug**

When connecting an ST-Link v2.1 (`ID 0483:374b STMicroelectronics ST-LINK/V2.1`) and trying to read the info it will lock up.

**To Reproduce**
```
$ target/debug/probe-rs-cli list
The following devices were found:
[0]: STLink V2-1 (VID: 1155, PID: 14155, STLink)

$ target/debug/probe-rs-cli info
DP info: 0x2ba01477

Available Access Ports:
IDR {
    REVISION: 0x2,
    DESIGNER: 0x23b,
    CLASS: MEMAP,
    _RES0: 0x0,
    VARIANT: 0x1,
    TYPE: AMBA_AHB3,
}
$ target/debug/probe-rs-cli info
Error processing command: UnknownError
```

**Expected behavior**
It should work :wink: 

**Desktop (please complete the following information):**
 - Linux (Arch)

**Additional context**
Add any other context about the problem here.

therealprof, 2019-11-12:
Uhm, no problems with that; just used that last Saturday while implementing/testing STLink-V3 support.

therealprof, 2019-11-12:
Maybe a regression since then?

rnestler, 2019-11-12:
> Uhm, no problems with that; just used that last Saturday while implementing/testing STLink-V3 support.

You know which version exactly? I tried a02917e08654b600031af4027a16ef8c5a46b60e and I get the same error...

rnestler, 2019-11-12:
What's even more weird is, that when I use the `st-info` utility from [stlink](https://github.com/texane/stlink) beforehand it works:
<details><summary>CLICK ME</summary>
<p>

```
$ RUST_LOG=info target/debug/probe-rs-cli info
Error processing command: UnknownError

$ st-info --probe
Found 1 stlink programmers
 serial: 303636424646333233353335343734
openocd: "\x30\x36\x36\x42\x46\x46\x33\x32\x33\x35\x33\x35\x34\x37\x34"
  flash: 262144 (pagesize: 2048)
   sram: 40960
 chipid: 0x0422
  descr: F3 device

$ RUST_LOG=info target/debug/probe-rs-cli info
DP info: 0x2ba01477

Available Access Ports:
IDR {
    REVISION: 0x2,
    DESIGNER: 0x23b,
    CLASS: MEMAP,
    _RES0: 0x0,
    VARIANT: 0x1,
    TYPE: AMBA_AHB3,
}
 INFO  probe_rs::memory::romtable > 	Reading component data at: e00ff000
 INFO  probe_rs::memory::romtable > 	Component class: RomTable
 INFO  probe_rs::memory::romtable > 	Component peripheral id: PeripheralID { REVAND: 0, CMOD: No, REVISION: 0, JEP106: Some(JEP106Code({ cc: 0x00, id: 0x20 } => Some("STMicroelectronics"))), PART: 422, SIZE: 1 }
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff000
 INFO  probe_rs::memory::romtable > 	Reading component data at: e000e000
 INFO  probe_rs::memory::romtable > 	Component class: GenericIPComponent
 INFO  probe_rs::memory::romtable > 	Component peripheral id: PeripheralID { REVAND: 0, CMOD: No, REVISION: 0, JEP106: Some(JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd"))), PART: c, SIZE: 1 }
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff004
 INFO  probe_rs::memory::romtable > 	Reading component data at: e0001000
 WARN  probe_rs::memory::romtable > Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0xb0)
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff008
 INFO  probe_rs::memory::romtable > 	Reading component data at: e0002000
 INFO  probe_rs::memory::romtable > 	Component class: GenericIPComponent
 INFO  probe_rs::memory::romtable > 	Component peripheral id: PeripheralID { REVAND: 0, CMOD: No, REVISION: 2, JEP106: Some(JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd"))), PART: 3, SIZE: 1 }
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff00c
 INFO  probe_rs::memory::romtable > 	Reading component data at: e0000000
 WARN  probe_rs::memory::romtable > Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0x1)
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff010
 INFO  probe_rs::memory::romtable > 	Reading component data at: e0040000
 WARN  probe_rs::memory::romtable > Component at 0xe0040000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff014
 INFO  probe_rs::memory::romtable > 	Reading component data at: e0041000
 INFO  probe_rs::memory::romtable > 	Component class: CoreSightComponent
 INFO  probe_rs::memory::romtable > 	Component peripheral id: PeripheralID { REVAND: 0, CMOD: No, REVISION: 0, JEP106: Some(JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd"))), PART: 925, SIZE: 1 }
 INFO  probe_rs::memory::romtable > Reading rom table entry at e00ff018
 INFO  probe_rs::memory::romtable > Entry consists of all zeroes, stopping.
Class1RomTable(
    CSComponentId {
        base_address: 0xe00ff000,
        class: RomTable,
        peripheral_id: PeripheralID {
            REVAND: 0x000000,
            CMOD: No,
            REVISION: 0x000000,
            JEP106: Some(
                JEP106Code({ cc: 0x00, id: 0x20 } => Some("STMicroelectronics")),
            ),
            PART: 0x000422,
            SIZE: 0x000001,
        },
    },
[... 47 lines cut ...]
                format: true    ,
                component_data: Class9RomTable(
                    CSComponentId {
                        base_address: 0xe0041000,
                        class: CoreSightComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000000,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x000925,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
        ],
    },
)
```

</p>
</details>

therealprof, 2019-11-12:
That is indeed weird. I'm pretty sure 43971475f1310e6da4dcf139971d7b9e27cb26a6 worked for me. I can't promise that I hadn't used pyOCD before so maybe it's an initialisation problem. I'm on macOS BTW, that might also contribute to it.

rnestler, 2019-11-12:
> That is indeed weird. I'm pretty sure 4397147 worked for me.

Doesn't work for me :confused: so maybe it really is something platform specific.

Yatekii, 2020-03-13:
@rnestler can you check if this is still a problem? Also, if it is, can you try to upgrade the firmware and try again?

rnestler, 2020-04-02:
> @rnestler can you check if this is still a problem? Also, if it is, can you try to upgrade the firmware and try again?

I'm currently in home office without a ST-Link in reach, but I'll see if I can get it at some point :slightly_smiling_face:

therealprof, 2020-04-02:
We have quite a few people (including me) who confirmed that ST-Link works just fine. Maybe we can just close this?

Yatekii, 2020-04-02:
Yeah true, since adamgreig confirmed this too, I guess it's safe to close.

For anyone in the future wondering: The issue was an USB3.0 Port and outdated drivers.
Maybe specificially the USB3.0 port was a problem and it only worked because of USB2.0 or it was a faulty port. Since many folks are using USB3.0, I assume the latter.

closed 2020-04-02 by Yatekii

rnestler, 2020-04-21:
I finally got my ST-Link and tested with latest master and it seems to work, so all fine :slightly_smiling_face:

iliis, 2022-05-30:
I've run into this as well. `cargo flash` just fails with

>Error Failed to open the debug probe.
>  Hint This could be a permission issue. Check our guide on how to make all probes work properly on your system: https://probe.rs/docs/getting-started/probe-setup

The firmware update can be downloaded from https://www.st.com/en/development-tools/stsw-link007.html (registration required, but it works on Win/Linux/Mac)