probe-rs/probe-rs #97: cargo flash docs not clear about "package"

brainstorm, 2019-12-29:
I got excited seeing that the blackmagic probe is now getting some probe-rs support:

https://twitter.com/yatekii/status/1211121477250113542

And tried it out on my [NRF52840 dongle](https://www.aliexpress.com/item/32972734429.html), but I'm struggling a bit with blinky [from the README.md](https://github.com/probe-rs/probe-rs/commit/21540e28b5dc67b139a10196642fb8eabe449619):

```
$ cargo flash --release --chip nRF52840 --example gpio_hal_blinky
missing field `package`
$ cargo flash --package=probe --release --chip nRF52840 --example gpio_hal_blinky
missing field `package`
$ cargo flash --release --chip nRF52840 --target thumbv6m-none-eabi --example gpio_hal_blinky --package=stlink
missing field `package`
$ cargo flash --package stlink --release --chip nRF52840 --example gpio_hal_blinky
missing field `package`
```

For more context, I'm using an STLink dongle reprogrammed with `blackmagic.bin` and it works well identifying the target on bare gdb:

```
(gdb) monitor swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 M3/M4
 2      Nordic nRF52 Access Port

(gdb) mon read deviceinfo
Part:		NRF52840
Variant:	AAC0
Package:	QIxx
Ram:		256K
Flash:		1024K
```

But it would be epic to cargo flash my firmware right away :)

Yatekii, 2019-12-29:
Hi!

Unfortunately the BMP is not supported and will most likely never be supported.
The BMP has everything probe-rs das on the probe itself. That was the idea and is a cool concept but has a few flaws. It has all the flashing algorithms contained in the firmware and the GDB as well.

We write software that has alle the flashing algorithms on the host and flashes them through the probe. Also, GDB, for us, is just a means to an end. We would want to support all sorts of debuggers directly instead of using GDB in the midterm future.

probe-rs currently supports CMSIS-DAP and STLinkv2 & above probes.

Hope that answers your question.
Best,
Yatekii

closed 2019-12-29 by Yatekii

brainstorm, 2019-12-31:
Makes a ton of sense, thanks for taking the time to explain in detail, I'll be testing it out with a STLink V2 probe soon then, thanks! ;)

brainstorm, 2019-12-31:
Actually @Yatekii, the original (documentation) issue remains with a STLink probe, the docs are not clear about "package":

```
$ cargo flash --package stlink --release --chip nRF52840 --example gpio_hal_blinky
missing field `package`
```

Yatekii, 2019-12-31:
Is this cargo project in a cargo workspace?

brainstorm, 2019-12-31:
I'm just trying to run the README.md example after I `cargo build` the whole `probe-rs` repo and install cargo flash:

```
$ RUST_LOG=info cargo flash --release --example  gpio_hal_ledbutton  --chip  nRF51822_xxAA
missing field `package`
``

Yatekii, 2019-12-31:
We found out on Matrix, that the issue was a missunderstanding with the README.md which I did not understand initially. The README.md has been adjusted in #101 for this reason.