probe-rs/probe-rs #108: Make autodetection more reliable & precise

Yatekii, 2019-12-31:
Does the nRF51822_xxAB have the same part number as the xxAC variant?

_Originally posted by @Tiwalun in https://github.com/probe-rs/probe-rs/pull/106_

__Known issues__

- Multiple chips could match, make sure we get the right one
- Do different variants have the same part number and only differ in memory, or do they also have different part numbers?
- Either way we should find out & potentially implement memory config readout from the ROMtables.

Yatekii, 2020-05-31:
For auto detection, one way to autodetect would be to hash parts of the ROM-tables, such as the chip metadata as well as memory sizes.

Maybe someone wants to work on this at oxidize impl days :)

bugadani, 2024-04-03:
I'll be honest, as a relative newcomer, this issue is kinda vague :) Does probe-rs have chip autodetection in any form right now? What triggers it?

Yatekii, 2024-04-03:
The bits that do auto detect are currently unused. But yes, we read the chip id to detect the exact chip. the target yamls have an optional field. we removed it tho because it was unreliable with - who would have thought - STMs.
If the field is not present in the target yaml we fo not try to detect it.

The problem is that ST violates the spec here and returns the same chipid for a whole family instead of per variant.

So the fix is: read the manufacturer, from that get a manufacturer specific procedure to read any indicators to the chip (ST for example has its own special chipid register) and compare with the chips we know.
For that it would be nice to have the targets sorted by family with per family config or similar.
We also wanted to look at sequences in a scripting lang or WASM but it never came to be.

bugadani, 2024-05-18:
No need to carry around three of these issues. Closing in favour of #2449

closed 2024-05-18 by bugadani