probe-rs/probe-rs #114: Implement more GDB stubs and improve overall features/stability

Yatekii, 2020-01-06:
We need to implement more GDB stubs.
- [ ] step (buggy)
- [ ] step in (not implemented)
- [ ] start (not implemented)
- [ ] mon reset (not implemented)
- [ ] next (not implemented)

Bugs
- [ ] High CPU usage after connection droped due to endless reads
- [ ] `--target` parsing at wrong place
- [ ] Reply contains invalid hex digit 84 (possibly due to `T` in reply packet?)

Features
- [ ] Add `--gdb` to `cargo flash`

david-sawatzke, 2020-01-06:
- [ ] "si" (step instruction) also doesn't work

Yatekii, 2020-01-07:
I implemented `mon reset` such that it resets and halts the core. Is that ok or should it run after? GDB displays the prompt afterwards so I felt it was apropriate that it also halts.

I am not sure if I got address reporting down properly as sometimes GDB complains that the PC is invalid. If someone else would like to have a look that would be great!

david-sawatzke, 2020-01-08:
openocd has options for reset, but defaults to continuing:
```
reset [run|halt|init]
      Reset all targets into the specified mode.Default reset mode is run,
      if not given.
```
I think just halting is fine for now

david-sawatzke, 2020-01-08:
Further suggestions:
- [ ] load
- [ ] semihosting

Yatekii, 2020-01-08:
> openocd has options for reset, but defaults to continuing:

I wonder why it shows the prompt then but yah guess that's just GDB level of stupidity.

> reset [run|halt|init]
>       Reset all targets into the specified mode.Default reset mode is run,
>       if not given.

Good you mention this, because GDB sends a hex encoded weirdo-command for whatever stupid reason, so I have to match the exact hex string (not in the mood to search the docs another 3 hours for yet another GDB oddity), which will then not work if someone writes mon reset init for example.

> [ ] semihosting

Is this GDB related? Before semihosting I'll do ITM, I also do not really have any interest in GDB & semihosting :) I do the minimum so anyone can take up the work if they like (so nobody :D).

> * [ ]  load

Yeah, thought about this too. This should actually be straight forward. Dunno how progress reporting works here tho.

david-sawatzke, 2020-01-24:
The latest cargo-flash release still has debug messages on my machine:
```
READ RACE WIN 13.993µs, Ok(160)
WRITE RACE WIN
READ RACE WIN 68.798µs, Ok(19)
WRITE RACE WIN
WRITE RACE WIN
WRITE RACE WIN
```

Yatekii, 2020-01-24:
This is kinda bad. A PR that removes them is very welcome, if not, I will try and clean it up once the API changes are done :)

Yatekii, 2022-01-30:
I think most is here and what is not will not get in unless someone will work on it in the future :)

Closing.

closed 2022-01-30 by Yatekii