probe-rs/probe-rs #115: Support ST chip identification

Tiwalun, 2020-01-06:
ST has a proprietary method to identify chips, based on the `DBGMCU_IDCODE` register. This is described in the reference manuals of the chips.

We should support this to properly identify ST chips.

See #113 for more discussion of this.

andresovela, 2020-08-14:
Hi @Tiwalun, I'd like to give this a shot this weekend, but I'd need some tips. It's my first time contributing. I can figure out the stuff out of the reference manual but I wouldn't know where to start in the code.

Tiwalun, 2020-08-14:
Welcome @andresovela :wave:, thanks for helping out!

The autodetect functionality currently starts with the `try_arm_autodetect` function in `session.rs`:

https://github.com/probe-rs/probe-rs/blob/c44a1ec444895594104c901361d645f3376c79aa/probe-rs/src/session.rs#L289

Currently, this function then tries to identify the chip by reading the ROM table, which is accesible over the debug port. It  tries to find a matching chip by comparing the manufacturer ID and peripheral ID from the ROM table with the stored values in the target files (in the `probe-rs/targets` subfolder).

Two changes should be necessary to support the STM chips:

- [ ] The autodect code should be changed so that it reads the `DBGMCU_IDCODE` register, when a STM32 chip is detected.
- [ ] The target description file format has to be adapted so that the ST-specific ID code can be stored.

If you have more questions, feel also free to ask in the [probe-rs channel on Matrix](https://matrix.to/#/#probe-rs:matrix.org).

andresovela, 2020-08-14:
Thanks, I'll see what I can do 👍

Tiwalun, 2020-08-20:
I just found https://docs.rs/stm32-device-signature/, it seems that the flash size can be read from some registers as well. I'm not sure if this works for any chips, however.

mvirkkunen, 2020-08-30:
If the target information is updated to include the address for `DBGMCU_IDCODE` (it seems to be different for some chip families), it would be very nice if `DBGMCU_CR` information could be added because it's part of the same subsystem and should be easy to do in the process. This will be useful for fixing #350.

diondokter, 2020-11-19:
Note that this is partially covered in #431, but it's not as much info because it's taken from the arm romtable.

Yatekii, 2020-11-19:
Yeah the problem here is that the values in the romtables are not unique to one chip but family, unfortunately. This has been confirmed by the author of pyOCD who's an ARM employee even.
Most manufacturers have custom registers in place tho, which uniquely identify the chip if the ARM provided IDs are not unique as they should be as per spec.
So now that 0.10.0 is out, my focus will be on a) hardware testing to prevent any kind of regressions and b) bringing up debug sequences which will bring vendor provided special procedures (such as chip unlocking, etc.) to probe-rs. In the same go, I would like to add a sequence type, which can determine the chip (on a per family basis), effectively making it possible to autodetect the chip!
I am not sure if I have written this down previously, but it has been a hot topic on Matrix some time ago :)