probe-rs/probe-rs #133: Flash an external ELF binary ?

NotSqrt, 2020-01-29:
Hello,

I want to put [some Rust in my C](https://rust-embedded.github.io/book/interoperability/rust-with-c.html), and I want to know if probe-rs will be capable of flashing the resulting binary (rust static lib linked with C code).
I digged in the code, and it seems it currently expect to flash a file from the target directory, not a provided external file.

Thanks !

Yatekii, 2020-01-29:
Hi!

probe-rs as a lib can flash any ELF binary. No matter the compiler.
You are right tho, that our utility - `cargo-flash` - is oriented towards rust projects.

We plan on providing a general utility in the future which can flash any ELF binary.
I hope this answers your question :)

If you are interested in working on this, feel free to ask specifics here or on Matrix!

Best,
Yatekii

Yatekii, 2020-01-29:
Actually, follow up:

@Tiwalun just hinted me that the `probe-rs-cli` might be able to flash an ELF binary. You could try ;)

Tiwalun, 2020-01-29:
To clarify, you should be able to run the following:

```bash
probe-rs-cli download <elf_file> --target <chip>
```
where `<chip>` is the chip you want to flash, and `<elf_file>` the path to an ELF file.

NotSqrt, 2020-01-29:
Thanks a lot !

I see that `--chip-description-path` is not exposed in `probe-rs-cli`, so I can't use the chip family description file that I extracted from a CMSIS-Pack (LPC chip with a Link2 debug probe with ARM’s CMSIS-DAP protocol).
That would be a nice addition !

Yatekii, 2020-01-29:
probe-rs-cli is more of a "demo" binary for us. A standalone application to work with targets is planned but not a top priority atm. Feel free to work on it :)

Also, does your Link2 have the CMSIS-DAP FW or the J-Link one? Because the latter is not upstream supported yet.

NotSqrt, 2020-01-29:
OK, sadly I don't have much time to work on it ..

Regarding the probe, it's this evaluation board, with the default firmware : https://www.nxp.com/design/microcontrollers-developer-resources/lpcxpresso-boards/lpcxpresso54114-board:OM13089
> On-board, high-speed USB based, Link2 debug probe with ARM’s CMSIS-DAP and SEGGER J-Link protocol options

Yatekii, 2020-01-29:
Yes, I know about the board, but it doesn't specify which firmware it's using ;)

Yatekii, 2020-03-30:
Fixed with cargo-flash 0.6.0.

closed 2020-03-30 by Yatekii