probe-rs/probe-rs #137: Clear breakpoints when detaching

Tiwalun, 2020-02-05:
Currently, we do not clear HW breakpoints when detaching. We also do not check when
attaching what the current state of the HW breakpoint unit is.

We should:

- [ ] Clear HW breakpoints when detaching
- [ ] Check the state of the breakpoint unit when attaching

Tiwalun, 2020-02-05:
When attaching, we should probably just clear all breakpoints to ensure we are in a consistent state.

elwerene, 2020-07-19:
Is it ok/fast enough to just clear all possible breakpoint units on attach? Otherwise it would need an implementation for each chip to get the current hardware breakpoints, as this is not yet implemented in probe-rs/I did not find it in the code.

PR #307 (2020-07-19): Clear all hardware breakpoints on attach
closes #137
closed 2020-07-19 by Yatekii