probe-rs/probe-rs #151: Support RISCV chips with only single program buffer.

Tiwalun, 2020-02-26:
The current RISCV debug implementation assumes that all targets have at least two program buffer slots. The debug specification also allows for implementations with only a single program buffer slot, and an implict `ebreak` at the end.

To support this, the number of program buffer slots has to be checked, and the usage of the slots has to be adapted.

See #144

ruabmbua, 2023-03-25:
Good news: seems like I got my hands on HW, that only supports a single progbuf slot. Enough reason to fix this I guess.

Although I think this particular hw might just have a broken debug implementation...

bugadani, 2025-04-28:
If anyone happens to know a device that acts like this, I'd love to source one.