probe-rs/probe-rs #168: Issue connecting to EFR32MG21 Cortex M33 chip on v0.5.0 using STLink V2

EnJens, 2020-03-07:
**Describe the bug**
probe-rs is unable to talk to the AP (CSW register) when using version 0.5.0
(Note, the goal is of course to run it with cargo-flash, but I'm using probe-rs-cli to verify the target yaml and proper connection to the chip)

**To Reproduce**
Run `probe-rs-cli info` with an EFR32MG21 connected with an ST-Link.
This also happens with any other commands, but info is nice and simple.

**Expected behavior**
That it would show me info about the chip.

**Stacktrace**
[probe-rs-cli-0.4.0.log](https://github.com/probe-rs/probe-rs/files/4302521/probe-rs-cli-0.4.0.log)
[probe-rs-cli-0.5.0.log](https://github.com/probe-rs/probe-rs/files/4302522/probe-rs-cli-0.5.0.log)

**Desktop (please complete the following information):**
Ubuntu

**Additional context**
I've verified Openocd is able to connect and talk to the chip (although it lacks specific support for the part, so no flashing).

Yatekii, 2020-03-07:
Thanks for filing this. We'll look into it.

If I understand this right, this is wroking with 0.4.0 but not working with 0.5.0?

EnJens, 2020-03-07:
@Yatekii correct, I've updated the description to clarify this

Yatekii, 2020-03-08:
The actuall error is missing in the log. It is:

Error processing command: A core architecture specific error occured: Failed to write register CSW, address 0x00000000

Yatekii, 2020-04-24:
@EnJens could you test if this is still an issue with the 0.6.1/2 releases?

EnJens, 2020-04-24:
I would, but `Error processing command: The firmware on the probe is outdated`
Not really sure I can update my stlinkv2 clone without bricking it

EnJens, 2020-04-24:
I decided to risk it - `probe-rs-cli info` works correctly now!

EnJens, 2020-04-24:
I tried a simple dump command which still fails (having added the target-gen generated yaml file):
```
RUST_LOG=debug cargo run --bin probe-rs-cli --release -- dump 0 16 --chip EFR32MG21A010F1024IM32
    Finished release [optimized] target(s) in 0.06s
     Running `target/release/probe-rs-cli dump 0 16 --chip EFR32MG21A010F1024IM32`
 DEBUG jaylink > libusb 1.0.23.11397
 DEBUG jaylink > libusb has capability API: true
 DEBUG jaylink > libusb has HID access: true
 DEBUG jaylink > libusb has hotplug support: true
 DEBUG jaylink > libusb can detach kernel driver: true
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 39, bNumInterfaces: 1, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 50, extra: None }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 0, bDeviceSubClass: 0, bDeviceProtocol: 0, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14152, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > STLink version: (2, 36)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 DEBUG probe_rs::probe::stlink                > Successfully initialized SWD.
 DEBUG probe_rs::architecture::arm::communication_interface > Debug Port version: DPv2
 DEBUG probe_rs::architecture::arm::communication_interface > Reading DP register DPIDR
 DEBUG probe_rs::architecture::arm::communication_interface > Read    DP register DPIDR, value=0x6ba02477
 DEBUG probe_rs::architecture::arm::communication_interface > DebugPort ID:  DebugPortId {
    revision: 0x6,
    part_no: 0xba,
    version: DPv2,
    min_dp_support: NotImplemented,
    designer: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
}
 DEBUG probe_rs::architecture::arm::communication_interface > Writing DP register ABORT, value=0x0000003c
 DEBUG probe_rs::architecture::arm::communication_interface > Writing DP register SELECT, value=0x00000000
 DEBUG probe_rs::architecture::arm::communication_interface > Requesting debug power
 DEBUG probe_rs::architecture::arm::communication_interface > Writing DP register CTRL/STAT, value=0x50000000
 DEBUG probe_rs::architecture::arm::communication_interface > Reading DP register CTRL/STAT
 DEBUG probe_rs::architecture::arm::communication_interface > Read    DP register CTRL/STAT, value=0xf0000040
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000010
 DEBUG probe_rs::architecture::arm::communication_interface > Reading register CSW
 DEBUG probe_rs::architecture::arm::communication_interface > Read register    CSW, value=0x43800050
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000EDF0
 DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
 DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x00180000
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000ED30
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register DRW, value=0x0000001F
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0x00000000
 DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Read first block with len 64 at address 0x000000
 DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW, block with len=16 words
 WARN  probe_rs::probe::stlink                                      > check_status failed: SwdDpFault
 DEBUG probe_rs::probe::stlink                                      > Current device mode: Jtag
Error processing command: A core architecture specific error occured: Failed to read register DRW at address 0x0000000c because: An error specific to a probe type occured: Command failed with status SwdDpFault
```

Yatekii, 2020-04-24:
@EnJens I am not sure you can read flash without unlocking first on every chip. Maybe try reading 0x20000000?

EnJens, 2020-04-24:
I'm not really convinced that's it - it works directly on openocd with zero chip-specific support...
I tried the debug command (with no binary) and I get a similar error on `halt` in there.
It's a bit spammy but:
```
RUST_LOG=debug cargo run --bin probe-rs-cli --release -- debug --chip EFR32MG21A010F1024IM32     
    Finished release [optimized] target(s) in 0.05s
     Running `target/release/probe-rs-cli debug --chip EFR32MG21A010F1024IM32`
 DEBUG jaylink > libusb 1.0.23.11397
 DEBUG jaylink > libusb has capability API: true
 DEBUG jaylink > libusb has HID access: true
 DEBUG jaylink > libusb has hotplug support: true
 DEBUG jaylink > libusb can detach kernel driver: true
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 39, bNumInterfaces: 1, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 50, extra: None }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 0, bDeviceSubClass: 0, bDeviceProtocol: 0, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14152, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > STLink version: (2, 36)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 DEBUG probe_rs::probe::stlink                > Successfully initialized SWD.
 DEBUG probe_rs::architecture::arm::communication_interface > Debug Port version: DPv2
 DEBUG probe_rs::architecture::arm::communication_interface > Reading DP register DPIDR
 DEBUG probe_rs::architecture::arm::communication_interface > Read    DP register DPIDR, value=0x6ba02477
 DEBUG probe_rs::architecture::arm::communication_interface > DebugPort ID:  DebugPortId {
    revision: 0x6,
    part_no: 0xba,
    version: DPv2,
    min_dp_support: NotImplemented,
    designer: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
}
 DEBUG probe_rs::architecture::arm::communication_interface > Writing DP register ABORT, value=0x0000003c
 DEBUG probe_rs::architecture::arm::communication_interface > Writing DP register SELECT, value=0x00000000
 DEBUG probe_rs::architecture::arm::communication_interface > Requesting debug power
 DEBUG probe_rs::architecture::arm::communication_interface > Writing DP register CTRL/STAT, value=0x50000000
 DEBUG probe_rs::architecture::arm::communication_interface > Reading DP register CTRL/STAT
 DEBUG probe_rs::architecture::arm::communication_interface > Read    DP register CTRL/STAT, value=0xf0000040
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000010
 DEBUG probe_rs::architecture::arm::communication_interface > Reading register CSW
 DEBUG probe_rs::architecture::arm::communication_interface > Read register    CSW, value=0x43800050
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000EDF0
 DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
 DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x00130003
 DEBUG probe_rs::architecture::arm::core::m33               > Core was halted when connecting
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000ED30
 DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
 DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x00000001
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register TAR, value=0xE000ED30
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register DRW, value=0x0000001F
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG rustyline                                            > initial cursor location: Some(1)
 DEBUG rustyline                                            > old layout: Layout { prompt_size: Position { col: 0, row: 0 }, default_prompt: false, cursor: Position { col: 0, row: 0 }, end: Position { col: 0, row: 0 } }
 DEBUG rustyline                                            > new layout: Layout { prompt_size: Position { col: 3, row: 0 }, default_prompt: true, cursor: Position { col: 3, row: 0 }, end: Position { col: 3, row: 0 } }
>>  DEBUG rustyline                                            > key: Char('h')
 DEBUG rustyline                                            > Emacs command: SelfInsert(1, 'h')
[... cut ...]
 DEBUG probe_rs::architecture::arm::communication_interface         > Reading register DRW, block with len=1 words
 WARN  probe_rs::probe::stlink                                      > check_status failed: SwdDpFault
 DEBUG probe_rs::probe::stlink                                      > Current device mode: Jtag
Error processing command: A core architecture specific error occured: Failed to read register DRW at address 0x0000000c because: An error specific to a probe type occured: Command failed with status SwdDpFault
```

EnJens, 2020-04-24:
also, trying to read e.g. 0x20000000 - which bizarely requires me to input it as decimal - gives me: error: Invalid value for '<loc>': number too large to fit in target type. I've verified it's actually an address in ram. I've attached the generated yaml file (Renamed because GitHub)

[EFR32MG21 Series.yaml.txt](https://github.com/probe-rs/probe-rs/files/4531916/EFR32MG21.Series.yaml.txt)

)

Yatekii, 2020-04-25:
the input is in HEX but you need to ommit the 0x afaik, sorry for that.

As expected from SiLabs, the flash algo looks kinda crooked and I am pretty sure we need to change some stuff about our algo because the sector size is not whole number factor of the page size ... you can try flashing tho :)

The error you get looks kinda familiar, I cannot really pinpoint the exact issue yet tho.

EnJens, 2020-04-25:
I gave flashing a shot too, and ultimately same error:
```
 DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0xfffffffe
 DEBUG probe_rs::flashing::flasher                          > Loading algorithm into RAM at address 0x20000200
 DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write block with total size 496 bytes to address 0x20000200
 DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
 DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0x20000200
 DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write first block with len 496 at address 0x20000200
 DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, block with len=124 words
 WARN  probe_rs::probe::stlink                                      > check_status failed: SwdApFault
 DEBUG probe_rs::probe::stlink                                      > Current device mode: Jtag
Error processing command: Something during memory interaction went wrong: A core architecture specific error occured: Failed to write register DRW at address 0x0000000c because: An error specific to a probe type occured: Command failed with status SwdApFault
```
I guess it doesn't even get to trying out the actual algorithm?

Yatekii, 2020-04-25:
Yeah, the failure happens before flashing takes place. Kinda hard to tell from here what's wrong. But the DRW register is the Data read write register, in short, where you write the actual data of the transfer once you set the address in TAR and have the rest configured via CSW. So it's really odd this fails. There is a few reasons this could happen, but idk atm what's so special about this core :/

EnJens, 2020-04-25:
Is there any way I can provide any useful info from some kind of openocd debug logs?
As previously mentioned, it works out of the box (except flashing of course) on there.

Yatekii, 2020-04-25:
Hmm idk how to actually log low level transfer calls in OpenOCD, but if you can get a log of what they do in terms of ARM register writes, that could help :)

EnJens, 2020-04-25:
This seems to be it :)
[debug.log](https://github.com/probe-rs/probe-rs/files/4533950/debug.log)
(Note that this is a slightly modified local openocd with flash driver implemented for EFR32MG21)

Yatekii, 2020-05-31:
Newer experiences show that maybe, for this chip, an attach under reset is required too!

Do you have any news on this @EnJens?

Maybe someone is in the mood to work on this on the impl days :)

Yatekii, 2020-11-13:
@EnJens could you try this with master of probe-rs?
Also with the same logs enabled. I think this is a general issue for some chips and I would like to verify this :)

EnJens, 2020-11-24:
I hope to get back to it in a week or two and give it a try! Sorry for the delay

Yatekii, 2020-11-24:
All good, take your time :)

mattico, 2021-06-04:
I had this issue with a DAPLink Probe connected to a STM32H743ZIT6 over SWD at 24000kHz. Reducing the speed to 20000kHz stops the issue from happening - or at least its so infrequent that I haven't seen it yet. Might just be a signal integrity issue?

Yatekii, 2022-01-30:
Since this did not get any more attention it seems to be resolved. Feel free to reopen if this is still an issue :)

closed 2022-01-30 by Yatekii