probe-rs/probe-rs #172: Support for the e-link32

hargoniX, 2020-03-08:
**Is your feature request related to a problem? Please describe.**
I can't find any non Windows software that allows me to communicate with the [e-link32](https://www.holtek.com/e-link32_pro) on the eval boards for my holtek chips.

**Describe the solution you'd like**
It would be pretty neat if probe.rs would have support for this debugger and maybe chips of holtek I'm using.

**Describe alternatives you've considered**
Using Windows and ARM Keil or other Windows tooling holtek provides seems to be the only way.

**Additional context**
I'd love to implement this myself however as I've never done anything related to this I am a little lost, I couldn't find any documentation etc. related to how the Holtek software actually talks to the e-link probe. If someone could tell me either how I should debug the software under Windows or can point me to some docs I oversaw I'd happily tackle this myself. In case there isn't any documentation I guess reverse engineering what their standalone tools are doing is the best way?

Yatekii, 2020-03-08:
It is unlinkely that any of the current maintainers will find time to implement this relatively unknown probe. It is hard enough to handle the three probes we have already :)

If you would like to invest the time and implement a driver I am more than willing to assist you with guidance. The matrix channel is a good place to ask more hands on questions.

Basically you need to implement two traits to make your probe work. 
It's for one the [DebugProbe](https://docs.rs/probe-rs/0.5.1/probe_rs/trait.DebugProbe.html) trait which is the generic probe control trait and [DAPAccess](https://docs.rs/probe-rs/0.5.1/probe_rs/architecture/arm/trait.DAPAccess.html) which is the trait that implements the ARM debug interface (DAP).

We would need to put this into the docs somewhere.

hargoniX, 2020-03-08:
It's just that on the dev boards they provide like for example [this one](https://www.atestore.de/en/development-kits/HT-ESK32-30501.html) there is one of these elink32 probes integrated so I thought it might be nice if you could just use the ones you get when buying the board?

therealprof, 2020-03-08:
https://matrix.to/#/!BHcierreUuwCMxVqOf:matrix.org?via=matrix.org&via=tchncs.de&via=x0f.org

@Yatekii Don't have we pointers to matrix on the homepage?

Yatekii, 2020-03-08:
We do have pointers, also in the readme. Maybe it's not obvius enough? I wanted to add some stuff to the webpage anyways.

therealprof, 2020-03-08:
It's quite possible that e-link is just a clone of something like STLink with a different sticker slapped on.

Tiwalun, 2020-03-09:
You could try capturing the USB communication using [Wireshark](https://www.wireshark.org/), then it would be easy to check if it's just a clone of a STLink.

Yatekii, 2020-03-09:
It is actually a CMSIS-DAP probe. We chatted and figured that. He is trying to make it work now :)

hargoniX, 2020-03-09:
Turns out that in order to read device strings under Fedora you need sudo because of the default SELinux limitations it provides, if i run it with that it reads the string correctly and detects it as a CMSIS-DAP probe, thanks for your help!

closed 2020-03-09 by hargoniX

Yatekii, 2020-03-09:
Hurray glad we were able to resolve this so quickly :)