probe-rs/probe-rs #184: STLink-V3 SWD speeds missing

korken89, 2020-03-15:
Hi,
The STLink-V3 supports up to 24 MHz SWD, however the implementation only seems to go up to 4.6 MHz from recent additions.
Putting the issue here so it is not forgotten.

Disasm, 2020-03-15:
That's strange, it works for me. What are you doing to set frequency?

korken89, 2020-03-16:
I did a quick code review and compared the implementation of the st link module to the datasheet, the max speed variant is 4.6 MHz.

How were you able to get it above that?

korken89, 2020-03-16:
Just to be clear, I am referring to this: https://github.com/probe-rs/probe-rs/blob/c97b681abd8be72bf91bb9dc17dd4df50e8038da/probe-rs/src/probe/stlink/constants.rs#L96

korken89, 2020-03-16:
Ah, sorry I misunderstood the usage, here is the special case of the V3 handled: https://github.com/probe-rs/probe-rs/blob/c97b681abd8be72bf91bb9dc17dd4df50e8038da/probe-rs/src/probe/stlink/mod.rs#L58

Closing.

closed 2020-03-16 by korken89