probe-rs/probe-rs #190: Using the nRF52840 target with the CMSIS-DAP by accident will panic

Yatekii, 2020-03-21:
**Describe the bug**
When you attach a CMSIS-DAP probe andtry to flash an ELF binary with --chip nrf52840, an assertion will be triggered:

```
yatekii@arcturo:~/repos/probe-rs/nrf52-hal/boards/nRF52840-DK$ cargo flash --release --chip nrf52840 --example blinky --disable-progressbars
    Finished release [optimized + debuginfo] target(s) in 0.02s
    Flashing /home/yatekii/repos/probe-rs/nrf52-hal/target/thumbv7em-none-eabihf/release/examples/blinky
 WARN  probe_rs::config::registry > Found chip nRF52840_xxAA which matches given partial name nrf52840. Consider specifying it's full name.
thread 'main' panicked at 'assertion failed: `(left == right)`
  left: `4`,
 right: `1`', /home/yatekii/repos/probe-rs/probe-rs/probe-rs/src/probe/daplink/mod.rs:377:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**Expected behavior**
We should
a) remove all assertions as well as panics. panics are not properly catchable and should really not be part of our lib as it will always crash the users program, which is not preferable. Imagine your entire IDE crashing because flashing failed. Great stuff.
b) We should just handle this response properly. We need to investigate what the actual answer of DAP is here. I don't know why the response is 4 bytes. I was convinced this is always 1 byte. Even if it is expected to be 1 byte, we should not panic if it's more.

**Desktop (please complete the following information):**
 - Linux (Ubuntu, 19.10)

evanrichter, 2020-07-19:
not sure why its not linked correctly, but PR #312 addresses this issue

Yatekii, 2020-07-19:
Fixed in #312. Thanks a lot!

closed 2020-07-19 by Yatekii