probe-rs/probe-rs #209: (Better) support for cheap ST-link clones

birkenfeld, 2020-04-02:
As requested in https://github.com/probe-rs/cargo-flash/issues/20, here are the details about my setup.

I'm using an ST-link clone from Aliexpress, as [shown here](http://blog.linuxbits.io/2016/02/15/cheap-chinese-st-link-v-2-programmer-converted-to-black-magic-probe-debugger/) (no modifications applied).  The USB interface registers as
```
Bus 001 Device 008: ID 0483:3748 STMicroelectronics ST-LINK/V2
```

Flashing works but I have to select the correct chip (in my case, stm32f429vgtx) and set the speed argument to 1800 kHz, and I get the following output:
```
    Flashing /home/gbr/devel/stm_display/firmware/target/thumbv7em-none-eabihf/release/display_firmware
        WARN probe_rs::flashing::flasher > Error while waiting for core halted: An error with the usage of the probe occured: Operation timed out.
[... same warning ca. 20 times]
        WARN probe_rs::flashing::flasher > Error while waiting for core halted: An error with the usage of the probe occured: Operation timed out.
     Erasing sectors ✔ [00:00:02] [###########] 128.00KB/128.00KB @  44.14KB/s (eta 0s )
 Programming pages   ✔ [00:00:07] [###########] 128.00KB/128.00KB @  13.10KB/s (eta 0s )
```

When flashing with OpenOCD/gdb, it seems to recognize that the speed needs to be reduced,
 the output is as follows:

```
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.225347
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x20016419
Info : flash size = 1024kbytes
Info : Single Bank 1024 kiB STM32F42x/43x/469/479 found
undefined debug reason 7 - target needs reset
semihosting is enabled
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08006586 msp: 0x10010000, semihosting
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
Info : Padding image section 0 with 4 bytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x20000046 msp: 0x10010000, semihosting
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08006586 msp: 0x10010000, semihosting
```

The OpenOCD config is:
```
source [find interface/stlink-v2.cfg]
transport select hla_swd
source [find target/stm32f4x.cfg]
```
It is also much quicker (3.5s) compared to `cargo-flash` which takes about 11s.

Yatekii, 2020-04-02:
Ok, so I don't think this has anything to do with the clone, but rather with the Firmware itself.

We would have to look at how we set the speed exactly.

Could you maybe add the first few lines of a log of cargo flash? You can enable it via `--log=debug`. The first few lines should contain the speed selected.
HINT: the resulting log is huge, so maybe log it into a file :) Also, if you do that, use `--disable-progressbars` so it will log properly :)

birkenfeld, 2020-04-02:
ok, interesting. I attach two logs, one with --speed and one without.

However, looking at the the files apparently 1800 kHz is selected anyway? So I made some more test runs, and I've now also reproduced the error with the argument.

So there must be another reason for the problem, although it is strange that in the few times I tried before, `--speed` always semeed to fix it.

[cargo_flash_nospeed.log](https://github.com/probe-rs/probe-rs/files/4421840/cargo_flash_nospeed.log)
[cargo_flash.log](https://github.com/probe-rs/probe-rs/files/4421841/cargo_flash.log)

Yatekii, 2020-04-02:
Thanks a lot for the provided information!
I don't see the actuall error anywhere :/ Mind posting that?

birkenfeld, 2020-04-02:
Ah indeed since it is the only thing printed to stderr.

Error failed to flash /home/gbr/devel/stm_display/firmware/target/thumbv7em-none-eabihf/release/display_firmware: Something during memory interaction went wrong: A core architecture specific error occured: Failed to write register DRW at address 0x0000000c because: USB Communication Error

Yatekii, 2020-04-02:
Thanks!

This is real odd, because the set speed should not affect USB at all. It is only a setting for SWD.

birkenfeld, 2020-04-02:
Isn't the USB dongle talking to the chip via SWD?

Yatekii, 2020-04-02:
Yes it is. But this error means that there was really something wrong with the USB communication. not with SWD. But the speed setting only affects SWD.

Odd, odd ...

Disasm, 2020-04-02:
I was able to update the firmware on my ST-LINK clone with the official firmware update utility and it works great!

birkenfeld, 2020-04-03:
@Disasm good idea! In the past I updated all my genuine ST-Links (on Nucleo boards) since they had some problems with OpenOCD before that, but I didn't try updating the Chinese one yet.  Which version of the firmware did you update to?

birkenfeld, 2020-04-03:
Ok, I just went ahead and tried :) With the newest version (V2J36S7) it seems to *mostly* work without `--speed`, but still fails about 1 in 10 times.

(I'm also still getting the warnings about halting the core, and the process still takes 11 seconds compared to OpenOCD's 3 seconds.)

Yatekii, 2020-04-03:
Glad to hear this worked for you!

I am sure there is some kinks here and there, as things are not really well documented and the codebases of different tools also tell different things ;)

About the speed: this is something to be worked on. Factor 4 is a lot. (Even tho I see 9s in your paste? I also thought the speed difference was about factor two.)
Just know that we are working on it, but it is not a top priority atm :)

Also, was this a fresh write to an erased chip or did you already flash that before? Because OpenOCD does page verification while we just always write everything ...

birkenfeld, 2020-04-03:
> About the speed: this is something to be worked on. Factor 4 is a lot. (Even tho I see 9s in your paste? I also thought the speed difference was about factor two.)

I compared `time cargo flash...` against `time gdb ...`.

> Also, was this a fresh write to an erased chip or did you already flash that before? Because OpenOCD does page verification while we just always write everything ...

Yes, it's always the same chip where the always the same firmware is flashed. If OpenOCD checks if it can omit writes, it won't be fair.

Yatekii, 2020-04-03:
Ok, yeah, I am sure OOCD omits some writes. But we should do that too, so that's no excuse :) Just an explanation why it diverges so much. Because normally it's about factor two, I thought and that is because some things need to be optimized :)

Yatekii, 2020-05-31:
I am closing this, as the remaining issues are elsewhere, specifically with our error/warn game, st-chip stability and speed (this has been addressed in some commits by @adamgreig).

closed 2020-05-31 by Yatekii