probe-rs/probe-rs #210: ZZ support / reusability as C library

aep, 2020-04-02:
Hi,

i was wondering what's needed to use this outside of rust.

the library first approach is great, and would make this very reusable. since rust has its own incompatible ABI unfortunately to make it work outside of rust, you'd have to add an extra standard ABI layer. Would that be something you'd be willing to maintain on top?

Yatekii, 2020-04-02:
I personally have any interest in maintaining that. There is a reason we all want to use Rust and I have zero interest in pushing C any further.
If someone else wants to take care of this, I accept any PR with C bindings, I take no issue with that. Just maintaining it myself is not worth my time.

But as I can see, zetz is written in Rust. I don't see why you couldn't just use the probe-rs crate as is?

I am already thinking of how we could make probe-rs useable from the outside. And I am thinking of a Server which you can control via let's say JSON (or whatever it is) commands. That way, we don't have to carry a fragile C API and are also directly compatible with any language that can open a Socket and serialize/deserialize JSON.

The ITM tracer and the VSCode plugin are already implemented like that.

aep, 2020-04-02:
Ok that's unfortunate

closed 2020-04-02 by aep