probe-rs/probe-rs #233: Panic on `probe.attach("somemcu")` when no MCU is connected

korken89, 2020-04-24:
**Describe the bug**
Panic on `probe.attach("somemcu")` when no MCU is connected.

Panic message: `thread 'main' panicked at 'index out of bounds: the len is 1 but the index is 18446744073709551615', /rustc/4fb7144ed159f94491249e86d5bbd033b5d60550/src/libcore/slice/mod.rs:2842:1`

**To Reproduce**
Steps to reproduce the behavior:
1. Run `probe.attach("somemcu")` with no MCU connected

**Expected behavior**
The return of attach should have been an error

Tiwalun, 2020-04-24:
With what probe does this happen?

korken89, 2020-04-24:
CMSIS-DAP running dap42

Tiwalun, 2020-04-24:
I only have a micro:bit to test CMSIS-DAP, where I unfortunately cannot disconnect the MCU to reproduce this. Could you try getting a backtrace, to determine where this happens?

Yatekii, 2020-11-13:
Is this still a problem, btw @korken89?
I am pretty sure I do not have this issue when the target is not powered.

korken89, 2020-11-14:
This is still happening since my last update, but not sure if I'm on 0.8 it
0.9. I will check when I'm back in the office :)

Yatekii, 2022-03-05:
How is the status of this, @korken89? :) Not sure if this is still a thing, but if yes, a backtrace would indeed help :)

closed 2023-01-14 by Yatekii