probe-rs/probe-rs #247: [0.6.2] nRF52840_xxAA: erase_sector fails with error code 1

japaric, 2020-05-15:
## **Describe the bug**

I get "erase_sector fails with error code 1" from both `cargo-flash` (tool) and `probe_rs::flashing::download_file` (library). I've tested with these combinations of probes: (CMSIS-DAP + nRF52840) and (J-Link + nRF52840).

## **To Reproduce**

The ELF I'm using is basically what you get from compiling the snippet with opt-level = 'z':

``` rust
#[cortex_m_rt::entry]
fn main() -> ! {
    loop { asm::nop() }
}
```

``` console
$ cargo flash --chip nrf52840
INFO probe_rs::flashing::download > Found 2 loadable sections:
INFO probe_rs::flashing::download >     .vector_table at 00000000 (1024 byte0)
INFO probe_rs::flashing::download >     .text at 00000400 (112 byte0)
INFO probe_rs::flashing::flasher  > Erasing sector at address 0x00000000
INFO probe_rs::flashing::flasher  > Done erasing sector. Result is 1. This took 20.939951ms
Error failed to flash app: The execution of 'erase_sector' failed with code 1
```

``` rust
let probes = Probe::list_all();
let probe = probes[0].open()?;
let sess = probe.attach("nRF52840_xxAA")?;
let core = sess.attach_to_core(0)?;
flashing::download_file(&sess, path, Format::Elf)?;
```

## **Expected behavior**

I expect both versions to be able to flash the device. OpenOCD is able to program the binary (tested both a nop loop and a blinky program):

``` console
$ openocd -f interface/cmsis-dap.cfg -f target/nrf52.cfg
(..)

$ telnet localhost 4444
> program path/to/elf
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0000104c msp: 0x20040000
** Programming Started **
nRF52840-xxAA(build code: C0) 1024kB Flash, 256kB RAM
Flash protection of this nRF device is not supported
Adding extra erase range, 0x00000470 .. 0x00000fff
** Programming Finished **
```

## **Stacktrace**

See logs in the STR section (I can provide trace logs as well)

## **Desktop**
- Arch Linux x86_64
- cargo-flash 0.6.0
- probe_rs 0.6.2

## **Additional context**

AFAIK the device has no way to set write protection for Flash (it can set write protection for the "User Configuration" section but that's at a different address: 0x1000_1000). Both `cargo flash` and `probe_rs::flashing` also fail after OpenOCD flashed a binary (I tested this in case OpenOCD did remove write protection).

Any other info I can provide to help debug this?

Yatekii, 2020-05-15:
Thanks for reporting this, I'll have a look into it right away. This definitely should work (I test stuff on my devkit with the 840 and the onboard JLink).

The binary is very simple too, so I'd definitely expect that to work!

Yatekii, 2020-05-15:
@japaric so I just successfully flashed a blinky example onto my nRF52840 with

```
cargo flash --chip nrf52840 --release --example blinky
```

on the probe-rs master and 0.6.2.

Here is a complete log

```
yatekii@arcturo:~/repos/probe-rs/nrf-hal/boards/nRF52840-DK$ RUST_BACKTRACE=1 ~/repos/probe-rs/cargo-flash/target/debug/cargo-flash --chip nrf52840 --probe-selector 1366:1015:000683899362 --release --example blinky --log info
    Finished release [optimized + debuginfo] target(s) in 0.02s
    Flashing /home/yatekii/repos/probe-rs/nrf-hal/target/thumbv7em-none-eabihf/release/examples/blinky
        INFO cargo_flash > Protocol speed 0 kHz
        WARN probe_rs::config::registry > Found chip nRF52840_xxAA which matches given partial name nrf52840. Consider specifying its full name.
        INFO probe_rs::flashing::download > Found 3 loadable sections:
        INFO probe_rs::flashing::download >     .vector_table at 00000000 (256 byte0)
        INFO probe_rs::flashing::download >     .text at 00000100 (3272 byte0)
        INFO probe_rs::flashing::download >     .rodata at 00000DC8 (544 byte0)
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception
        INFO probe_rs::flashing::flasher           > Erasing sector at address 0x00000000
        INFO probe_rs::flashing::flasher           > Done erasing sector. Result is 0. This took 95.986903ms
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Breakpoint), new reason is Request
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception
        INFO probe_rs::flashing::flasher           > Flashing page at address 0x000000 with size: 4096
        INFO probe_rs::flashing::flasher           > Flashing took: 233.992089ms
     Erasing sectors ✔ [00:00:00] [###############################################################################]   4.00KB/  4.00KB @  10.72KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [###############################################################################]   4.00KB/  4.00KB @   4.52KB/s (eta 0s )
```

So I wonder what is broken for you.

First I thought that it could be the security feature the nRFs have but that would fail much earlier :/
(The nRFs can lock the MemoryAP which you can only unlock through a special commandsequence on a ControlAP, but as I said, this would explode way faster.).

The bad thing here is: The error code you are getting is the return code of the flash algorithm binary blob. And I have no clue what this specific blob means with a return value of 1.

Maybe we can look at the blob disassembly.

To be honest, I don't think a trace helps, but could you provide a trace with loglevel DEBUG please?

japaric, 2020-05-15:
[DEBUG log](https://gist.github.com/japaric/2b3a2ac82ee9736f8d2891b257656e4d)

japaric, 2020-05-15:
[probe-rs-cli info](https://gist.github.com/japaric/17cc3a223b12a97737b1cb706f384b8c) with the CMSIS-DAP probe (nRF52840 MDK)

Yatekii, 2020-05-17:
After some digging, I most likely found the culprit here: http://openocd.zylin.com/#/c/5522/

@japaric could you try and erase the entire chip via OOCD/Segger tooling and then flash with cargo-flash afterwards to confirm that this is the culprit? :) At least for @jamesmunns this resolved the issue afaik (this is only a temporary solution ofc!).

For the real solution we need custom hooks on a per chip/manufacturer basis. I don't want this in the probe-rs core as this results in the crazy mess OOCD is.
@cbiffle is working on this right now (pdsc -> WASM with a probe-rs executor). I am not sure when it hits, but when it does, it should be straightforward to implement an unlocking routine in notime!

Yatekii, 2020-05-31:
Okay, it is confirmed that a chip erase fixes the issue. This is most likely caused by accidentally written flash lock bytes.
The flash can be cleared with the [nrf-recover](https://github.com/thalesfragoso/nrf-recover) tool.
We should step up our error game here and report properly that this is most likely because of a locked flash and that an erase would help.

@therealprof kindly started to clean up errors the last two days and we can use the same technique to improve erroring.

Dirbaio, 2020-09-03:
Maybe the cause is usage of the [ACL](https://infocenter.nordicsemi.com/topic/ps_nrf52840/acl.html?cp=4_0_0_5_2) peripheral?

The nRF52840 has 2 ways of blocking flash access:

- APPROTECT in the UICR: entirely blocks the debug port (except for the "recover" operation), so this should've failed much earlier, so I don't think that's it.
- ACL peripheral: this is intended for runtime protection (eg a bootloader write-protects itself before booting an application, so the application can't overwrite it). They get cleared on reset.

If it's the ACL, resetting before flashing should fix the issue.

Also, the docs on ACL say " debugger write access to a write-protected region will be Write-Ignored (WI). ", so maybe return value `1` from the flash algorithm means "verify failed"?

Yatekii, 2020-09-10:
How do I get notified about this 7 days later? oO (sorry for offtopic but this is weird ...)

Hmm I agree with all theories. But since a full chip erase solves the problem, and otherwise it's seemingly permanent, I wouldn't expect it to be the runtime ACL? Or what do I miss here?

Yatekii, 2022-01-30:
I will close this since an erase fixes this. Also, newer structures in probe-rs allow to unlock automatically with given permission :)

closed 2022-01-30 by Yatekii