probe-rs/probe-rs #258: probe-rs-server

Yatekii, 2020-05-31:
I started working on a central server which should coordinate different tools accessing a single probe.

This would allow access for RTT, GDB and ITM all concurrently.

An idea here would be to use JsonRPC and then let different consumer/clients poll the probe-rs-server for new ITM data, memory reads/writes, core status, set breakpoints, etc.

The idea would be that every binary tool (cargo-flash, cargo-embed, etc.) can open or connect to a server, which would make it highly flexible in a way such that you could open a server once and connect your RTT logger and GDB to it and you just flash in a separate terminal window, attaching to the same server. Once flashing is done, RTT and GDB would get notified and know that they can resume operations.

Maybe someone has interest on continuing this work during oxidize impl Days :)

hargoniX, 2020-07-11:
I'll be looking into this!

Yatekii, 2022-03-05:
Closing this as we kind of informally decided against such a thing and with the implementation of microsoft DAP, the DAP debugger could be considered as a server for this.

closed 2022-03-05 by Yatekii