probe-rs/probe-rs #266: Flash timeout

FrozenDroid, 2020-06-04:
**Describe the bug**
When trying to flash a debug build to a C030-R412M (STM32F437, ST-link/v2-1), it times out when erasing.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `cargo embed`
2. Wait till it gets at around ~256kb erased
3. ??
3. no profit

**Expected behavior**
No timeout

**Desktop (please complete the following information):**
 - NixOS 20.03

**Additional context**
It only starts to freeze when it's at about 256kb flash erased. So, if I flash a release build which is smaller, and thus erases less, it works.

memory.x
```
MEMORY
{
  FLASH : ORIGIN = 0x08000000, LENGTH = 1024K
  RAM : ORIGIN = 0x20000000, LENGTH = 192K
}
```

FrozenDroid, 2020-06-04:
Correction. It fails at around 128kb. 
With the help of @Yatekii, we managed to increase the timeout during erasing, which helps. This is still a bit hacky so we should find a better solution than just loosening the timeout by feel.

Yatekii, 2020-06-13:
This will be fixed in #271. cargo-flash and cargo-embed then need to be adjusted too.

closed 2020-09-05 by Yatekii