probe-rs/probe-rs #278: Add a getting started for people not acquainted with rust.

UweBonnes, 2020-06-17:
Please add a not what steps are needed to get from a clones repo to exectuables able to talk to a debugger.

Thanks

adamgreig, 2020-06-17:
Are you looking to use probe-rs as a user or a developer?

For users, you should just need to "cargo install cargo-flash", there's a quickstart and more info on the website: https://probe.rs/. There's no need to clone the repository.

As a developer I guess some familiarity with Rust will be required; `cargo build` from the cloned repository is enough to get the basic probe-rs utilities built, but you'll want [cargo-flash](https://github.com/probe-rs/cargo-flash) or [cargo-embed](https://github.com/probe-rs/cargo-flash) checked out too and likely modify their `probe-rs` dependency to use your local checkout, in `Cargo.toml` something like:

```toml
[patch.crates-io]
probe-rs = { patch = "../probe-rs/probe-rs" }
```

UweBonnes, 2020-06-17:
I am looking as a user,

As a blackmagic debug probe developper, I hope to find some debugger software with code that can talk to a GD32FV Longan board with some of the debuggers I have  so that I can either spy on the communciation with the program itself or with a logic analyzer to get Riscv started in BMP.

adamgreig, 2020-06-17:
Oh, cool! Then you probably do just want `cargo install cargo-flash` without needing to check anything out; when running set `RUST_LOG=trace` to emit all the communication buffers.

I don't know about the Longan board though. Possibly something like this will work for you:

```
RUST_LOG=trace cargo flash --chip riscv --elf main.elf
```

UweBonnes, 2020-06-18:
I have (re)found git@github.com:ruabmbua/blackmagic.git, that does a lot of riscv work inside the BMP framework. https://github.com/UweBonnes/blackmagic/tree/ruabmbua extracts those changes and adapts them to BMP head. So probably I do not have to spy with your probe-rs.

Yatekii, 2020-06-23:
Ok so, I guess this specific issue is resolved. But I think, we are at a point, where not only rust folks could be interested.
So we should maybe make a cargo-independent version of cargo-embed/flash and add a guide how to build.

benmkw, 2020-08-29:
> So we should maybe make a cargo-independent version of cargo-embed/flash and add a guide how to build.

So a cargo subcommand such as cargo flash is just a binary named cargo-flash which is installed in the users path. Thus when applying some small fixes cargo-embed I just ran cargo build and then used the executable in the target folder named cargo-embed (https://github.com/probe-rs/cargo-embed) directly much like `./target/release/cargo-embed --help` so I'm not sure a different name would help, maybe its just that not everyone knows how cargo subcommands work (https://doc.rust-lang.org/book/ch14-05-extending-cargo.html) and how to run these binaries without cargo (just like any other program that is build by cargo and results in a binary in the target folder). 
(steps are the same for cargo-embed or cargo-flash, both use probe-rs inside and wrap it with more user friendly commands/ inside an executable)

closed 2020-10-31 by bors[bot]