probe-rs/probe-rs #284: DebugProbe: `speed` method

dbrgn, 2020-06-26:
In the definition of the `DebugProbe` trait, the following method is defined:

```rust
    /// Get the currently used maximum speed for the debug protocol in kHz.
    ///
    /// Not all probes report which speed is used, meaning this value is not
    /// always the actual speed used. However, the speed should not be any
    /// higher than this value.
    fn speed(&self) -> u32;
```

This is confusing. Should this method return the max speed that is *supported* by this protocol or probe? Or is it the speed being *used* right now, as set by `set_speed`?

Suggestions:

- If the method should return the max supported speed (independently from `set_speed`), rename the method to `max_speed`. Otherwise, clarify docs.
- The use of an `u32` to indicate kHz is not ideal. I would have expected Hz from a unitless value. What about a `Frequency` or `Speed` newtype that wraps an `u32` and has appropriate `From` and `Into` implementations?

adamgreig, 2020-06-27:
I believe this is the speed we asked the probe to use, and it might use any speed up to this value (and ideally uses exactly this value), but we might not be able to tell what speed it actually uses or what its maximum supported speed is.

In other words, this definitely isn't the maximum _supported_ speed, and it's best thought of as the current speed, but in reality the actual speed might be lower. In that sense I think the docs are correct but perhaps they could be expanded to clarify.

Yatekii, 2020-06-28:
I agree with adam on this one. Improving the docs is always possible and very welcome, but I think the choice of names here is good.

Concerning using a newtype: I would def. see this as an improvement :)

commit b42c1456 (2022-03-11): Merge #1039 #1040 #1042

1039: Document the main Error enum r=Dirbaio a=Yatekii

1040: Document the debug module r=Dirbaio a=Yatekii

1042: Clarify documentation for probe speed getter (see #284) r=Dirbaio a=Yatekii

Fix #284.
PR #1042 (2022-03-11): Clarify documentation for probe speed getter (see #284)
Fix #284.
closed 2022-03-11 by bors[bot]