probe-rs/probe-rs #289: cargo-flash appears to leave ST-Link v3 in an unusable state

ryan-summers, 2020-07-06:
**Describe the bug**
When running `cargo flash` on a local repository with a [Nucleo H743ZI](https://www.st.com/en/evaluation-tools/nucleo-h743zi.html) (which has an on-board ST-Link v3 probe), the ST-Link on board will illuminate the COM LED of the ST-Link from red to green. Additional attempts to run `cargo-flash` will fail indicating `failed attaching to target`.

**To Reproduce**
1. Run `cargo flash`
2. Observe the COM led on the ST-Link transition from RED to GREEN
3. Attempt to re-run `cargo flash`
4. Observe that probe-rs fails to attach to the target
5. Unplug the ST-Link and replug it. Observe the COM led is now RED
6. Repeat step 1, observe success.

**Expected behavior**
`cargo flash` should properly ensure that the probe is in a usable state after disconnection.

**Stacktrace**
N/A

**Desktop (please complete the following information):**
Ubuntu 20.04

**Additional context**

Section from the datasheet for the Nucleo indicating COM LED state
[image]

Sample of the failure log:
```
ryan@Gabriel:~/repositories/quartiq/minimq/examples/minimq-stm32h7$ cargo flash --chip stm32h743zitx
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
    Flashing /home/ryan/repositories/quartiq/minimq/examples/minimq-stm32h7/target/thumbv7em-none-eabihf/debug/minimq-stm32h7
        WARN probe_rs::probe::stlink > check_status failed: Other(196)
       Error failed attaching to target

Caused by:
    0: An error with the usage of the probe occured
    1: An error specific to a probe type occured
    2: Command failed with status Other(196)
```

Yatekii, 2020-07-06:
Hmm it's odd that this happens to you again.
I definitely had this in the past, but it wasn't a problem recently afaik. I think @therealprof uses it a lot with this or similar devkits. Maybe he likes to share his experience :)

@ryan-summers could you maybe run the same command but with --log=DEBUG such that we can see what's going on in more detail? You could paste it into a gist if it gets long =)

ryan-summers, 2020-07-06:
Yeah, this is 100% reproducible on my end, so I'll grab a log once I'm off work.

therealprof, 2020-07-06:
Hm, not really. I do use a standalone ST-Link v3/SET occasionally to program my custom projects.

Disasm, 2020-07-06:
Can't reproduce: I'm able to run `cargo flash` twice with stlinkv3-mini and stm32f401 board. Probably the problem is somehow connected to the firmware you run on your target.

ryan-summers, 2020-07-07:
@Disasm This is unrelated to loading firmware. If I leave out any chip specification out (such that it never programs the device) or I reconnect the device after the firmware load, the problem is resolved.

Here's a gist showing the two outputs (take note that both fail because I omitted the `--chip` flag for brevity): https://gist.github.com/ryan-summers/35c2796ef3dc66d975765c1b362f01e9

Disasm, 2020-07-07:
@ryan-summers could you try flashing a simple firmware (like blinky without wfi/wfe) with `cargo-flash`?

Yatekii, 2020-07-07:
Yeah, it fails here: https://github.com/probe-rs/probe-rs/blob/e71105f3498e4106ffbdedd00ae0da3d2820f280/probe-rs/src/probe/stlink/mod.rs#L131 Which means that the STLink itself (not the USB device) is in broken state. I guess we need to reset it first somehow :)

Maybe you could add a log::DEBUG here: https://github.com/probe-rs/probe-rs/blob/e71105f3498e4106ffbdedd00ae0da3d2820f280/probe-rs/src/probe/stlink/mod.rs#L335 to see what mode it is in and if it differs in each call :)

Disasm, 2020-07-07:
I wonder if we should switch to SWD if the current mode is Jtag. Mode doesn't contain a variant for SWD so probably Jtag mode is correct for SWD too?

Yatekii, 2020-07-07:
Ah, I was sure we already log it but I didn't find it at first in your log: https://gist.github.com/ryan-summers/35c2796ef3dc66d975765c1b362f01e9#file-cargo-flash-failure-log-L23 here you can see that the mode differs from https://gist.github.com/ryan-summers/35c2796ef3dc66d975765c1b362f01e9#file-cargo-flash-good-log-L19.

Yap, we are in Jtag mode and here: https://github.com/probe-rs/probe-rs/blob/e71105f3498e4106ffbdedd00ae0da3d2820f280/probe-rs/src/probe/stlink/mod.rs#L337 we have no case to leave it :)

Yatekii, 2020-07-07:
> I wonder if we should switch to SWD if the current mode is Jtag. Mode doesn't contain a variant for SWD so probably Jtag mode is correct for SWD too?

SWIM is the SWD mode :) They call it Serial Wire Interface Module instead of Serial Wire Debug ... because we gotta be conistant /s

Disasm, 2020-07-07:
Yes, initial mode is the same for me (first MassStorage, then Jtag).

therealprof, 2020-07-07:
> > I wonder if we should switch to SWD if the current mode is Jtag. Mode doesn't contain a variant for SWD so probably Jtag mode is correct for SWD too?

> SWIM is the SWD mode :) They call it Serial Wire Interface Module instead of Serial Wire Debug ... because we gotta be conistant 

Nope, SWIM is the mode for the STM8.

> I think the correct mode we gotta be in is MassStorage TBH :)

I always disable MS mode now. So annoying and useless.

Yatekii, 2020-07-07:
Oh, mea culpa then :(

Hmm I am out of ideas ... yeah, I don't like the ST-Link for certain reasons :)

Disasm, 2020-07-07:
I downgraded probe firmware to V3J2M1 and now I have exactly this problem. No problems on all more modern versions. So I'd recommend upgrading the firmware with stlink updater utility.

Yatekii, 2020-07-07:
Ah, so it's a FW problem you're saying? :)

ryan-summers, 2020-07-07:
I was wondering if the firmware on the ST-Link may be an issue. This is a built-in st-link, so it's likely been sitting in a warehouse for a while...

Should we add information to users and/or indicate supported firmware versions?

Disasm, 2020-07-07:
@Yatekii yes, at least newer FW doesn't expose this problem. However, probably it's a good idea to add a version check for V3 firmware just like we have for V2 ones.

Yatekii, 2020-07-07:
Who is making a PR? ;P

Disasm, 2020-07-07:
@Yatekii will make in a moment, need to test it first

Yatekii, 2020-07-07:
"fixed" in #291 :)

closed 2020-07-07 by Yatekii