probe-rs/probe-rs #297: Improve J-Link speed

Yatekii, 2020-07-14:
The current implementation of the J-Link driver is a very rudimentary one I wrote.

There is a lot of room for improvement, as the current solution is basically the worst you could imagine.

While it works, it would be nice to have some kind of batching much like the CMSIS-DAP driver has.

Fixing this requires a little knowledge/research of how the ADIv52 works but there is nice docs and a reference implementantion in the CMSIS-DAP driver, so this should be suitable to first contributors and more advanced probe-rs contributors :)

ryankurte, 2020-09-30:
i've been using `cargo-flash` a bit and was feeling rather confused by how long it was taking (especially compared to just `JLinkExe`ing it), extremely glad this is on the list~!

Javier-varez, 2020-12-31:
I have some changes for this, although currently they depend on a fix for the BitIter split_off function in the JayLink crate which hasn't been released yet (I just fixed it yesterday). Just empirically with my config it seems to halve the flashing time from 3 seconds with a test elf file down to 1.5, so it is nice.

https://github.com/Javier-varez/probe-rs/tree/jlink_swd_batching

It would be nice to get some feedback on how to proceed here. I don't think using the git revision for the JayLink crate is a good idea, so it might be worth it to just wait until a new release of the crate.

Tiwalun, 2020-12-31:
Thanks for working on this, @Javier-varez 

It seems our work overlapped a bit, I was working on the same stuff today: #515.

Please have a look there, I will have at your work, to see if we had the same ideas :wink:

Javier-varez, 2021-01-01:
Oh, sorry, I should have commented here I was planning on working on this topic. I didn't think someone else would be working on it, since it was a long-running issue.

I will have a look at your PR and let you know 😃 Did you come across the split_off function issue of dropping more bits than needed in your changes? It seems I only came across it when batching more than one SWD read/write transaction.

Thanks!

Yatekii, 2021-01-01:
No need to be sorry :) We appreciate your work very much :)

Also, two oppinions on a topic is better than one ;)

Yatekii, 2022-01-30:
I consider this resolved :)

closed 2022-01-30 by Yatekii