probe-rs/probe-rs #300: Suddenly unable to flash to `STM32F407VGTx` (STM32F407G-DISC1 ST-LINK V2.1)

mitchmindtree, 2020-07-17:
Thanks for your work on `cargo-flash`!

## The Bug

After (successfully) flashing a particular example onto my STM32F407G-DISC1 last night, `cargo-flash` no longer seems able to attach itself successfully to my board - as a result, I can't seem to flash a new program onto it.

## Details

Last night I was trying some different programs on my new STM32F407G-DISC1 using `cargo flash`. I was able to edit and flash the examples at the [stm32f407g-disc1 board support crate](https://github.com/stm32-rs/stm32f407g-disc) without any issues. Things were looking great! Here's an example of the command I was using:

```
cargo flash --release --example gpio_hal_blinky --chip STM32F407VGTx
```

I then tried adapting the [`rtic-examples` bluepill example](https://github.com/rtic-rs/rtic-examples/tree/master/rtic_v5/bluepill_blinky) to see if I could blink the stm32 crate with blinky. See [here for the adapted example](https://github.com/mitchmindtree/stm32f4xx-rtic-blinky) that I flashed onto the board with the following command:

```
cargo flash --release --chip STM32F407VGTx
```

Everything appeared to work nicely! Here's the output of the seemingly successful flash:

```
    Finished release [optimized + debuginfo] target(s) in 1m 07s                                                                              
    Flashing /home/mindtree/programming/rust/stm32f4xx-rtic/target/thumbv7em-none-eabihf/release/stm32f4xx-rtic                               
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception     
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Breakpoint), new reason is Request    
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception     
     Erasing sectors ✔ [00:00:00] [#################################################################]  16.00KB/ 16.00KB @  37.44KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [#################################################################]   4.00KB/  4.00KB @   3.47KB/s (eta 0s )
```

I can see that the example appears to run as expected, blinking the LED each time the `blink` software task gets called.

However, any of the following times I have attempted to flash a program `cargo-flash` gives a `JtagNoDeviceConnected` error. The first time I encountered the error, the output looked like this:

<details>
  <summary><b>Error output</b></summary>
  
  stdout+stderr:
  
  ```
   Compiling stm32f4xx-rtic v0.1.0 (/home/mindtree/programming/rust/stm32f4xx-rtic)                                                           
    Finished release [optimized + debuginfo] target(s) in 0.75s                                                                               
    Flashing /home/mindtree/programming/rust/stm32f4xx-rtic/target/thumbv7em-none-eabihf/release/stm32f4xx-rtic                               
        WARN probe_rs::probe::stlink > check_status failed: JtagNoDeviceConnected
       Error failed attaching to target

Caused by:
    0: An error with the usage of the probe occured
    1: An error specific to a probe type occured
    2: Command failed with status JtagNoDeviceConnected

Stack backtrace:
   0: cargo_flash::main_try
   1: cargo_flash::main
   2: std::rt::lang_start::{{closure}}
   3: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:52 
      std::panicking::try::do_call
             at src/libstd/panicking.rs:297
      std::panicking::try
             at src/libstd/panicking.rs:274
      std::panic::catch_unwind
             at src/libstd/panic.rs:394
      std::rt::lang_start_internal
             at src/libstd/rt.rs:51 
   4: main
   5: __libc_start_main
   6: _start
[... 199 lines cut ...]
```rust
        // Setup clocks
        let rcc = device.RCC.constrain();
        let _clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(72.mhz())
            .pclk1(36.mhz())
            .freeze();
```

This config is left over from the bluepill config, I managed to miss this when updating the code! I think likely it should have looked like this:
```rust
        let rcc = device.RCC.constrain();
        let _clocks = rcc.cfgr.sysclk(168.mhz()).freeze();
```

Is it possible that my odd clock settings might have caused some issues with the debug interface `cargo-flash` tries to use? Or do you think something else might be going on? Any advice on other approaches to try would be appreciated!

## System Info

cargo-flash 0.8.0
rustc 1.46.0-nightly (feb3536eb 2020-06-09)
cargo 1.45.0-nightly (40ebd5220 2020-06-01)
nixos, linux 5.7.2

Yatekii, 2020-09-09:
This is being tracked in #350 :)

closed 2020-09-09 by Yatekii

sjmackenzie, 2021-03-15:
This worked for me:
* make NRST low by connecting NRST to GND
* power off the board by disconnecting VDD
* power it on, then exactly 1 second later disconnect the NRST from GND
* then run: `openocd -f interface/stlink.cfg -f target/stm32f1x.cfg -c "init" -c "halt" -c "wait_halt" -c "stm32f1x mass_erase 0" -c "sleep 200" -c "reset run" -c "shutdown"`

jamesmunns, 2021-08-30:
Just as a note (I'll mention it on the other issue), but it is also (at least sometimes) possible to fix this using `cargo flash`'s `--connect-under-reset` option.