probe-rs/probe-rs #301: Index out of bounds when USB devices with no language support are connected

jonas-schievink, 2020-07-17:
**Describe the bug**

This line indexes into the list of supported languages, which may be empty for some devices (encountered in the wild: Saleae logic analyzers):
https://github.com/probe-rs/probe-rs/blob/2bad253cd304e0db6ac9951aba9b58107d6103f3/probe-rs/src/probe/daplink/tools.rs#L33

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a "bad" USB device
2. Use any probe-rs tooling that enumerates USB devices (so most of them)

**Expected behavior**
The probe should just be filtered out instead of causing a panic.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - macOS Mojave
 - Windows 10
 - Linux (Ubuntu, Gentoo, Debian, Fedora, etc.)

**Additional context**
Add any other context about the problem here.

Yatekii, 2020-07-17:
Uh, this is a very recent change. I thought whether this could produce any problems but tested it with like 5 probes. What I didn't think of was that there could be non-probe devices with that issue.

Should be an easy fix, thanks for reporting!

kellerkindt, 2020-07-19:
I'd like to give this a try as it seems reasonable for a first time contributor (like me)?

Yatekii, 2020-07-19:
Fixed in #305 :)

closed 2020-07-19 by Yatekii

dbrgn, 2020-08-20:
Hehe, just ran into this as well, also with a Saleae Logic device (a clone, although I also own the original). Thanks for fixing, @kellerkindt!