probe-rs/probe-rs #310: Flashing Chips with TrustZone

obraunsdorf, 2020-07-19:
It seems like the target yaml files that correspond to TrustZone enabled chips contain flashing algorithms for both secure world and non-secure world. But it seems like there is no support in probe-rs to flash to the respective region based on the the binary's `.text` section's address or some other indicator.
I don't know how to implement such support or even if probe-rs is the right place for that.
What do you think?

Yatekii, 2020-07-19:
In the target description file, you can see that every flash algorithm has a memory range attached. probe-rs determines for every sector in flash in that range it fits and which flash algorithm it has to use :) So the user doesn't have to configure anything, other than properly linking such that the sections are at the right memory locations :)

adamgreig, 2020-07-19:
Does it matter that the targets only have a single `Flash` entry at 08000000, even though they have two algorithms for the two ranges?

Yatekii, 2020-07-19:
Hmm this is actually a good question. I think so far we have not really tested this with non contiguos memory locations. So I am pretty sure this would fail if there is two different flashes, which I guess is the case for TrustZone based things?

So we would need all flashes there is :)

adamgreig, 2020-07-19:
I _think_ the device has one flash region, but it is aliased at both 0x0800_0000 and 0x0C00_0000, with one alias for nonsecure access only and one access which may be secure. I think the linker script and ELF have to contain the right alias, so probe-rs would end up trying to write to one or the other, but I'm not totally sure of the details.

Yatekii, 2020-07-19:
Oh, I see. Well, if we only have a flash region from N to O and the ELF section is not contained in that, no matter the list of flash algorithms, probe-rs will complain. So we definitely have to include all regions for that :)