probe-rs/probe-rs #315: write batching can cause run() to return while core is still halted

bcantrill, 2020-07-20:
With #221, batched writes were introduced for DAPLink.  This is a huge performance win, but it relies on reads (or a detach) to actually issue any batched writes.  Normally, this isn't an issue, as many writes are often followed by reads to observe changed state.  However, there are classes of writes that In particular, `run()` will issue only writes; if those writes are batched, control flow will return to the caller without the core being, in fact, resumed.  In discussing this issue with @adamgreig and @Yatekii, we agreed that the best way to fix this would be to introduce a `flush` operation on the `Core`, exporting that to users of `Core` and calling it explicitly in the `run` implementations.  Here is a prototype that implements this:

  https://github.com/oxidecomputer/probe-rs/commit/a12a8ffc0d537fb74d31fc5dc1e763a5deda9fc8

This approach seemed to work well; assuming others agree, once `itm` lands (see #256), happy to turn this into a PR...

Yatekii, 2020-07-20:
This is looking good to me! And I also just discussed this with @Tiwalun and he also mentions that a `flush()` method was already discussed once when we had issues with the ST-Links :)

So I think this is good to PR, thank you very much!

adamgreig, 2020-08-10:
This was merged in 523c04065145b83150fb1d4170f02895d6704d3a.

closed 2020-08-10 by adamgreig