probe-rs/probe-rs #324: `reset_and_halt` resumes currently-halted execution before halting

adamgreig, 2020-07-24:
I'm debugging connecting under reset for targets that go to sleep or disable their debug interface immediately on boot. In this case the core is already halted by the time the flashing routine is called (to prevent it going to sleep). The flashing routine calls `reset_and_halt` which calls `debug_core_start`:

https://github.com/probe-rs/probe-rs/blob/29df73ca886c5272f8887d16294c3e41c149f8ae/probe-rs/src/architecture/arm/core/m4.rs#L576-L578

However `debug_core_start` writes DHCSR from 0, without first reading its current value:

https://github.com/probe-rs/probe-rs/blob/29df73ca886c5272f8887d16294c3e41c149f8ae/probe-rs/src/architecture/arm/core/mod.rs#L16-L21

I think `C_DEBUGEN` should always already be set at this point without needing to call `debug_core_start`, but because it overwrites DHCSR, if the core is already halted (for example because we attached under reset and then requested a halt on start), it will clear `C_HALT`, causing the processor to resume execution. At that point the processor immediately goes to sleep or whatever and the next command to read some other register fails.

I'm not sure what the ideal solution is here (or this would be a PR!), should we just remove the call to `debug_core_start`, or make a new call `debug_ensure_started` which reads DHCSR and then sets `C_DEBUGEN`, or should `debug_core_start` always read DHCSR first and then only modify it to set `C_DEBUGEN`?

adamgreig, 2020-07-24:
@Tiwalun in https://github.com/probe-rs/probe-rs/pull/325#issuecomment-663358506 you said 

> To solve the issue with reset_and_halt, I would suggest to remove the call to debug_core_start entirely, it shouldn't be necessary in that function. The debug mode of the processor should already be active at that point, as it's set when connecting.

I can't actually see where it's set. There's no other calls to `debug_core_start` except the ones we're proposing to remove and the new one for attaching under reset. The only other place `C_DEBUGEN` gets set is in `core::halt`/`core::step`/`core::run`. As it happens `Flasher::init` calls `halt()` so it gets set eventually, but you could definitely create a session and get a core and end up with debug not enabled. Perhaps `reset` and `reset_and_halt` should set `C_DEBUGEN` (as they do now) but without clearing `C_HALT`? Or should we add a call to `debug_core_start` somewhere else (maybe in session irrespective of whether we're attaching under reset)?

Tiwalun, 2020-07-24:
I thought we set it already, but it seems that you're right :sweat_smile: .

I would suggest that we add a call to `debug_core_start` after https://github.com/probe-rs/probe-rs/blob/29df73ca886c5272f8887d16294c3e41c149f8ae/probe-rs/src/architecture/arm/communication_interface.rs#L211-L219

This would then fit with the sequence described in https://arm-software.github.io/CMSIS_5/Pack/html/debug_description.html#usage_of_sequences .

The goal is to add support for these sequences soon, so we should start working in that direction.

adamgreig, 2020-07-24:
That would first be called at the start of `Session::new()` when it calls `get_target_from_selector`, right? So in that case we could remove it from below where it's currently inside the `AttachMethod::UnderReset` if statement?

Tiwalun, 2020-07-24:
I've made a PR, but decided to keep it inside the `Session` for now. The `debug_core_start` function needs a `Core`, which is hard to get in the `ArmCommunicationInterface`.

PR #326 (2020-07-24): Always call debug_core_start when attaching
Always call `debug_core_start` when attaching. This ensures that the debug mode is always activated when a `Session` is created, regardless of the attach method.

The `reset_and_halt` function now doesn't  call the `debug_core_start` function anymore, which fixes #324 .

Tested with a board which requires attach under reset, and a board which does not require it.
closed 2020-07-24 by Yatekii