probe-rs/probe-rs #343: Ethernet-based probe support

mitchmindtree, 2020-08-03:
## Motivation / Idea

I plan to work with a large number (~500) of `STM32F107`s over an Ethernet network. In order to simplify the process of uploading new firmware to each node, I'm using a fork of [`blethrs`](https://github.com/airborneengineering/blethrs), an Ethernet bootloader for STM32F (currently only the F407) devices.

My fork of `blethrs` has a custom CLI. The original does too, though it is written in python. Both CLIs are extremely basic and no-where near as nice or well-thought-out as `cargo embed` or `cargo flash`. It would be great if, rather than rolling a separate tool, we could simply implement an Ethernet-based `probe-rs` `DebugProbe` and take advantage of the existing `probe-rs` infrastructure!

As someone relatively new to the embedded ecosystem I'm not entirely sure what the best approach to something like this should be.

## Desired Features

It would be nice to have all the existing goodies in `probe-rs`, but in particular I'm interested in:

1. **Discovery** - It would be great to have the ability to enumerate all of the available Ethernet devices. Perhaps this would require some standard way of doing device discovery. Perhaps some sort of multicast-based approach like [SSDP](https://en.wikipedia.org/wiki/Simple_Service_Discovery_Protocol)? 
2. **Flashing** - Uploading new firmware via Ethernet would be great. Perhaps the existing work around `blethrs` could be leveraged?
3. **RTT** - It's incredibly useful to be able to use `rtt_target` and `panic_rtt_target` and suddenly have basic console access to the device. Having something like this over Ethernet would be just as useful.
4. Probably many more probe-rs features I'm yet to discover or take advantage of!

## Further Thoughts

One thing that occurs to me is that support for probe protocols like SWD and JTAG appear to be built *into* the processors themselves and have dedicated pins on the chip that allow special control to interrupt a running process at any point. On the other hand, this hypothetical Ethernet probe would likely need some specific bootloader-esque code on the device itself. Not only bootloading, but runtime interaction such as "halting" a device or sending/receiving via RTT would need some sort of integration in the app firmware too.

Perhaps some `eth-probe-rs` crate could provide some setup and Ethernet interaction functions that have to be plugged into the main device firmware in a particular way? Compared to the existing probes that do not require any firmware integration at all, I imagine this approach might appear very prone to failure. It *might* be a little easier if the probe could consume the whole ethernet device on the firmware side and register its own interrupts, however if we want to be able to share the ethernet device with the rest of the firmware code (as I do in my case) then this isn't really feasible.

If you or anyone reading this could give any advice or thoughts on how you would approach something like this, that would be greatly appreciated! My current project has a looming deadline, so I'm trying to weigh up the feasibility of implementing something like this in time or if I'm better off just sticking with my fork of `blethrs` and hacking in my own project-specific discovery and RTT alternative.

Yatekii, 2020-08-23:
Hi! Sorry for the huge delay. I was on holiday when I saw the issue pop up and completely forgot about it ...

So an ETH based probe (as well as a Wifi & BLE one) is my ultimate goal.
We don't need anything special other than to use TCP (or UDP) packets as the transport layer instead of USB HID/WinUSB and then have a CMSIS-DAP capable probe.
If we have that, everything else will work already. So basically the chain of command would look like this:

probe-rs (control layer) -> ETH-probe-driver (TCP/UDP packet rx/tx) -> ETH -> ETH-probe (implements the CMSIS-DAP protocol from arm via TCP/UDP packets instead of USB) -> SWD/JTAG -> Core

There is nothing more required to flash, use RTT, etc. We'd just need a probe that is capable of this and according firmware :)
There are J-Links that have ETH support, but discovery is quite lacking and I am 99% sure the protocol is not documented. Also, those probes are 800$+.

We have an OSS probe in the pipeline (beta PCBs are underway). After we have a more or less polished firmware for that, we can start looking elsewhere.

Does this help a little? Sorry I didn't answer earlier. I hope it didn't mess too much with your deadline :/

Ridtt, 2023-10-09:
How is this topic going? Want to build smart house with alot of stm32's connected via ethernet. All goals (like discovery, rtt, flashing) would be super useful. Switching on async rust in full time job too, so those features would be constantly reused, its just a question where they would be implemented. As part of probe-rs would ultimative feature.

kokjo, 2025-03-30:
... So i have made an TCP/IP based probe using an esp32c3: https://github.com/kokjo/esp-swd-probe

Currently there is no discovery and i have simple hardcoded the address:port into an serial number in a `ProbeFactory`
a fork of the probe-rs project for working with this probe is available here: https://github.com/probe-rs/probe-rs

I'm unsure of how to proceed with the development of this probe as it requires some modification to core probe-rs features(probe enumeration and selection/opening) to really be usable.

These projects, esp-swd-probe and the fork of probe-rs, is only really meant as a toy and are not production ready.

Ping, @Yatekii any suggestions for a way forward?

bugadani, 2025-03-30:
> I'm unsure of how to proceed with the development of this probe as it requires some modification to core probe-rs features(probe enumeration and selection/opening) to really be usable.

Take a look at what is being done in https://github.com/probe-rs/probe-rs/pull/3195 for the blackmagic probe. Similar things may work for your probe, too.

bugadani, 2025-03-30:
@kokjo have you taken a look at https://github.com/[probe-rs/dap-rs](https://github.com/probe-rs/dap-rs) yet? I have a very basic, and old implementation of it at https://github.com/bugadani/espdap, it would be awesome if we could get a standard CMSIS-DAP working over TCP.

kokjo, 2025-03-30:
@bugadani with `dap-rs` it would be pretty simple to make a tcp based shim that simply transfers dap/usb packets over tcp to the probe-rs host. i was not aware of the dap-rs project and i made the esp-swd-probe code mostly as an exercise to understand the swd protocol better.

bugadani, 2025-03-30:
My main point with dap-rs is that we could use the CMSIS-DAP driver, we would just need to implement a TCP/IP transport instead of USB for it. You wouldn't need a completely new driver, and that driver wouldn't need to bit-bang SWD, which would probably make the end result a bit faster, and more flexible.

kokjo, 2025-03-30:
@bugadani I fully understand that. Do you think that the transport part of CMSIS-DAP dirver in probe-rs is easy make more flexible?

Also minor correction: My probe driver(the probe-rs part) does not do bit-banging, the bit-banging happens on the probe which is where it needs to happen anyway... My probe pretty much just implements `swj_sequence` as well as read/write of dp/ap registers. However there is currently no acceleration of multiple memory accesses as there are in CMSIS-DAP so every read/write to dp/ap requires a roundtrip over wifi which is somewhat slow.