probe-rs/probe-rs #349: STLINK-V3: Intermittent failure to write register

mattico, 2020-08-18:
I'm using an STLINK-V3 in VCP mode to program a custom board with an STM32H743BIT6. I'm using [probe-run](https://github.com/knurling-rs/probe-run) built with HEAD (c44a1ec), but I believe the issue is in probe-rs.

When attempting to write registers, there is an intermittent failure. It fails on different registers at different addresses. Usually there's an `SwdApError` but once it printed something like "access port 0 does not exist for this device". Sometimes the error occurs immediately upon starting, sometimes it doesn't happen for minutes. Enabling more logging seems to slow down the incidence of the issue.

Thinking it was an intermittent communication issue, I added retry logic: https://github.com/probe-rs/probe-rs/compare/master...mattico:retry-probe-error This did not solve the problem.

I'd appreciate some help narrowing this down.

```
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Read register    DRW, value=0x00010001
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Writing register CSW, value=0x63000012
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 0, 0, 12, 0, 0, 63] to STLink, timeout: 1s
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Writing register TAR, value=0x20000024
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 4, 0, 24, 0, 0, 20] to STLink, timeout: 1s
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface] Read first block with len 8 at address 0x20000024
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Reading register DRW, block with len=2 words
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 45, 0, 0, c, 0] to STLink, timeout: 1s
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 45, 0, 0, c, 0] to STLink, timeout: 1s
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface] Finished reading block
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Writing register CSW, value=0x63000012
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 0, 0, 12, 0, 0, 63] to STLink, timeout: 1s
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Writing register TAR, value=0x200003C4
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 4, 0, c4, 3, 0, 20] to STLink, timeout: 1s
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface] Read first block with len 60 at address 0x200003c4
[2020-08-18T19:07:04Z DEBUG probe_rs::architecture::arm::communication_interface] Reading register DRW, block with len=15 words
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 45, 0, 0, c, 0] to STLink, timeout: 1s
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] Error reading register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 45, 0, 0, c, 0] to STLink, timeout: 1s
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] Error reading register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 45, 0, 0, c, 0] to STLink, timeout: 1s
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] Error reading register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:04Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 45, 0, 0, c, 0] to STLink, timeout: 1s
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:04Z WARN  probe_rs::probe::stlink] Error reading register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:05Z DEBUG probe_rs::architecture::arm::communication_interface] Writing register CSW, value=0x63000012
[2020-08-18T19:07:05Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 0, 0, 12, 0, 0, 63] to STLink, timeout: 1s
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] Error writing register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:05Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 0, 0, 12, 0, 0, 63] to STLink, timeout: 1s
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] Error writing register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:05Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 0, 0, 12, 0, 0, 63] to STLink, timeout: 1s
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] Error writing register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:05Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f2, 46, 0, 0, 0, 0, 12, 0, 0, 63] to STLink, timeout: 1s
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] check_status failed: SwdApError
[2020-08-18T19:07:05Z WARN  probe_rs::probe::stlink] Error writing register, retrying ProbeSpecific(CommandFailed(SwdApError))
[2020-08-18T19:07:05Z WARN  probe_rs::session] Could not clear all hardware breakpoints: ArchitectureSpecific(RegisterWriteError { address: 0, name: "CSW", source: ProbeSpecific(CommandFailed(SwdApError)) })[2020-08-18T19:07:05Z TRACE probe_rs::probe::stlink] Getting current mode of device...
[2020-08-18T19:07:05Z TRACE probe_rs::probe::stlink::usb_interface] Sending command [f5] to STLink, timeout: 1s
[2020-08-18T19:07:05Z DEBUG probe_rs::probe::stlink] Current device mode: Jtag
Error: Error communicating with probe: A core architecture specific error occured

Caused by:
    0: A core architecture specific error occured
    1: Failed to read register DRW at address 0x0000000c because: An error specific to a probe type occured
    2: An error specific to a probe type occured
    3: Command failed with status SwdApError
error: process didn't exit successfully: `probe-run --chip STM32H743BITx --defmt target\thumbv7em-none-eabihf\release\mainboardfw-rs` (exit code: 1)
```

Yatekii, 2020-08-19:
Hi!

I am quite positive that this is due improperly (or not at all) initializing the debug modules of the H7 cores. I thought @hargoniX used some of them successfully. Did you check that the memory ranges etc are correct in the yaml description file of those cores?

hargoniX, 2020-08-19:
I was working with an st-link v2 + another variant of the H743 (mine is a H743ZIT6) but your variant looks at least closely related for me, maybe the stlink v3 could be another potential error source here if the YAML files turn out to be fine.

mattico, 2020-08-19:
The memory ranges in HEAD are correct. I'll see if I can test with an STLINK-V2.

mattico, 2020-08-26:
This error still happened when using a J-Link EDU with the WinUSB driver. It is probably an issue in how probe-rs is initializing the debug registers on the H7.

Yatekii, 2021-02-26:
Does this still happen?

mattico, 2021-02-27:
I've been using the J-Link without issues, but the STLINK-V3 didn't work a few days ago if I remember correctly. I'll try to test it more deliberately.

Yatekii, 2021-03-01:
That is very sad :(

Could you try with today's master? Because I currently use an onboard ST-Link v3 on my STM32H747-DISCO board very successfully and without issues. But only after fixing: https://github.com/probe-rs/probe-rs/pull/553 which was 8 days ago while your commit seems to be 12 days ago?

Could you also be so kind and try with `cargo-flash` https://github.com/probe-rs/cargo-flash/tree/improve-flash-errors-yatekii from this branch, so we could rule out any differences in config that might be triggering the issue?

Also, could you check the FW version of your ST-Link? It should be V3J7Mx, which is the newest one I believe.

mattico, 2021-03-01:
That fixes it! Thank you!

I'm happy to stop using the J-Link because its flashing and RTT are very slow (with probe-rs).

Yatekii, 2021-03-01:
> I'm happy to stop using the J-Link because its flashing and RTT are very slow (with probe-rs).

Is it still that slow with master? It should have received a significant speed bump, even tho it is still slower than their official tools (This will always be because we cannot use the proprietary fast API :().

> The error only occurs when there are many RTT messages sent quickly (or some other activity?).

I get that too. I wonder why. Guess we need to investigate that :) Will do in the comming days!

Tiwalun, 2023-01-18:
This might be fixed by #1406. Can you please check if the current version (0.14.2) still has this issue?

mattico, 2023-02-01:
I've been using the hs-probe for a long time, so I'm not sure. Okay with closing this, though.

closed 2023-02-01 by mattico