probe-rs/probe-rs #352: Attaching to FE310 fails with HaltResume

yvt, 2020-08-26:
**Describe the bug**
Attaching to FE310-G002 fails with `Error occured during execution of an abstract command: HaltResume`.

**To Reproduce**

```shell
git clone https://github.com/probe-rs/probe-rs.git
git checkout 79a91c1f5e11090a8344546e7b0db97f32963d98
cargo run -p probe-rs-cli -- debug -c riscv
```

This produces the following output:

```
Error: A core architecture specific error occured

Caused by:
    Error occured during execution of an abstract command: HaltResume
```

**Expected behavior**
`probe-rs-cli` successfully attaches to the target and displays `>>`.

**Desktop (please complete the following information):**
 - macOS Catalina

**Stacktrace**

The root cause of `HaltResume` (“The abstract command couldn’t execute because the hart wasn’t in the required state (running/halted), or unavailable.”) is this call to `abstract_cmd_register_read`:

https://github.com/probe-rs/probe-rs/blob/79a91c1f5e11090a8344546e7b0db97f32963d98/probe-rs/src/architecture/riscv/mod.rs#L70-L72

`write_csr` was called by:

https://github.com/probe-rs/probe-rs/blob/79a91c1f5e11090a8344546e7b0db97f32963d98/probe-rs/src/architecture/riscv/mod.rs#L377

Going up the stack trace, we eventually reaches here:

https://github.com/probe-rs/probe-rs/blob/79a91c1f5e11090a8344546e7b0db97f32963d98/probe-rs/src/session.rs#L129

Yatekii, 2020-08-26:
Thanks for reporting this :)

Maybe @Tiwalun or @Disasm have any ideas :)

Tiwalun, 2020-08-26:
A quick bisect shows that 696ad2f61892d27626c2e619c0fe019df8ade5df seems to be the commit causing this.

As a workaround, commenting out this line works:

https://github.com/probe-rs/probe-rs/blob/79a91c1f5e11090a8344546e7b0db97f32963d98/probe-rs/src/session.rs#L129

commit ade1c152 (2020-08-28): Merge #355

355: Halt RISCV cores on attach r=Yatekii a=Tiwalun

To clean all HW breakpoints on RISCV, core needs to be halted.

This is also more consistent with the behaviour on ARM.

Should fix #352
PR #355 (2020-08-28): Halt RISCV cores on attach
To clean all HW breakpoints on RISCV, core needs to be halted.

This is also more consistent with the behaviour on ARM.

Should fix #352
closed 2020-08-28 by bors[bot]