probe-rs/probe-rs #357: "no probe found" when unsufficient permissions to open the USB device

japaric, 2020-09-01:
**Describe the bug**
On Linux, if you forget to set udev rules to use a (USB) probe as a non-root user `Probe::list_all` will not include it.

**To Reproduce**
Steps to reproduce the behavior:
`probe-rs-cli list` returns "no probe found"
`sudo probe-rs-cli list` returns the probe
`Probe::list_all()` returns an empty list when the program is not run as root

**Expected behavior**
`probe-rs-cli list` returning "failed to open USB devices due to permissions" would make easier to debug the issue.
`Probe::list_all()` returning all devices, even those which cannot be opened, would be helpful.

**Desktop (please complete the following information):**
 - Arch Linux

adamgreig, 2020-09-01:
What sort of probe is this with? The cmsis-dap probes should open in HID mode without permissions but require udev rules for v2 mode, but probably st-links (and perhaps jlinks) require the udev rules all the time? Each type of probe is responsible for generating its own list of devices, so this behaviour is per-probe-backend rather than generic.

japaric, 2020-09-02:
Probe: `LPC11U3x CMSIS-DAP v1.0.7 (VID: 1fc9, PID: 0132, Serial: <redacted>, DAPLink)`
I got the empty list on a fresh Ubuntu install. `lsusb` did list the probe with all its USB interfaces (VCOM, HID, etc.)

The device is detected on my Arch Linux system but I have several udev rules enabled. If I add a udev rule to set the mode of the device to "0000" then the device is no longer listed by `probe-rs-cli list` unless I run the command as root.

adamgreig, 2020-09-05:
I've been looking into this a bit (it turns out if you apt install openocd on Ubuntu it adds its own rules file which allows access to all devices with CMSIS-DAP in the ID...).

If the user doesn't have permission to open the device (via a udev rule), then both hidapi and rusb cannot read the product string, and the only way to detect a CMSIS-DAP device is from that product string containing "CMSIS-DAP". I think that even though you could talk to the device over HID without permission, you can't read the product string so can't know it's a probe. Consequently as far as I can tell there's no general way to determine that there's a probe we don't have permission to open. If anyone knows any other way I'd love to hear it. We can't just warn that there was _some_ device we couldn't open, because almost all other system USB devices will appear like that.

We could ship a list of known USB VID:PID pairs for probes and warn if any of those VID:PIDs are found which cannot be opened, which might at least point some users in the right direction some of the time. We could also change the error message when no devices are found to suggest checking drivers are installed/permission rules are set, perhaps with a pointer to a README.

stappersg, 2020-10-11:
> We could also change the error message when no devices are found to suggest checking drivers are installed/permission rules are set, perhaps with a pointer to a README.

Has the advantage over 
> We could ship a list of known USB VID:PID pairs for probes and warn if any of those VID:PIDs are found which cannot be opened, which might at least point some users in the right direction some of the time.

of never having a incomplete list.