probe-rs/probe-rs #364: nrf52 flash algorithm disallows overwriting a softdevice

Dirbaio, 2020-09-03:
**Describe the bug**
the nrf52 flash algorithm has a hardcoded check that disallows erasing/writing pages when a softdevice is present.

**To Reproduce**
- `pyocd erase --chip --target NRF52840`
- `pyocd flash --target NRF52840 third_party/nrf5_sdk/components/softdevice/s140/hex/s140_nrf52_7.0.1_softdevice.hex `
- try to use probe-rs to flash something that overwrites 0x0 -- 0x27000 (the size of that softdevice).

relevant logs: 

```

[2020-09-03T17:22:01Z INFO  probe_rs::flashing::download] Found 4 loadable sections:
[2020-09-03T17:22:01Z INFO  probe_rs::flashing::download]     .vector_table at 00027000 (256 byte0)
[2020-09-03T17:22:01Z INFO  probe_rs::flashing::download]     .text at 00027100 (108888 byte0)
[2020-09-03T17:22:01Z INFO  probe_rs::flashing::download]     .rodata at 00041A58 (22204 byte0)
[2020-09-03T17:22:01Z INFO  probe_rs::flashing::download]     .data at 20010000 (7620 byte0)
...
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::loader] Using builder for region (0x00000000..0x00100000)
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::loader] Algorithm nrf52xxx - start: 0x000000 - size: 0x200000
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::loader] Algorithm nrf52xxx_uicr - start: 0x10001000 - size: 0x001000
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::loader] Algorithm nrf52xxx_sde - start: 0x000000 - size: 0x200000
..
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::flasher] Full Chip Erase enabled: false
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::flasher] Double Buffering enabled: false
[2020-09-03T17:22:01Z DEBUG probe_rs::flashing::flasher] Initializing the flash algorithm.
...
[2020-09-03T17:22:01Z INFO  probe_rs::flashing::flasher] Erasing sector at address 0x00020000
...
Error: Error while flashing

Caused by:
    The execution of 'erase_sector' failed with code 103
```

**Additional context**

Excerpts from the flash algorithm decompiled with Ghidra:

```c

undefined4 erase_sector(uint param_1)

{
  uint uVar1;
  int iVar2;
  int unaff_r9;
  
  *(undefined4 *)(unaff_r9 + 4) = 2;
  *(undefined4 *)(unaff_r9 + 8) = 0;
  *(undefined4 *)(unaff_r9 + 0xc) = 0;
  *(undefined4 *)(unaff_r9 + 0x10) = 0;
  uVar1 = read_ficr_codepagesize();
  if (param_1 != (param_1 / uVar1) * uVar1) {
    *(undefined4 *)(unaff_r9 + 0xc) = 3;
    *(uint *)(unaff_r9 + 0x10) = param_1;
    return 0x65;
  }
  *(undefined4 *)(unaff_r9 + 8) = 1;
  uVar1 = ret0();
  if ((uVar1 <= param_1) && (uVar1 = total_flash_size(), param_1 < uVar1)) {
    *(undefined4 *)(unaff_r9 + 8) = 2;
    iVar2 = if_addr_in_softdevice_2_else_3(param_1);   // <<<<<<< HERE is the check
    if (iVar2 == 3) {
      do_erase_sector(param_1);
      return 0;
    }
    return 0x67; // THIS is the 103 return code that we/re seing
  }
  *(undefined4 *)(unaff_r9 + 0xc) = 4;
  *(uint *)(unaff_r9 + 0x10) = param_1;
  return 0x66;
}

undefined4 if_addr_in_softdevice_2_else_3(uint param_1)

{
  undefined4 uVar1;
  
  if (_DAT_00003004 == 0x51b1e5db) {
    uVar1 = 3;
    if (param_1 < _DAT_00003008) {
      uVar1 = 2;
    }
    return uVar1;
  }
  return 3;
}

```

_DAT_00003004 is a magic value all softdevices contain, and _DAT_00003008 is the size.

Yatekii, 2020-09-03:
Ok, interesting!

Sadly we have no control over the algorithm. We can write our own ofc, but we do not have the sources for the exported ones.

We had some interesting return codes with other algorithms too ...

Dirbaio, 2020-09-03:
I've checked that the "softdevice erase" algorithm is exactly the same except without that check. PR opened!

The return codes I've seen in the algo are
- 1: verify failed 
- 101: unaligned address (must be page multiple in erase_sector, word multiple in write)
- 102: address out of flash
- 103: address is in softdevice area
- 104: trying to write to a word that's not erased (not 0xFFFFFFFF)
- 105: invalid nvmc config mode or something

hope it's useful :) 

Also interestingly, the erase_all command skips softdevice sectors if there's a softdevice.

Yatekii, 2020-09-03:
Awesome, thank you!

I guess we could add a hashmap of all the available codes to the target definition file. It will be hard to get it for everything but it's better than nothing.
Technically the returncodes are not specified in the spec. The spec just knows 0/1.

Yatekii, 2020-09-03:
>     * 104: trying to write to a word that's not erased (not 0xFFFFFFFF)

Does this mean it checks every single word? Seems highly imperformant wow.

Dirbaio, 2020-09-03:
yup, it reads the word to check if it's 0xFFFFFFFF before writing to it

```c

undefined4 program_page(uint write_addr,uint write_len,int data)

{
  uint i;
  int iVar1;
  int unaff_r9;
  
  *(undefined4 *)(unaff_r9 + 4) = 3;
  *(undefined4 *)(unaff_r9 + 8) = 0;
  *(undefined4 *)(unaff_r9 + 0xc) = 0;
  *(undefined4 *)(unaff_r9 + 0x10) = 0;
  if ((write_addr & 3) != 0) {
    *(undefined4 *)(unaff_r9 + 0xc) = 3;
    *(uint *)(unaff_r9 + 0x10) = write_addr;
    return 0x65;
  }
  *(undefined4 *)(unaff_r9 + 8) = 1;
  i = get_flash_start();
  if ((write_addr < i) || (i = total_flash_size(), i <= write_addr)) {
    *(undefined4 *)(unaff_r9 + 0xc) = 4;
    *(uint *)(unaff_r9 + 0x10) = write_addr;
    return 0x66;
  }
  *(undefined4 *)(unaff_r9 + 8) = 3;
  i = total_flash_size();
  if (i < write_len + write_addr) {
    *(undefined4 *)(unaff_r9 + 0xc) = 4;
    *(uint *)(unaff_r9 + 0x10) = write_len + write_addr;
    return 0x66;
  }
  *(undefined4 *)(unaff_r9 + 8) = 4;
  iVar1 = ret0();
  if (iVar1 != 0) {
    *(undefined4 *)(unaff_r9 + 0xc) = 2;
    return 0x67;
  }
  *(undefined4 *)(unaff_r9 + 8) = 5;
  i = if_addr_in_softdevice_2_else_3(write_addr);
  if (i < 2) {
    *(undefined4 *)(unaff_r9 + 0xc) = 2;
    *(uint *)(unaff_r9 + 0x10) = write_addr;
    return 0x67;
  }
  if ((i == 2) && (iVar1 = do_verify(write_addr,write_len,0xff), iVar1 != 0)) {
    *(undefined4 *)(unaff_r9 + 4) = 3;
    *(undefined4 *)(unaff_r9 + 8) = 5;
    return 0x67;
  }
  *(undefined4 *)(unaff_r9 + 4) = 3;
  *(undefined4 *)(unaff_r9 + 8) = 6;
  if (write_len >> 2 == 0) {
    return 0;
  }
  i = 0;
  do {
    if (*(int *)(write_addr + i * 4) != -1) {
      *(undefined4 *)(unaff_r9 + 0xc) = 5;
      *(int *)(unaff_r9 + 0x10) = write_addr + i;
      return 0x68;
    }
    *(undefined4 *)(write_addr + i * 4) = *(undefined4 *)(data + i * 4);
    nvmc_wait_ready();
    i = i + 1;
  } while (i < write_len >> 2);
  return 0;
}
```

jacobrosenthal, 2020-09-03:
Is there any reason this would be nrf52 specific, and not also cover nrf51 as well? Could you check that @Dirbaio 

Its a refrain in the channel that some nordics are 'locked' and so the nrf-recover tool was built to do a mass erase.

But with this knowledge we now know theyre not locked, they have a softdevice on them and refuse to program with the current flash algorithms

hargoniX, 2020-09-03:
I'm already looking into it right now, the nrf51 target yml file also has an sde algorithm so I will quickly check whether using that as default fixes it, if it does I'll pr the fix in.

Dirbaio, 2020-09-03:
> Its a refrain in the channel that some nordics are 'locked' and so the nrf-recover tool was built to do a mass erase.

No, that's another lock mechanism: APPROTECT in the UICR (for nrf52, for nrf51 it's something else). 

Its purpose is so when you manufacture a device, bad guys can't read your firmware to manufacture clones. It disables the debug port entirely, except for the "recover" operation.

Recover erases the entire chip and removes APPROTECT. Nordic designed it like this so you don't brick the chip if you fuck up, while not allowing stealing firmware (that's why it erases everything before unlock).

The "don't overwrite the softdevice" seems to me more like a convenience noob-friendly "safety" feature, not a security feature. Either way it has nothing to do with APPROTECT / recover.

jacobrosenthal, 2020-09-03:
Yeah I don't think most chips are actually locked though. It's really rare
in open source. People thought they were because of the algorithm blocking
when sd present. It just so happens mass erase fixes both problems

Dirbaio, 2020-09-03:
in case anyone else is interested, [full decompile of nrf52 algo](https://gist.github.com/Dirbaio/bc1e40b98db79f371c9323aa3c5522a8)

closed 2020-09-03 by Yatekii

peter9477, 2022-07-22:
Note regarding the APPROTECT stuff: new revs of many of their chips (at least nRF52) now do have that protection enabled by default, and (as far as I can tell after just starting into this) actually re-protected automatically after every hard reset (or power cycle) unless the firmware actually turns it off again.  See [this PDF](https://infocenter.nordicsemi.com/pdf/in_141_v1.1.pdf) or google "nordic approtect rev 3 changes".