probe-rs/probe-rs #373: Trying to connect to a STM32F407VETx

coreychristian, 2020-09-12:
**Describe the bug**
Cannot probe the device using my ST-Link (clone).
I have tried connecting the ST-Link to my board https://os.mbed.com/users/hudakz/code/STM32F407VET6_Hello/shortlog/ 
probe-rs cannot seem to connect to the board. I have connected the board using the four wires, swdio, clk, ground and vcc, and have also tried grounding the boot 0 and 1 pins. I am fairly new to SWD, so am wondering if I am doing something wrong or if perhaps the issue lies with my hardware. I have been able to flash my ST-Link to the latest version available on the ST website, and can connect to my STM board using USART. Any help would be much appreciated.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect all four wires to the ST-Link and the Target
2. Try to open probe-rs-cli using RUST_LOG=probe_rs=debug probe-rs-cli info (also tried --connect-under-reset)
3. I notice that the first time I run the probe command, the board seems to be in DFU mode, it then  goes into MassStorageMode, and only returns to DFU if I unplug and plus it back into my laptop.

**Expected behavior**
No errors, info relating to the board being displayed in the console, this will hopefully allow me to progress to the next step where I can connect and flash RUST programs to my board.

**Stacktrace**
DEBUG probe_rs::probe::daplink::tools > Attempting to open 0483:3748 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 39, bNumInterfaces: 1, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 50, extra: None }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 0, bDeviceSubClass: 0, bDeviceProtocol: 0, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14152, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > STLink version: (2, 32)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 WARN  probe_rs::probe::stlink                > check_status failed: JtagGetIdcodeError
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
Error: An error with the usage of the probe occured

Caused by:
    0: An error specific to a probe type occured
    1: Command failed with status JtagGetIdcodeError

**Desktop **
 - Latest version of Linux Ubuntu

Tiwalun, 2020-09-22:
Hi @coreychristian :wave:, sorry for the late reply.

This seems like a problem with the reset, and you assumed correctly that `--connect-under-reset` would fix it. Unfortunately, the `info` subcommand doesn't support that flag (which is actually a bug).

Pleas try running `probe-rs-cli debug --connect-under-reset`. It should give you a very basic debug CLI if it works.

80avin, 2021-08-21:
Hi @Tiwalun 
This doesn't work.

I am connecting ST-Link v2 to EFR32BG22C224F512 chip via SWD and getting this error in every command.

Yatekii, 2021-08-21:
Hi @80avin

Unfortunately the `EFR32BG22C224F512` is on a whole other level of brokenness. This is actually a different issue (could you open one please?) because this one is with a STM32 which is well supported but requires this flag sometimes.
`EFR32BG22C224F512` is actually just broken :/

Yatekii, 2022-03-05:
I will close this, as it appears to be an issue with the EFR32 chips. If the **STM32** problem persists, please reopen.

closed 2022-03-05 by Yatekii