probe-rs/probe-rs #383: JLink probe unreliable under macOS

ryankurte, 2020-09-30:
**Describe the bug**
Both `probe-rs` cli or `cargo-flash` appear to be _super unreliable_ on macOS. In all cases `JLinkExe` can talk to the probe and target with no issue.

**To Reproduce**
1. Run `cargo-flash` or `probe-rs-cli` with any operation

Sometimes the operation will complete, usually a `DAP read` or other timeout will occur.

**Expected behavior**
The operation should complete.

**Stacktrace**

A failure from `RUST_LOG=probe_rs=debug cargo run --bin probe-rs-cli -- reset --chip ATSAMD21G18AU`
```
 DEBUG probe_rs::architecture::arm::communication_interface > Writing register CSW, value=0x63000012
 DEBUG probe_rs::probe::jlink                               > DAP line busy, retries remaining 4.
 DEBUG probe_rs::probe::jlink                               > DAP line busy, retries remaining 3.
 DEBUG probe_rs::probe::jlink                               > DAP line busy, retries remaining 2.
 DEBUG probe_rs::probe::jlink                               > DAP line busy, retries remaining 1.
 DEBUG probe_rs::probe::jlink                               > DAP line busy, retries remaining 0.
 ERROR probe_rs::probe::jlink                               > DAP read timeout.
 WARN  probe_rs::session                                    > Could not clear all hardware breakpoints: ArchitectureSpecific(RegisterWriteError { address: 0, name: "CSW", source: Timeout })
```

A failure from `probe-run --chip ATSAMD21G18AU`
```
       ERROR probe_rs::probe::jlink > DAP read timeout.
     Erasing sectors ✔ [00:00:00] [########################################################################################################################################] 128.00KB/128.00KB @ 148.93KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [----------------------------------------------------------------------------------------------------------------------------------------]       0B/117.00KB @       0B/s (eta 0s )
       Error failed to flash /Users/ryankurte/projects/scrollboi/target/thumbv6m-none-eabi/debug/scrollboi

Caused by:
    0: Error while flashing
    1: Something during the interaction with the core went wrong
    2: A core architecture specific error occured
    3: Failed to write register CSW at address 0x00000000 because: Operation timed out
    4: Operation timed out
```

**Desktop (please complete the following information):**
 - macOS Catalina

**Additional context**

- Running against `ATSAMD21G18AU` core on https://github.com/probe-rs/probe-rs/pull/382, though i think this is a _debugger_ issue rather than a _target_ one. yet to try another IC.
- Protocol issues perhaps related to #297?

ryankurte, 2020-10-11:
this also appears to be the case on linux with `probe-run`, flashing / connecting works maybe 1/4 of the time and otherwise results in an error along the lines of:

```
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
     Running `probe-run --chip ATSAMD21G18 --defmt target/thumbv6m-none-eabi/debug/scrollboi`
  (HOST) INFO  flashing program
  (HOST) INFO  success!
Error: A core architecture specific error occured

Caused by:
    0: Failed to write register CSW at address 0x00000000 because: Operation timed out
    1: Operation timed out
```

Yatekii, 2020-10-11:
Can you try and lower the speed to 100kHz please? Just as a test, I know this is not feasible for actual development :)

Yatekii, 2020-10-12:
`--speed 100` in `cargo flash` :) Not sure the CLI supports it.

ryankurte, 2020-10-12:
noo dice, somehow today i'm not getting a single file through (though `JLinkExe` is still happy). attached a log with slightly more information from `cargo-flash 0.9.0` (with probe.rs patched with #382).

[cargo-flash-log.txt](https://github.com/probe-rs/probe-rs/files/5367507/cargo-flash-log.txt)

benmkw, 2020-11-22:
I checked again after the last jlink changes (@Tiwalun) because I got timeouts before as well and now I get:

```
       DEBUG probe_rs::probe::jlink                 > Sucessfully switched to SWD
       DEBUG probe_rs::probe::jlink                 > Attached succesfully
```

but then
```
       Error failed attaching to target
             
             Caused by:
                 0: An error with the usage of the probe occured
                 1: An error specific to the selected architecture occured
                 2: Target device did not respond to request.
```

<details>
  <summary> TRACE </summary>

```
λ  (master↑1|●1✚ 6) ~/.cargo/my_target/debug/cargo-embed --target thumbv7m-none-eabi --release                                                                                                                                          181ms 
       DEBUG probe_rs_cli_util > Running 'cargo' in directory /Users/benediktmandelkow/Downloads/oxidize2020/my_boards/stm32bp_blink
    Finished release [optimized + debuginfo] target(s) in 0.03s
      Config default
      Target /Users/benediktmandelkow/.cargo/my_target/thumbv7m-none-eabi/release/stm32_blink
       DEBUG jaylink           > libusb 1.0.23.11397
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: false
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: false
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x105,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 3 interfaces
       TRACE jaylink           > active configuration descriptor: ConfigDescriptor {
    bLength: 0x9,
    bDescriptorType: 0x2,
    wTotalLength: 0x62,
    bNumInterfaces: 0x3,
    bConfigurationValue: 0x1,
    iConfiguration: 0x4,
    bmAttributes: 0x80,
    bMaxPower: 0x32,
    extra: Some(
        [
            0x8,
            0xb,
            0x0,
            0x2,
            0x2,
            0x2,
            0x0,
            0x5,
        ],
    ),
}
[... 227 lines cut ...]
       TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, f0, ff, ff, ff, ff, 1, 0]
       Error failed attaching to target
             
             Caused by:
                 0: An error with the usage of the probe occured
                 1: An error specific to the selected architecture occured
                 2: Target device did not respond to request.
        INFO cargo_embed                            > Metadata {
    chip: Some(
        "Unspecified(\"stm32f103\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "100",
    ),
    release: "0.10.0",
    commit: "8a4c101-modified",
}
```

</details>

(I'm not in a hurry about this, just posting in case the logs are helpful)

jkelleyrtp, 2020-12-11:
Same issues here...

Yatekii, 2020-12-11:
We do this already :) Not 1:1 ofc, but when we encounter issues we sometimes have a look at other codebases.

@Tiwalun did a lot of fixes on current master. So if we still have issues it might be a chip specific problem and not JLink maybe.

benmkw, 2020-12-12:
> a lot of fixes on current master

The logging changed a bit but the error is the same for me:

```
       DEBUG probe_rs::probe::jlink                 > Performing line reset!
       TRACE jaylink                                > write 30 bytes: [cf, 0, 61, 0, ff, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, 0, ff, ff, ff, ff, ff, ff, 53, a, 0, 0, 0, 0, 0]
       TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, f0, ff, ff, ff, ff, 1, 0]
       DEBUG probe_rs::probe::jlink                 > line reset ack: [true, true, true]
       TRACE jaylink                                > write 30 bytes: [cf, 0, 61, 0, ff, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, 0, ff, ff, ff, ff, ff, ff, 53, a, 0, 0, 0, 0, 0]
       TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, f0, ff, ff, ff, ff, 1, 0]
       DEBUG probe_rs::probe::jlink                 > line reset ack: [true, true, true]
        INFO cargo_embed                            > The target seems to be unable to be attached to.
        INFO cargo_embed                            > A hard reset during attaching might help. This will reset the entire chip.
        INFO cargo_embed                            > Set `general.connect_under_reset` in your cargo-embed configuration file to enable this feature.
       Error failed attaching to target
             
             Caused by:
                 0: An error with the usage of the probe occured
                 1: An error specific to the selected architecture occured
                 2: Target device did not respond to request.
        INFO cargo_embed                            > Metadata {
    chip: Some(
        "Unspecified(\"stm32f103\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "100",
    ),
    release: "0.10.1",
    commit: "e4d5b22-modified",
}
```

or with attach under reset 

```
       DEBUG probe_rs::probe::jlink                 > Performing line reset!
       TRACE jaylink                                > write 30 bytes: [cf, 0, 61, 0, ff, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, 0, ff, ff, ff, ff, ff, ff, 53, a, 0, 0, 0, 0, 0]
       TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, f0, ff, ff, ff, ff, 1, 0]
       DEBUG probe_rs::probe::jlink                 > line reset ack: [true, true, true]
       TRACE jaylink                                > write 30 bytes: [cf, 0, 61, 0, ff, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, 0, ff, ff, ff, ff, ff, ff, 53, a, 0, 0, 0, 0, 0]
       TRACE jaylink                                > read 14 bytes: [0, 0, 0, 0, 0, 0, 0, f0, ff, ff, ff, ff, 1, 0]
       DEBUG probe_rs::probe::jlink                 > line reset ack: [true, true, true]
       Error failed attaching to target
             
             Caused by:
                 0: An error with the usage of the probe occured
                 1: An error specific to the selected architecture occured
                 2: Target device did not respond to request.
        INFO cargo_embed                            > Metadata {
    chip: Some(
        "Unspecified(\"stm32f103\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "100",
    ),
    release: "0.10.1",
    commit: "e4d5b22-modified",
}
```

I checked on linux and its the same except I get the line reset ack two times and in between it does read 30 bytes, read 14 bytes.

Tiwalun, 2020-12-12:
Hard to tell if this a probe-rs problem, or if the connection is not possible for another reason.

Have you checked if it works with OpenOCD, or the JLink tools? This would rule out a setup problem.

benmkw, 2020-12-12:
I totally agree, unfortunately I only got back to doing a bit of embedded through rust so I’ve not really set anything else up but if I do I’ll be sure to report back.

LongLiveCHIEF, 2021-04-24:
I'm getting the same issues (`atsamd21g18a`), and I'm using a legit jlink edu mini.  I've also ensured it's not hardware related, as I've setup a platform.io project and and successfully flashed using swd.

LongLiveCHIEF, 2021-04-24:
To elaborate on my previous point, I even have to use openocd to restore the original bootloader after each flash attempt because the error described here make it impossible has the effect that the chip will no longer go into bootloader mode.

LongLiveCHIEF, 2021-04-24:
...and I'm on Linux, so it's not just macOS.

LongLiveCHIEF, 2021-04-24:
I think this is related to #507 .

ryankurte, 2021-04-24:
> Hard to tell if this is a probe-rs problem, or if the connection is not possible for another reason.

no issues using JLink tools directly, only via probe-rs for me, @LongLiveCHIEF have you tried using `JLinkExe` directly?  i guess another option would be to capture the USB interactions for both for comparison.

LongLiveCHIEF, 2021-04-25:
@ryankurte see my comments in #507.  I think this is possibly a problem with the actual CMSIS pack from ARM for the `atsamd21g18au` chip specifically.

LongLiveCHIEF, 2021-04-25:
> no issues using JLink tools directly, only via probe-rs for me, @LongLiveCHIEF have you tried using JLinkExe directly?

Not with a rust project, but i did write a quick blinky sample in C++ using platformio, and that worked. Platformio downloads the JLink tools, and executes using the `JLinkExe` official tool.  The example below worked perfectly with the same probe and chip, at the very least confirming that my probe setup with that particular probe was correct and there are no hardware/firmware problems with the probe itself.

```cpp
// src/main.cpp
#include <Arduino.h>

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);  
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(100);
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
}
```
and
```
[platformio]
default_envs = xiao

[env:xiao]
platform = atmelsam
framework = arduino
board = seeed_xiao
debug_tool = jlink          ; uses probe instead of uf2 flashing
upload_protocol = jlink   ; SWD interface (jlink-jtag for jtag)
```

Yatekii, 2022-03-05:
I changed the title of this, because it is not related to JLink most likely and surely not macOS.

Generally the ATSAM cores are/were always super flaky.

If you could give this another try @ryankurte & @LongLiveCHIEF that'd be nice :)

pustekuchen91, 2023-11-21:
I'm trying to debug an ATSAME53N20A with `probe-rs`

But i'm also getting errors:

`ERROR probe_rs::cmd::dap_server::server::startup: probe-rs-debugger session ended: An ARM specific error occurred.`

```
INFO: Triggered DAP Event: Event {
    body: Some(
        Object {
            "message": String("An ARM specific error occurred.\n\tError using access port\n\t\tFailed to write register CSW at address 0x00000000\n\t\t\tAn error occurred in the communication with an access port or debug port.\n\t\t\t\tTarget device responded with a FAULT response to the request.\n"),
            "severity": String("error"),
        },
    ),
    event: "probe-rs-show-message",
    seq: 405,
    type_: "event",
}
INFO: Triggered DAP Event: Event {
    body: Some(
        Object {
            "message": String("Debug Adapter terminated unexpectedly with an error: ProbeRs(Arm(AccessPort { address: ApAddress { dp: Default, ap: 0 }, source: RegisterWrite { address: 0, name: \"CSW\", source: Dap(FaultResponse) } }))\n"),
            "severity": String("error"),
        },
    ),
    event: "probe-rs-show-message",
    seq: 407,
    type_: "event",
}
INFO: Triggered DAP Event: Event {
    body: Some(
        Object {},
    ),
    event: "terminated",
    seq: 409,
    type_: "event",
}
INFO: Triggered DAP Event: Event {
    body: Some(
        Object {
            "exitCode": Number(1),
        },
    ),
    event: "exited",
    seq: 411,
    type_: "event",
}
An ARM specific error occurred.
	Error using access port
		Failed to write register CSW at address 0x00000000
			An error occurred in the communication with an access port or debug port.
				Target device responded with a FAULT response to the request.

Debug Adapter terminated unexpectedly with an error: ProbeRs(Arm(AccessPort { address: ApAddress { dp: Default, ap: 0 }, source: RegisterWrite { address: 0, name: "CSW", source: Dap(FaultResponse) } }))

```

I detected that in ATSAME53N20A.svd some errors with the H1CR Register were present, which i fixed. Could be an invalid SVD file an cause for this error, maybe there are more issues in the files.

Yatekii, 2023-11-22:
So this chip most likely uses a custom debug sequence that we did not properly implement yet. You can maybe check the CMSIS-Pack for that chip :)