probe-rs/probe-rs #391: ISP AVR support

pitaj, 2020-10-06:
**Is your feature request related to a problem? Please describe.**
I am trying to program an ATtiny84 with rust, but right now it looks like I'll have to define a custom runner using avrdude to flash the device.

**Describe the solution you'd like**
Probe.rs has full support for interfacing AVR MCUs to provide flashing and debugging capabilities.

**Describe alternatives you've considered**
Partial support for just flashing devices with AVRDUDE would be a great first step.

**Additional context**
Rust recently merged support for AVR. AVR microcontrollers are very popular, thanks to the Arduino project.

jwagen, 2020-12-05:
For supporting AVR it might be worth implementing support for EDBG for AVR. The specification for the protocol can be found here: [http://ww1.microchip.com/downloads/en/DeviceDoc/50002630A.pdf]( http://ww1.microchip.com/downloads/en/DeviceDoc/50002630A.pdf)
The AVR programming and debugging is implemented as an extension to CMSIS-DAP using vendor commands. It is should be possible to program and debug most AVRs using a supported EDBG debugger. This is also somewhat relevant to #308 I think.

There is also an implementation of the EDBG protocol written in python that can be used as a reference.
https://pypi.org/project/pymcuprog/

pitaj, 2020-12-05:
@jwagen EDBG would be useful but most hobbyists don't have an EDBG debugger as they're quite expensive.

jwagen, 2020-12-05:
Yes a the standalone debuggers such as Atmel ICE is a lot more expensive than the cheaper hobbyist programmers. There exist a couple of lower cost debuggers that should support EDBG that might be interesting.
The snap debugger supports AVR:
https://www.microchip.com/Developmenttools/ProductDetails/PG164100
And the the Curiosity Nano series of dev kits.
https://www.microchip.com/developmenttools/ProductDetails/DM320115

Of course support for the rely cheap AVRASP, Arduino and similar programmers would be better for low cost. And as you said these programmers are also more common among hobbyist. For debugging I think an EDBG compatible debugger is required though.

I guess EDBG support for AVR and AVRDUDE integration could be two separate issues.

closed 2020-12-09 by Tiwalun