probe-rs/probe-rs #392: Unbuntu dependency libusb-1.0 not required

ninjasource, 2020-10-08:
Hi,

I set up a fresh Linux Mint 20 pc today and followed the Ubuntu instructions:
```
sudo apt install -y libusb-dev libusb-1.0 libftdi1-dev
cargo install probe-run
```
Resulting in the following error:
`E: Unable to locate package libusb-1.0`

However, the following works and is sufficient for an install:
`sudo apt install -y libusb-dev libftdi1-dev`

So I guess you can remove that libusb-1.0 dependency.

Tiwalun, 2020-12-11:
Thanks for your report, and sorry for the late reply. 

We have updated the requirements, and now have the following instructions:

    sudo apt install -y libusb-1.0-0-dev libftdi1-dev

closed 2020-12-11 by Tiwalun

ninjasource, 2020-12-12:
Thanks for your great tool!