probe-rs/probe-rs #394: TPIU not found for STM32H753VITx

diondokter, 2020-10-18:
I'm trying to set up some ITM functionality for my STM32H753VIT6.
On the session I call setup_swv:

```rust
session.setup_swv(
            &SwoConfig::new(cpu_frequency)
                .set_mode_uart()
                .set_baud(interface_frequency),
        )?;
```

At some point this will call: https://github.com/probe-rs/probe-rs/blob/043b1b49bf83a14e321d05269e8db083d31fde2b/probe-rs/src/architecture/arm/memory/romtable.rs#L451 
This function will not find the TPIU for the H753, but does work for my F407.
The function searches for a component part with value `0x9A1`, defined here: https://github.com/probe-rs/probe-rs/blob/043b1b49bf83a14e321d05269e8db083d31fde2b/probe-rs/src/architecture/arm/memory/romtable.rs#L607.

I don't know where these values come from, so I can't check if it's the same for the H7 series.

These are the part codes found on the H7 (read from the log output): 
```
0x00C, 0x002, 0x00E, 0x001, 0x975, 0x906
```
On the F4:
```
0x00C, 0x002, 0x003, 0x001, 0x9a1, 0x925
```

**Desktop:**
 - Windows 10

**Additional context**
Probe: STLINKv3

Yatekii, 2020-10-18:
Did you setup the TPIU properly? If the debug facilities on the core are not activated the TPIU will not be found (yes, the ROM tables are not immutable as ROM might suggest ...). I am not sure how much is required for the H7. Might well be mich more init sequence things required than for the F4.

Also, I just tested this with an F407 and an nRF52840, so I would not claim for this to be 100% working on all targets.
The 0x9A1 code can be found at various places, I can't tell you where exactly I took it from. Even with ARM there is very cluttered docs.

I think all the CoreSight components are "documented" in pyOCD. But even tho those guys work for ARM it was determined a bit empirically too :D

diondokter, 2020-10-18:
So, what I've found now (a bit rambly):

The H7 docs say: `The TRACECK output must be enabled by setting the TRACECLKEN bit in the DBGMCU_CR register before trace is sent to the TPIU.`

That seems to be the only setup required.
___
The H7 has a different `DBGMCU` address.
Probe-rs defines the CR register as:
```rust
const DBGMCU: u32 = 0xE004_2004;
```
This is consistent with the F4: https://docs.rs/stm32f4/0.12.1/src/stm32f4/stm32f407/mod.rs.html#511
The H7 however, has a different location: https://docs.rs/stm32h7/0.12.1/src/stm32h7/stm32h743/mod.rs.html#3226
(I'm linking to the docs, because I can give actual links)

Looking up a random F7, it has the same address as the F4. So it's not a Cortex M4/7 thing.
___
The DBGMCU layout is different.
So not only is the location different, the bits are as well.

The F4 and F7 have the same layout.
___
Even if I change the probe-rs code to use the H7 location and H7 bits, the TPIU is still not found in the rom table.
It appears that the rom table is read before the DBGMCU register is set.

But even if I reread the rom table after DBGMCU register is set, it still doesn't show any new part.
___
Any idea what I could now look at? They don't make it easy to make stuff like this work, now do they? 😜

diondokter, 2020-10-18:
Wait a minute:
[image]
[image]

Why are they not present?

adamgreig, 2020-10-18:
I believe that's because they're instead described in the System ROM table 2; the TPIU is associated with the chip-wide debug subsystem rather than the Cortex-M7 core debug system. Or something like that.

[image]
[image]
[image]

diondokter, 2020-10-18:
Oh, that'd be annoying!

But what does `entry_present` even mean?
Looking at the RomTable::try_parse method, I can see that the value is false for some, but the address is valid:
[image]

adamgreig, 2020-10-18:
It means there's no TPIU accessible via the AHB-AP debug system or through the Cortex-M7 PPB, instead the TPIU must be accessed via the APB-AP and that's why it's described in the System ROM table 2 instead of the Cortex-M7 ROM table.

However I don't know how to turn this annoying knowledge into getting probe-rs to actually talk to the TPIU...

[image]

Yatekii, 2020-10-18:
>D6.2.2 ROM Table entries that are marked not present
The descriptions of the debug components are stored in sequential locations in the ROM Table, starting at the ROM
Table base address. However, a ROM Table entry can be marked not present by setting the PRESENT field of the
entry to a value that indicates that the entry is not present.
When scanning the ROM Table, an entry that is marked as not present must be skipped. Unless the entry has the
value 0x00000000, however, you must not assume that an entry that is marked not present represents the end of the
ROM Table. For example, a ROM Table might be generated using static configuration tie-offs that indicate the
presence or absence of particular devices, giving not present entries in the ROM Table.

From the reference manual ;)

adamgreig, 2020-10-18:
The values in Table 589 (the System ROM table 2) are the right ones for the TPIU, saying its address from a debugger's point of view is 0xE00F5000 and from the system bus (i.e. over AHB-AP, I guess) is 0x5C015000. The "TPIU" entry in the Cortex-M7 PPB ROM table which has an address of 0xE0041000 is not valid (since it's marked as not-present).

So I think in principle if you hardcoded probe-rs to access the TPIU at E00F5000, that should work, or if you wrote to it from your own firmware at 0x5C015000, that should also work. I'm not sure if probe-rs picks up the second set of ROM tables (on the APB-AP) or how to make it do so if it doesn't, though.

Yatekii, 2020-10-18:
Well, I think parsin the ROMtable is important such that wer actually check if there is a TPIU present. We could add an override tho.

I am not sure how we access the APB-AP but it should 100% be possible. It is just work :)

diondokter, 2020-10-18:
It may not even have to be hardcoded, though.

The address of the system rom table is given by the `ArmProbeInterface` of `StlinkArmDebug`.
The function `ap_information` when using access_port 2 gives back the correct address.

So what if not only the first ApInformation is used, but all?

Don't know if that would work with non-stlink probes, though.

diondokter, 2020-10-18:
Yes! That seems to work.
I forced the system to use the ap index 2 instead of 0.

I get back a part with the code 0x912.

According to blackmagic, that's a TPIU. It's also at the correct address.
https://github.com/blacksphere/blackmagic/blob/9cd3193a89ef075cc3f24963503a2118d5fa0e4b/src/target/adiv5.c#L210

Yatekii, 2020-10-18:
YAAAY! Nice :) Yes, so we only used the first AP (MemoryAP) for simplicity reasons in the past :) And especially in recent months I didn't have much time to get into things again :/ So this was def. the plan from the get go :) Great to hear it works :)

diondokter, 2020-10-18:
I'll try putting some time into this.

1. To use all Memory APs
2. To better integrate part codes (probably by looking at what BlackMagic does)
3. To support the H7's DBGMCU

That's not gonna be today, though. Monday evening is the first time I can spend with it again

Yatekii, 2020-10-18:
Sounds great, thanks!

No hurry, we all have other stuff to do unfortunately :/

diondokter, 2020-11-16:
Ok, so I have done some work on this and I quite like the results. But I'm running into another issue.
Sometimes the rom table can't be read properly.

```text
2020-11-16 20:45:55,723 INFO  [probe_rs::architecture::arm::memory::romtable] Reading rom table entry at e00e0018
2020-11-16 20:45:55,723 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e0018, length=1
2020-11-16 20:45:55,735 INFO  [probe_rs::architecture::arm::memory::romtable] Entry consists of all zeroes, stopping.
2020-11-16 20:45:55,735 INFO  [probe_rs::architecture::arm::memory::romtable] Parsing entry at e00e1000
2020-11-16 20:45:55,735 INFO  [probe_rs::architecture::arm::memory::romtable] Parsing entry at e00e2000
2020-11-16 20:45:55,736 INFO  [probe_rs::architecture::arm::memory::romtable] Parsing entry at e00e3000
2020-11-16 20:45:55,736 INFO  [probe_rs::architecture::arm::memory::romtable]  Reading component data at: e00e3000
2020-11-16 20:45:55,736 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e3ff0, length=4
2020-11-16 20:45:55,737 DEBUG [probe_rs::architecture::arm::memory::romtable] CIDR: [0, 0, 0, 0]
2020-11-16 20:45:55,737 WARN  [probe_rs::architecture::arm::memory::romtable] Component at 0xe00e3000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
2020-11-16 20:45:55,738 WARN  [probe_rs::architecture::arm::memory::romtable] Component at 0xe00e3000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
2020-11-16 20:45:55,738 WARN  [probe_rs::architecture::arm::memory::romtable] Component at 0xe00e3000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
2020-11-16 20:45:55,739 DEBUG [probe_rs::architecture::arm::memory::romtable] Reading debug id from address: e00e3fd0
2020-11-16 20:45:55,739 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e3fd0, length=4
2020-11-16 20:45:55,739 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e3fe0, length=4
2020-11-16 20:45:55,739 DEBUG [probe_rs::architecture::arm::memory::romtable] Raw peripheral id: [0, 0, 0, 0, 0, 0, 0, 0]
2020-11-16 20:45:55,741 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e3fcc, length=1
2020-11-16 20:45:55,741 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e3fbc, length=1
2020-11-16 20:45:55,742 DEBUG [probe_rs::architecture::arm::memory::romtable] Dev type: 0, arch id: 0
2020-11-16 20:45:55,742 INFO  [probe_rs::architecture::arm::memory::romtable]  Component class: GenericVerificationComponent
2020-11-16 20:45:55,743 INFO  [probe_rs::architecture::arm::memory::romtable]  Component peripheral id: PeripheralID { REVAND: 0, CMOD: No, REVISION: 0, JEP106: None, PART: 0, SIZE: 1, dev_type: 0, arch_id: 0 }
2020-11-16 20:45:55,754 INFO  [probe_rs::architecture::arm::memory::romtable] Parsing entry at e00e4000
2020-11-16 20:45:55,754 INFO  [probe_rs::architecture::arm::memory::romtable]  Reading component data at: e00e4000
2020-11-16 20:45:55,754 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e4ff0, length=4
2020-11-16 20:45:55,755 DEBUG [probe_rs::architecture::arm::memory::romtable] CIDR: [0, 0, 0, 0]
2020-11-16 20:45:55,755 WARN  [probe_rs::architecture::arm::memory::romtable] Component at 0xe00e4000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
2020-11-16 20:45:55,757 WARN  [probe_rs::architecture::arm::memory::romtable] Component at 0xe00e4000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
2020-11-16 20:45:55,757 WARN  [probe_rs::architecture::arm::memory::romtable] Component at 0xe00e4000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
2020-11-16 20:45:55,757 DEBUG [probe_rs::architecture::arm::memory::romtable] Reading debug id from address: e00e4fd0
2020-11-16 20:45:55,757 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e4fd0, length=4
2020-11-16 20:45:55,758 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e4fe0, length=4
2020-11-16 20:45:55,760 DEBUG [probe_rs::architecture::arm::memory::romtable] Raw peripheral id: [0, 0, 0, 0, 0, 0, 0, 0]
2020-11-16 20:45:55,773 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e4fcc, length=1
2020-11-16 20:45:55,774 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e4fbc, length=1
2020-11-16 20:45:55,774 DEBUG [probe_rs::architecture::arm::memory::romtable] Dev type: 0, arch id: 0
2020-11-16 20:45:55,776 INFO  [probe_rs::architecture::arm::memory::romtable]  Component class: GenericVerificationComponent
2020-11-16 20:45:55,776 INFO  [probe_rs::architecture::arm::memory::romtable]  Component peripheral id: PeripheralID { REVAND: 0, CMOD: No, REVISION: 0, JEP106: None, PART: 0, SIZE: 1, dev_type: 0, arch_id: 0 }
2020-11-16 20:45:55,776 INFO  [probe_rs::architecture::arm::memory::romtable] Parsing entry at e00e5000
2020-11-16 20:45:55,776 INFO  [probe_rs::architecture::arm::memory::romtable]  Reading component data at: e00e5000
2020-11-16 20:45:55,792 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e5ff0, length=4
2020-11-16 20:45:55,792 DEBUG [probe_rs::architecture::arm::memory::romtable] CIDR: [d, f0, 5, b1]
2020-11-16 20:45:55,792 DEBUG [probe_rs::architecture::arm::memory::romtable] Reading debug id from address: e00e5fd0
2020-11-16 20:45:55,792 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e5fd0, length=4
2020-11-16 20:45:55,793 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e5fe0, length=4
2020-11-16 20:45:55,809 DEBUG [probe_rs::architecture::arm::memory::romtable] Raw peripheral id: [1, b1, 1b, 0, 4, 0, 0, 0]
[... cut ...]
2020-11-16 20:45:55,810 DEBUG [probe_rs::probe::stlink] Read mem 32 bit, address=e00e5fbc, length=1
2020-11-16 20:45:55,811 DEBUG [probe_rs::architecture::arm::memory::romtable] Dev type: 0, arch id: 0
```

Some CIDR's are read as all 0. It usually only breaks or starts working again when it is reflashed, but there's no guarantee and I can't find the pattern...

I've made #431 to show the progress so far. When it works, it works. But I don't know if this bug is my new code, in existing code or a hardware bug (either STM's or mine).

Yatekii, 2020-11-16:
The PR looks nice already! Thanks for the hard work :)

I have seen CIDRs read all 0 already. So I think there is a bug in the Romtable parsing I once fixed and ill-merged somehow, that's in again. I cannot remember what it was tho. I will have another look some day. There is an open issue about it (#320).

closed 2021-01-18 by diondokter

ryan-summers, 2022-05-20:
For anyone that comes here after the fact, for the STM32H7xx components, you still need to explicitly enable the debug component clocks in the DBGMCU to get them to show up in the ROM table. Otherwise, even with the fixes to search all ROM tables, the entries will be zero and the TPIU may not be found

Specifically, enable the `D3DBGCKEN` and `D1DBGCKEN` bits in the DBG MCU to get the TPIU to show up.

The TPIU is part of the D1 domain and SWO etc. are part of D3.

Yatekii, 2022-05-20:
I wonder if it would be feasibly doable to enable DBGMCU when we can for romtables or if that is bad. because we read some components when we connect. we could disable it after reading them.

ryan-summers, 2022-05-20:
The reason we don't is because the DBG-MCU is a vendor-specified component, and indeed changes between variants (e.g. F4/F7 have different DBGMCU than H7). We'd have to encode information about DBGMCU layouts into probe-rs, which I believe there's aversion to.

noppej, 2022-05-20:
@Yatekii ... why could we not use a custom sequence for this?

Yatekii, 2022-05-20:
yup, sequences are the way to go here :) there might even be a fitting one already :)