probe-rs/probe-rs #395: Failed attaching to target (ATSAMD51N20A)

bradleyharden, 2020-10-18:
**Describe the bug**
I first mentioned this issue in #308. Now that #382 has landed, I think I can upgrade it from a PR comment to an actual issue.

With the original version of #308, I was not able to flash my device, but I was at least able to get cargo-embed working well enough to see RTT output. However, if I build everything from `master` now, I can't even connect to the target. I get [this](https://gist.github.com/bradleyharden/047e35ef1f3118089e5dfadc74a02924) instead. I get the same error with `cargo flash` and with `cargo embed` configured **not** to flash.

**To Reproduce**
1. Build cargo-embed (6d2768b) & cargo-flash (5460330) with probe-rs (0dc90f2b)
2. Use a WeAct CMSIS-DAP probe with the latest firmware
3. Connect to a custom board with an ATSAMD51N20A
4. Run `cargo flash --log debug --connect-under-reset`
5. Observe the error

**Expected behavior**
Chip is flashed, or RTT output is observed

**Stacktrace**
https://gist.github.com/bradleyharden/047e35ef1f3118089e5dfadc74a02924

**Desktop (please complete the following information):**
Ubuntu 18.04

bradleyharden, 2020-10-19:
I also have a J-Link probe. I tried it again, with the same install of cargo-flash and cargo-embed. The errors were different but the result was the same.

https://gist.github.com/bradleyharden/7fb757a813df9fc3585dc1f714486799

jacobrosenthal, 2020-11-05:
I took a few shots at getting this going. 
probe-rs is eating the error here and that read is crashing the probe https://github.com/probe-rs/probe-rs/blob/dd320571a47fccde0330a6b973efe414fe0c1918/probe-rs/src/architecture/arm/ap/mod.rs#L148

So its overall failing without even getting into the loop here https://github.com/probe-rs/probe-rs/blob/8262576839a8f4dd6c01350205a9229f82caf353/probe-rs/src/architecture/arm/communication_interface.rs#L247

So basically just reading ap 2 pisses off the device, I reset after each ap in the loop, and the only 2 that seems to cooperate are 0 and 1, so Im just limiting there
https://github.com/jacobrosenthal/probe-rs/commit/c4ed971fd6495c14c0f0bee2e56ceae087284fb0

Current jlink
https://gist.github.com/jacobrosenthal/b2095345bb6e003b1554ab3849ffe218
current cmsis
https://gist.github.com/jacobrosenthal/a75bcf1f80832341f1b77a6946a691b6

Heres a .sal logic decode of the cmsis. My jtag is going to be harder to break out to sniff
[cmsis atsamd.zip](https://github.com/probe-rs/probe-rs/files/5495456/cmsis.atsamd.zip)

jacobrosenthal, 2020-11-05:
Jlink with --speed 1 gets a bit further
https://gist.github.com/jacobrosenthal/122f7e2f00c4449e6bff778d99b06367
`1: The execution of 'erase_sector' failed with code 1`

Deleted the old capture file. Heres a better one
Heres a .sal logic decode of the cmsis. My jtag is going to be harder to break out to sniff
[cmsis atsamd.zip](https://github.com/probe-rs/probe-rs/files/5495591/cmsis.atsamd.zip)

jacobrosenthal, 2020-11-06:
Tried chip erasing to make sure something bootloader or locking wasnt screwing up, but doesnt help either probe.

Yatekii saw the fault in the cmsis log and recommended printing CTRL/STAT 
https://github.com/jacobrosenthal/probe-rs/commit/4733fa935329545d684a95268ed5f6d6d118106f
https://gist.github.com/jacobrosenthal/63301f7c4d27118eb412c2182ecac996
which finds that sticky_err IS being set.. but we dont know what to do with that knowledge

"What causes a STICKYERR in a CoreSight Debug Access Port?" https://developer.arm.com/documentation/ka001433/1-0/?search=5eec6f69e24a5e02d07b2655 hasnt given me any ideas

jacobrosenthal, 2020-11-07:
I find understanding the logs with batching to be troublesome, so I set max_writes = 1 for the cmsis and it looks like theyre failing at the exact same place, which is good to know. 

I dug up a jlink I could get probes on and the result is
[jlink atsamd.zip](https://github.com/probe-rs/probe-rs/files/5505362/jlink.atsamd.zip)
[image]

The decoder eats it on the last packet, ill have to manually decode that one.

Yatekii, 2020-11-07:
Thanks a lot for all the digging, I appreciate it very much =)

Hmm ok, but the batching should execute the commands in the exact same order as if there was no batching :) But yeah, it's an additional layer of abstraction ...

That is a common bug on the Logic SWD decoder afaik :(

bradleyharden, 2020-11-07:
@jacobrosenthal, thanks for all your hard work. I wish I had time to dig into it myself, but that's not going to happen this weekend. At this point, it will also take me time to even catch up to you. Good luck with it.

jacobrosenthal, 2020-11-08:
tldr
if you disable bootprotect and keep speeds down, both cmsis and jlink can program sam parts. 
If you want to keep bootprotect on, we need to teach probe-rs erases to be load address aware

---------------
So last we left our hero, erase was failing with code 1
Turns out it WAS bootprotect screwing us up.

Also turns out theres like a decade of bugs and patches and openocd version and hacks and memory accesses to try to disable it

havent figured out the magic incantations to do it within jlinkexe yet but thats left as an exercise for the reader

Openocd with my cmsis probe cant seem to do it either
```
openocd -f interface/cmsis-dap.cfg -c "cmsis_dap_vid_pid 0xc251 0xf001" -f target/atsame5x.cfg -c "init; halt; atsame5 bootloader 0; reset; shutdown"
Info : Listening on port 3333 for gdb connections
Error: SAM: NVM lock error
Error: SAM: NVM programming error
Error: Failed to erase block containing 00804000
Error: Couldn't erase user row
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory at 0x00804004
```

however with jlink and openocd i was sucessful checking or setting the bootloader. Im using openocd from head, but I think this has been in trunk for a while..hopefully. Ideally we find a probe-rs solution instead of relying on openocd

`openocd -f interface/jlink.cfg -c "transport select swd;" -f target/atsame5x.cfg -c "init; halt; atsame5 bootloader 0; reset; shutdown"`

```
jacob@j:~/Downloads/atsamd/boards/edgebadge (qspi-badge)$ cargo flash --example ferris_img --release --speed 200
    Finished release [optimized + debuginfo] target(s) in 0.08s
    Flashing /home/jacob/Downloads/atsamd/boards/edgebadge/target/thumbv7em-none-eabihf/release/examples/ferris_img
     Erasing sectors ✔ [00:00:00] [####################################################################################################################]  24.00KB/ 24.00KB @  27.44KB/s (eta 0s )
 Programming pages   ✔ [00:00:08] [####################################################################################################################]  21.50KB/ 21.50KB @   2.23KB/s (eta 0s )
jacob@j:~/Downloads/atsamd/boards/edgebadge (qspi-badge)$ 
```

cmsis does work after boot prot is unlocked
```
jacob@j:~/Downloads/atsamd/boards/edgebadge (qspi-badge)$ cargo flash --example ferris_img --release --speed 100
    Finished release [optimized + debuginfo] target(s) in 0.06s
    Flashing /home/jacob/Downloads/atsamd/boards/edgebadge/target/thumbv7em-none-eabihf/release/examples/ferris_img
     Erasing sectors ✔ [00:00:00] [####################################################################################################################]  24.00KB/ 24.00KB @  36.20KB/s (eta 0s )
 Programming pages   ✔ [00:00:06] [####################################################################################################################]  21.50KB/ 21.50KB @   2.90KB/s (eta 0s )
jacob@j:~/Downloads/atsamd/boards/edgebadge (qspi-badge)$ 
```
Those are the fastest speeds that work.

This seems to be a long standing issue. My guess is wait handling is poor?

ok so what about WITH boot protection but avoiding those sectors
```
jacob@j:~/Downloads/atsamd/boards/pygamer (qspi-badge)$ cargo flash --example ferris_img --release --speed 100
    Finished release [optimized] target(s) in 0.09s
    Flashing /home/jacob/Downloads/atsamd/boards/pygamer/target/thumbv7em-none-eabihf/release/examples/ferris_img
     Erasing sectors ✔ [00:00:00] [--------------------------------------------------------------------------------------------------------------------]       0B/ 32.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [--------------------------------------------------------------------------------------------------------------------]       0B/ 22.00KB @       0B/s (eta 0s )
       Error failed to flash /home/jacob/Downloads/atsamd/boards/pygamer/target/thumbv7em-none-eabihf/release/examples/ferris_img

Caused by:
    0: Error while flashing
    1: The execution of 'erase_sector' failed with code 1
jacob@j:~/Downloads/atsamd/boards/pygamer (qspi-badge)$ 
```
Logs show
```
        INFO probe_rs::flashing::flasher                                  > Erasing sector at address 0x00000000
```

so its attempting to erase protected bits. 
[... 25 lines cut ...]
Flash write discontinued at 0x00004134, next section at 0x00008000
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
atsame5.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x80000003 pc: 00000000 msp: 0x464c4558
Polling target atsame5.cpu failed, trying to reexamine
atsame5.cpu: hardware has 6 breakpoints, 4 watchpoints
> 
```

However providing a bin and address works
```
> program /home/jacob/Downloads/atsamd/boards/pygamer/ferris_img.bin 16384 verify reset
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000574 msp: 0x20012ab8
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
> 
```

jacobrosenthal, 2020-11-08:
And fwiw Ive confirmed this works with cargo flash, cargo embed as well as probe run (with outstanding speed patch `cargo install --git https://github.com/jacobrosenthal/probe-run --branch samd`)

Speed is rather miserable but ah well

jacobrosenthal, 2020-11-09:
More on sector erase
memory.x
```
MEMORY
{
  /* Leave 16k for the default bootloader on the PyGamer */
  FLASH (rx) : ORIGIN = 0x00000000 + 16K, LENGTH = 512K - 16K
  RAM (xrw)  : ORIGIN = 0x20000000, LENGTH = 192K
}
_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```
nm sorted we can see it shouldnt be interacting under 0x4000
```
jacob@j:~/Downloads/atsamd/boards/pygamer (qspi-badge)$ cargo nm --example ferris_img --release -- --numeric-sort
    Finished release [optimized] target(s) in 0.07s
00004004 R __RESET_VECTOR
00004008 R __EXCEPTIONS
00004008 R __reset_vector
00004040 R __INTERRUPTS
00004040 R __eexceptions
00004260 T Reset
00004260 R _stext
0000430a t <() as core::fmt::Debug>::fmt::h5cfe4bdc114d19fb
0000431a t core::ptr::drop_in_place::h28d7670aebb06621
0000431c T main
00004326 t ferris_img::__cortex_m_rt_main::h3e88d509e0e5907f
```

Relevant snip where we can see probe-rs erasing 0x00000000
```
jacob@j:~/Downloads/atsamd/boards/pygamer (qspi-badge)$ RUST_LOG=probe_rs::flashing cargo flash --example ferris_img --release --speed 100 
    Finished release [optimized] target(s) in 0.08s
    Flashing /home/jacob/Downloads/atsamd/boards/pygamer/target/thumbv7em-none-eabihf/release/examples/ferris_img
...
       TRACE probe_rs::flashing::builder  > Added Sector (0x00000000..0x00002000)
       TRACE probe_rs::flashing::builder  > Added Page (0x00000000..0x00000200)
       TRACE probe_rs::flashing::builder  > Added Sector (0x00004000..0x00006000)
...
       DEBUG probe_rs::flashing::flasher  > Waiting for routine call completion.
       WARN probe_rs::flashing::flasher  > sector_erase 0x00000000
       INFO probe_rs::flashing::flasher  > Erasing sector at address 0x00000000
       DEBUG probe_rs::flashing::flasher  > Calling routine 20000249(Some(0), None, None, None, init=false)
...
       INFO probe_rs::flashing::flasher  > Done erasing sector. Result is 1. This took 76.838258ms
     Erasing sectors ✔ [00:00:00] [--------------------------------------------------------------------------------------------------------------------]       0B/ 32.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [--------------------------------------------------------------------------------------------------------------------]       0B/ 22.00KB @       0B/s (eta 0s )
       Error failed to flash /home/jacob/Downloads/atsamd/boards/pygamer/target/thumbv7em-none-eabihf/release/examples/ferris_img

Caused by:
    0: Error while flashing
    1: The execution of 'erase_sector' failed with code 1
```

jacobrosenthal, 2020-11-09:
[ferris_img.zip](https://github.com/probe-rs/probe-rs/files/5513684/ferris_img.zip)

jacobrosenthal, 2020-11-11:
Ian on matrix reported stlinkv2/1 worked out of the box at full speed

I have an ST-LINK/V2 here
Bus 003 Device 024: ID 0483:3748 STMicroelectronics ST-LINK/V2

Failed with
DEBUG probe_rs::probe::stlink                > STLink version: (2, 29)
https://gist.github.com/jacobrosenthal/83849fc69b74de57608070722329666c

Upgraded firmware failure looks very similar
DEBUG probe_rs::probe::stlink                > STLink version: (2, 36)
https://gist.github.com/jacobrosenthal/7fa68cf6909242ff1b36d18eee7d749f

Though these devices/the code path seems to handle the ap search fault
```
   DEBUG probe_rs::probe::stlink                > Changing AP to 2, AP_BANK_SEL to 15
   DEBUG probe_rs::probe::stlink                > Writing DP register SELECT, value=0x020000f0
   TRACE probe_rs::probe::stlink::usb_interface > Sending command [f2, 46, ff, ff, 8, 0, f0, 0, 0, 2] to STLink, timeout: 1s
   TRACE probe_rs::probe::stlink::usb_interface > Sending command [f2, 45, ff, ff, c, 0] to STLink, timeout: 1s
   DEBUG probe_rs::probe::stlink                > Opening AP 2
   TRACE probe_rs::probe::stlink                > JTAG_INIT_AP 2
   TRACE probe_rs::probe::stlink::usb_interface > Sending command [f2, 4b, 2] to STLink, timeout: 1s
    WARN probe_rs::probe::stlink                > send_jtag_command 242 failed: SwdApFault
   DEBUG probe_rs::probe::stlink                > Reading register IDR
   DEBUG probe_rs::probe::stlink                > Changing AP to 0, AP_BANK_SEL to 15
```

ianrrees, 2020-11-11:
[capture of probe-rs-cli info.zip](https://github.com/probe-rs/probe-rs/files/5520970/capture.of.probe-rs-cli.info.zip)

This is a Saleae capture file for their Logic 2 Alpha software, of running `probe-rs-cli info` with a debug probe-rs built from 0f8120747bca519968ac42057b212de7f49ade09 .  Using an ST-Link v2.1, attached to an ATSAMD21G18AU.  [Here](https://github.com/probe-rs/probe-rs/files/5520984/probe-rs-cli.info.output.txt) is output from a different invocation with the same setup.

jacobrosenthal, 2020-11-16:
#419 has been merged 

The nitty gritty
In all cases, bootloader bits can be annoying
* If you have a bootloader you need to be avoiding that area (obviously). Though note you probably also need to be passing nmagic to to linker https://github.com/probe-rs/probe-rs/pull/426)
* Or you need to make sure your boot protection bits are cleared (sadly probably with openocd and a different incantation based on your chip)

Stlink v2.1 - We have reports that this has worked all along at full speed! Please report back your success
Stlink v2 - I havent seen it work and probably wont persue. Please let us know what you find.
Cmsis and jlink probes -- with speed set to 100 (sadly)

That fix will come out in probe-rs 0.10 and slowly filter out to cargo flash/embed and probe-run ecosystems

jacobrosenthal, 2020-11-29:
With the recent #462 merge, upstream (but not yet released) has

Stlink v2/v2.1 - fully working
jlink     - fully working
cmsis     - fully working
edbg      - are untested/wip pr at https://github.com/probe-rs/probe-rs/pull/308

bradleyharden, 2020-11-29:
Shall I close this then?

bradleyharden, 2020-11-29:
On Matrix, @jacobrosenthal says yes. Thanks @jacobrosenthal!

closed 2020-11-29 by bradleyharden