probe-rs/probe-rs #398: Having hard time flashing simple code

CGenie, 2020-10-22:
**Describe the bug**

I can't get `cargo flash` to work with even simple code files.

**To Reproduce**

I'm using STM32F411RE nucleo board. I cloned https://cgit.pinealservo.com/BluePill_Rust/blue_pill_base and adjusted target to `"thumbv7em-none-eabi"`. I also updated `memory.x` to:

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 512K
  RAM : ORIGIN = 0x20000000, LENGTH = 128K
}
```

However, `cargo flash --chip stm32f411retx --release` results in:

```
Finished release [optimized + debuginfo] target(s) in 0.01s
    Flashing /home/przemek/git-work/github/STM32/utah-rust-blue-pill-code/target/thumbv7em-none-eabi/release/blue_pill_base
     Erasing sectors ⠁ [00:00:00] [##################################################################################################################################################################]       0B/      0B @       0B/s (eta 0s )
       Error failed to flash /home/przemek/git-work/github/STM32/utah-rust-blue-pill-code/target/thumbv7em-none-eabi/release/blue_pill_base

Caused by:
    0: Error while flashing
    1: No flash memory contains the entire requested memory range 0x010000..0x010094.
```

**Expected behavior**

Expected flash to work without errors.

**Desktop (please complete the following information):**

ArchLinux.

```
rustc --version
rustc 1.49.0-nightly (1eaadebb3 2020-10-21)
```

adamgreig, 2020-10-22:
When you changed the target from to thumbv7**e**m-none-eabi, did you also update the first line of the [.cargo/config](https://cgit.pinealservo.com/BluePill_Rust/blue_pill_base/src/branch/master/.cargo/config) file from that repository? You have to set the linker script using a rustflag as per that file, and the one you've cloned only sets it for the original target.

Without that linker script, your executable file won't have the code in the right location, which could lead to the sort of error you're getting (which doesn't at first appear to be a probe-rs error, but rather a problem with your executable file).

Yatekii, 2020-10-22:
Normally, this error means that your linker script is not properly pulled in.

For me this mostly means I forgot the "rt" feature somewhere or didn't pull in the HAL crate in my main :)

The error is not very concise so we can definitely double down on this here :)

CGenie, 2020-10-23:
Ah, the `.cargo/config` file editing worked, linker script fix was what I needed! Thank you!

closed 2020-10-23 by CGenie