probe-rs/probe-rs #399: Consider all possible probe devices even if one probe device type fails during the scan.

Yatekii, 2020-10-22:
Currently, when scanning for debug probes, we exceute 3 different functions. One for J-Links, one for ST-Links and one for CMSIS-DAP devices. Let's assume we have this exact order. If the function to scan for J-Link devices fails due to any reason, we will never even scan for ST-Links or CMSIS-DAP devices.
Users of an ST-Link are not even interested in J-Links, so it makes no sense to return an error if J-Link enumeration fails.

I am not sure what the best solution is for this, but it certainly isn't just aborting the scan :)

adamgreig, 2020-10-25:
Is this actually _only_ a JTAG issue? In `Probe::list_all` we scan for DAPlink, FTDI, and STlink devices by calling their respective `list_x_devices()` methods, but those always return a Vec, not a Result (internally they product empty lists on errors).

However `list_jlink_devices()` returns a Result which we call `expect("Failed to list J_Link devices.")` on [here](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/mod.rs#L185).

Perhaps we just need to change `list_jlink_devices()` to always return a (possibly empty) Vec to match the other drivers?

Yatekii, 2020-10-25:
Ohh, I never considered that! Seems like it, good spot!

commit ed935361 (2020-11-26): Merge #401

401: Return plain Vec<DebugProbeInfo> from list_jlink_devices() r=Yatekii a=adamgreig

Change to return `Vec<DebugProbeInfo>` instead of `Result<impl Iterator>` to match the other `list_x_devices()` methods and prevent a jlink error from stopping other drivers operating. The original patch was only a two-line change but sadly rustfmt had other ideas!

Closes #399.

I don't have any j-link probes, so this is untested beyond checking it builds.
PR #401 (2020-11-26): Return plain Vec<DebugProbeInfo> from list_jlink_devices()
Change to return `Vec<DebugProbeInfo>` instead of `Result<impl Iterator>` to match the other `list_x_devices()` methods and prevent a jlink error from stopping other drivers operating. The original patch was only a two-line change but sadly rustfmt had other ideas!

Closes #399.

I don't have any j-link probes, so this is untested beyond checking it builds.
closed 2020-11-26 by bors[bot]