probe-rs/probe-rs #404: nRF91 gives me ERROR probe_rs::probe::jlink > DAP read timeout.

macthestack, 2020-10-26:
When I'm running ```cargo embed --target=thumbv8m.main-none-eabihf``` I get errors:

````
       DEBUG jaylink           > libusb 1.0.23.11397
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: false
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: false
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 1 interfaces
       DEBUG jaylink                                > J-Link interface is #0
       DEBUG jaylink                                > legacy caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG jaylink                                > extended caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG probe_rs::probe::jlink                 > Supported speeds: Speeds { base_freq: 96000000, min_div: 24 }
       DEBUG probe_rs::probe::jlink                 > Divider: 24
        INFO cargo_embed                            > Protocol speed 4000 kHz
       DEBUG probe_rs::probe::jlink                 > Attaching to J-Link
       DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
        INFO probe_rs::probe::jlink                 > J-Link: S/N: 801026105
       DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link EDU Mini V1 compiled Jul 17 2020 16:25:21
        INFO probe_rs::probe::jlink                 > J-Link: Hardware version: JLink 1.0.0
        INFO probe_rs::probe::jlink                 > J-Link: Target voltage: 3.29 V
       DEBUG probe_rs::probe::jlink                 > Sucessfully switched to SWD
       DEBUG probe_rs::probe::jlink                 > Attached succesfully
       DEBUG probe_rs::config::registry             > Searching registry for chip with name nRF9160_xxAA
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 4.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 3.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 2.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 1.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 0.
       ERROR probe_rs::probe::jlink                 > DAP read timeout.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 4.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 3.
       DEBUG probe_rs::probe::jlink                 > DAP line busy, retries remaining 2.
[... 127 lines cut ...]
    name: nrf91xx_uicr
    description: nRF91xxx UICR Erase
    default: true
    instructions: cLVA8gQGwPIABgAlCesGAEn4BlDA6QFVxWDQshRGAPDj+QAoHr8J6wYBwekCBAVGKEZwvRC1BEYAIADw1flA8gQBwPIAAUlEELHB6QIEA+CJaAApCL8QvUTyAAHC8gABASIKYEDyBALA8gACWfgCMEpES2BTaItgk2jLYNJoCmEQvQC/gLVA8gQAwPIAAAEhSfgAEEhEACHA6QERwWAA8MX5ACCAvQC/ELVA8gQAwPIAAAIhSfgAEEhEACTA6QFExGAA8FP5AUZI8gAAwPL/AP8iAPAL+DCxAPC4+QRGaSgUvwAsACQgRhC9AL9wtUDyBAzA8gAMBSNJ+AwwCesMA0/wAA6EB8PpAe7D+AzgBdAJ6wwBAyLB6QIgDOAJ6wwDT/ABDowHw/gE4AjQCesMAAMiwOkCIU/wZQ5wRnC9SPb/fsDy/w4J6wwDAiRwRVxgBdkJ6wwBBCLB6QIgDeALGA7xAQUJ6wwEAyarQmZgCNkJ6wwABCHA6QITT/BmDnBGcL1RsQAjT/AADgC/xlyWQgfRATOLQvnTC+BP8AAOcEZwvRhECesMAQUiT/ABDsHpAiBwRnC9AL8t6fBBQPIEBsDyAAYVRgRGAyAJ6wYCACOPB0n4BgDC6QEz02AG0AnrBgLC6QIBZSC96PCBSPb/cMDy/wAJ6wYCAiOEQlNgB9kJ6wYABCHA6QIUZiC96PCBChkBMAnrBgMDJ4JCX2AH2QnrBgAEIcDpAhJmIL3o8IEAILDrkQ8b0E/qkQgAJwzgVfgnAET4JwAA8Oj4ATdHRU/wAAAov73o8IFU+CcAATDu0AnrBgAFIeIZwOkCEmggvejwgXC1QPIEDMDyAAwEI0n4DDAJ6wwDT/AADoQHw+kB7sP4DOAF0AnrDAEDIsHpAiAM4AnrDANP8AEOjAfD+ATgB9AJ6wwAAyLA6QIhZSMYRnC9SPb/fsDy/w4J6wwDAiRwRVxgBdkJ6wwBBCLB6QIgDeALGA7xAQUJ6wwEAyarQmZgB9kJ6wwABCHA6QITZiMYRnC9ACW165EPCesMBk/wBAV1YBTQT+qRDgAhAL8EaFL4IVCsQgXRATFxRQDxBAD10wXgCesMAQYiA0bB6QIgGEZwvQAAQPIwEMDy/wABaAExHL8AaHBHQPbgcc/yAAEIeEloYfMLIHBHQPIgIMDy/wABaAExFL8AaE/0gFBwRwC/QPIkIMDy/wABaAExFL8AaE/0AHBwRwC/ACBwRwAgcEcQtf/34f8ERv/36v8A+wTwEL0AvwFEgUKcvwEgcEcD4IhCJL8BIHBHUPgEKwEyHL8AIHBH9OcAv3BHAL8AIHBHAyBwRwMohL9pIHBHgLVAsgWhUfggAEnyBFHF8gMBCGAA8Ar4ACCAvQAAAAACAAAAAQAAAAAAAABJ8gBAxfIDAAFoACn80HBHSfIMUMXyAwABIQFg8OcAv4GwAJAAmE/w/zEBYAGw5+dpIHBHsLX/96f/BEZAsQAlKEb/9+3///eD/wVEpUL30wAgsL0AAAAAAAAAAAAAAAAAAAAAAAAAAA==
    pc_init: 1
    pc_uninit: 53
    pc_program_page: 441
    pc_erase_sector: 169
    pc_erase_all: 133
    data_section_offset: 1136
    flash_properties:
      address_range:
        start: 16744448
        end: 16748544
      page_size: 4096
      erased_byte_value: 255
      program_page_timeout: 1000
      erase_sector_timeout: 3000
      sectors:
        - size: 4096
          address: 0
core: M33
````

I have no problem flashing the chip using for example JLinkExe.

macthestack, 2020-10-29:
When I set it to 100kHz I get the following:

```
      Config default
      Target /Users/martin/repos/external/nrf-hal/target/thumbv8m.main-none-eabihf/debug/blinky-button-demo
       ERROR probe_rs::probe::jlink > DAP read timeout.
     Erasing sectors ✔ [00:00:00] [------------------------]       0B/  4.00KB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [------------------------]       0B/  4.00KB @       0B/s (eta 0s )
       Error failed to flash /Users/martin/repos/external/nrf-hal/target/thumbv8m.main-none-eabihf/debug/blinky-button-demo
             
             Caused by:
                 0: Error while flashing
                 1: Something during memory interaction went wrong
                 2: A core architecture specific error occured
                 3: Failed to write register DRW at address 0x0000000c because: Operation timed out
                 4: Operation timed out
```

P.S. thanks for transferring me to the right place :)

jacobrosenthal, 2020-11-07:
@macthestack  Can you post your full debug log with the lowered speed?

macthestack, 2020-11-12:
@jacobrosenthal Sorry for taking so long. I will get back asap when I get the time to check it.

macthestack, 2020-11-14:
@jacobrosenthal

This is what I get:

```
       DEBUG jaylink           > libusb 1.0.23.11397
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: false
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: false
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 1 interfaces
       DEBUG jaylink                                > J-Link interface is #0
       DEBUG jaylink                                > legacy caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG jaylink                                > extended caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG probe_rs::probe::jlink                 > Supported speeds: Speeds { base_freq: 96000000, min_div: 24 }
       DEBUG probe_rs::probe::jlink                 > Divider: 960
        INFO cargo_embed                            > Protocol speed 100 kHz
       DEBUG probe_rs::probe::jlink                 > Attaching to J-Link
       DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
        INFO probe_rs::probe::jlink                 > J-Link: S/N: 801026105
       DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link EDU Mini V1 compiled Jul 17 2020 16:25:21
        INFO probe_rs::probe::jlink                 > J-Link: Hardware version: JLink 1.0.0
        INFO probe_rs::probe::jlink                 > J-Link: Target voltage: 3.30 V
       DEBUG probe_rs::probe::jlink                 > Sucessfully switched to SWD
       DEBUG probe_rs::probe::jlink                 > Attached succesfully
       DEBUG probe_rs::config::registry             > Searching registry for chip with name nRF9160_xxAA
       DEBUG probe_rs::architecture::arm::communication_interface > Debug Port version: DPv2
       DEBUG probe_rs::architecture::arm::communication_interface > Reading DP register DPIDR
       DEBUG probe_rs::architecture::arm::communication_interface > Read    DP register DPIDR, value=0x6ba02477
       DEBUG probe_rs::architecture::arm::communication_interface > DebugPort ID:  DebugPortId {
    revision: 0x6,
    part_no: 0xba,
    version: DPv2,
[... 213 lines cut ...]
       DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x000402e4
       DEBUG probe_rs::flashing::flasher                          > Loading algorithm into RAM at address 0x20000200
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write block with total size 1368 bytes to address 0x20000200
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0x20000200
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write first block with len 512 at address 0x20000200
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, block with len=128 words
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 4.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 3.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 2.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 1.
[... cut ...]
 Programming pages   ✔ [00:00:01] [-------------------------------------------------------------------------]       0B/ 12.00KB @       0B/s (eta 0s )
       Error failed to flash /Users/xxx/repos/xxx/xxx/xxx/target/thumbv8m.main-none-eabihf/debug/xxx
             
             Caused by:
                 0: Error while flashing
                 1: Something during memory interaction went wrong
                 2: A core architecture specific error occured
                 3: Failed to write register DRW at address 0x0000000c because: Operation timed out
                 4: Operation timed out
```

Yatekii, 2020-11-14:
can you please pass `--nmagic` to your linker command in `.cargo/config` ?

macthestack, 2020-11-14:
my config now looks like this:

```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = 'arm-none-eabi-gdb'
rustflags = [
    "-C", "link-arg=--nmagic",
    "-C", "link-arg=-Tlink.x",
]

[build]
target = "thumbv8m.main-none-eabihf" # Cortex-M33 (with FPU)

```

```
       DEBUG jaylink           > libusb 1.0.23.11397
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: false
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: false
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 1 interfaces
       DEBUG jaylink                                > J-Link interface is #0
       DEBUG jaylink                                > legacy caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG jaylink                                > extended caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG probe_rs::probe::jlink                 > Supported speeds: Speeds { base_freq: 96000000, min_div: 24 }
       DEBUG probe_rs::probe::jlink                 > Divider: 960
        INFO cargo_embed                            > Protocol speed 100 kHz
       DEBUG probe_rs::probe::jlink                 > Attaching to J-Link
       DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
        INFO probe_rs::probe::jlink                 > J-Link: S/N: 801026105
       DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link EDU Mini V1 compiled Jul 17 2020 16:25:21
        INFO probe_rs::probe::jlink                 > J-Link: Hardware version: JLink 1.0.0
[... 224 lines cut ...]
       DEBUG probe_rs::architecture::arm::communication_interface > Reading register DRW
       DEBUG probe_rs::architecture::arm::communication_interface > Read register    DRW, value=0x00040f2a
       DEBUG probe_rs::flashing::flasher                          > Loading algorithm into RAM at address 0x20000200
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write block with total size 1368 bytes to address 0x20000200
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register CSW, value=0x63000012
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register TAR, value=0x20000200
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Write first block with len 512 at address 0x20000200
       DEBUG probe_rs::architecture::arm::communication_interface         > Writing register DRW, block with len=128 words
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 4.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 3.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 2.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 1.
       DEBUG probe_rs::probe::jlink                                       > DAP line busy, retries remaining 0.
       ERROR probe_rs::probe::jlink                                       > DAP read timeout.
     Erasing sectors ✔ [00:00:01] [-------------------------------------------------------------------------]       0B/ 12.00KB @       0B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [-------------------------------------------------------------------------]       0B/ 12.00KB @       0B/s (eta 0s )
       Error failed to flash /Users/martin/repos/viva/edge/beebox/target/thumbv8m.main-none-eabihf/debug/beebox
             
             Caused by:
                 0: Error while flashing
                 1: Something during memory interaction went wrong
                 2: A core architecture specific error occured
                 3: Failed to write register DRW at address 0x0000000c because: Operation timed out
                 4: Operation timed out
```

macthestack, 2020-11-14:
I have a logic analyzer if that's helpful...

Yatekii, 2020-11-14:
Hmm it cannot write the first byte of the flash algorithm to RAM. So somehow RAM writes are locked.
You could try if running the https://crates.io/crates/nrf-recover utility fixes the issue. Otherwise we'll have to dig deeper how the nRF can lock RAM access through the AP :)

macthestack, 2020-11-14:
I'm getting `Error Could not find Nordic's CtrlAP`

I'll try to do the same with nrfjprog if possible

macthestack, 2020-11-14:
just did a `nrfjprog --recover` but it still doesn't work.

could I be using the wrong values in my nrf91.yaml file?

Yatekii, 2020-11-14:
It tries to write to 0x20000200 which looks fine to me :/

macthestack, 2020-11-14:
I'm using a segger edu mini connecting using a 6 pin tagconnect to an actinius icarus if that's of any use...

jacobrosenthal, 2020-11-14:
Can you try my branch of cargo embed. It has a jlink change that I think does retries better and might print more 
`cargo install --git https://github.com/jacobrosenthal/cargo-embed --branch samd --force`

 with speed low as usual

macthestack, 2020-11-15:
now I get:

```
       DEBUG jaylink           > libusb 1.0.23.11397
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: false
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: false
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 012: ID 1366:0101
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 007: ID 21a9:1004
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 009: ID 2109:0101
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 008: ID 058f:8468
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 007: ID 1a40:0801
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 005: ID 2109:0813
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 004: ID 2109:2813
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 003: ID 046d:082d
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 002: ID 2109:2813
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 000 Device 001: ID 2109:0813
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 002: ID 2188:6533
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 005 Device 001: ID 2188:0747
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 004 Device 001: ID 0411:01de
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 1 interfaces
       DEBUG jaylink                                > J-Link interface is #0
       Error An error specific to a probe type occured
             
             Caused by:
                 USB error while taking control over USB device: Access denied (insufficient permissions)
```

Yatekii, 2020-11-15:
Same difference; can't load the flash algorithm :/

jacobrosenthal, 2020-11-15:
Yeah. But this tells you the busy retry was a lie and was blocking us from seeing the sticky bits debug print.  Now we know its in nack state but it blew up so we cant see the damn sticky bits here either :)

macthestack, 2020-11-28:
@jacobrosenthal is there something I can do from my end to give you more information on what's going on?

Tiwalun, 2020-11-28:
@macthestack I did some work to improve the error handling for the JLink recently (see #443),  could you please try again with the current master version?

The panic should also be fixed, so we should see better what happens in the log.

macthestack, 2020-11-28:
Hi,

This is what I get now:

```
     Config default
      Target /Users/martin/repos/viva/edge/beebox/target/thumbv8m.main-none-eabihf/debug/beebox
       DEBUG jaylink           > libusb 1.0.23.11397
       DEBUG jaylink           > libusb has capability API: true
       DEBUG jaylink           > libusb has HID access: false
       DEBUG jaylink           > libusb has hotplug support: true
       DEBUG jaylink           > libusb can detach kernel driver: false
       DEBUG jaylink           > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink           > scanning 1 interfaces
       DEBUG jaylink           > J-Link interface is #0
       DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 002: ID 0403:6015
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 004 Device 004: ID 21a9:1004
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 004 Device 002: ID 1366:0101
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 001: ID 0411:01de
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 004 Device 001: ID 2188:0747
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 001: ID 2188:6533
       DEBUG probe_rs::probe::daplink::tools > No device matches
       DEBUG probe_rs::probe::daplink::tools > Attempting to open 1366:0101 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0x0,
    bDeviceSubClass: 0x0,
    bDeviceProtocol: 0x0,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x101,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 1 interfaces
       DEBUG jaylink                                > J-Link interface is #0
       DEBUG jaylink                                > legacy caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
       DEBUG jaylink                                > extended caps: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO cargo_embed                            > Protocol speed 0 kHz
       DEBUG probe_rs::probe::jlink                 > Attaching to J-Link
       DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
        INFO probe_rs::probe::jlink                 > J-Link: S/N: 801026105
       DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GET_HW_VERSION | READ_CONFIG | WRITE_CONFIG | SPEED_INFO | GET_MAX_BLOCK_SIZE | GET_HW_INFO | RESET_STOP_TIMED | SELECT_IF | GET_COUNTERS | GET_CPU_CAPS | EXEC_CPU_CMD | SWO | REGISTER | INDICATORS | TEST_NET_SPEED | GET_CAPS_EX
        INFO probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link EDU Mini V1 compiled Jul 17 2020 16:25:21
[... 373 lines cut ...]
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
```

*** The DAP FAULTS repeat for every 0x200 followed by this:

```
       DEBUG probe_rs::architecture::arm::memory::adi_v5_memory_interface > Finished reading block
       ERROR probe_rs::flashing::flasher                                  > Failed to verify flash algorithm. Data mismatch at address 0x20000200
       ERROR probe_rs::flashing::flasher                                  > Original instruction: 0xe00abe00
       ERROR probe_rs::flashing::flasher                                  > Readback instruction: 0x000000
       ERROR probe_rs::flashing::flasher                                  > Original: [e00abe00, 62d780d, 24084068, d3000040, 1e644058, 1c49d1fa, 2a001e52, 4770d1f, f240b5b0, f2c00504, 20000500, 105eb09, 5f849, 1e9c1, 461060c8, f0004614, b120fabf, 105eb09, 402e9c1, eb09bdb0, 21010005, 20006041, bf00bdb0, 4604b510, f0002000, f240faad, f2c00104, 44490100, e9c1b110, e0030402, 29006889, bd10bf08, 100f244, 100f2c2, 600a2201, 204f24 [... cut ...]
 Programming pages   ✔ [00:00:01] [-------------------------------------------------------------------------------------------------------------------------------------------------------]       0B/128.00KB @       0B/s (eta 0s )
       Error failed to flash /Users/martin/repos/viva/edge/beebox/target/thumbv8m.main-none-eabihf/debug/beebox

             Caused by:
                 0: Error while flashing
                 1: The RAM contents did not match the expected contents after loading the flash algorithm.
```

macthestack, 2020-11-28:
Here's a sample with trace logging:

```
       TRACE jaylink                                                      > write 18 bytes: [cf, 0, 38, 0, ff, 83, ff, ff, ff, ff, ff, 4, 2, a, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 8 bytes: [0, 44, 0, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP wrote 20.
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, 4, 0, 0, 0, 0, 0]
       TRACE probe_rs::probe::jlink                                       > DAP read 0.
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, 7c, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, 4, 0, 0, 0, 0, 0]
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, f4, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, f0, ff, ff, ff, 7f, 0]
       DEBUG probe_rs::probe::jlink                                       > DAP FAULT
       TRACE jaylink                                                      > write 16 bytes: [cf, 0, 2f, 0, ff, 3, 0, 0, 0, 0, 34, 2, 0, 0, 0, 0]
       TRACE jaylink                                                      > read 7 bytes: [0, 64, 4, 0, 0, 7e, 0]
       TRACE probe_rs::probe::jlink                                       > DAP read 4026531875.
       DEBUG probe_rs::probe::jlink                                       > Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4026531875,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: false,
    c_dbg_rst_req: false,
    trn_cnt: 0,
    mask_lane: 0,
    w_data_err: false,
    read_ok: false,
    sticky_err: true,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}
```

Tiwalun, 2020-11-28:
The initial comunication between probe and chip seems fine, until it tries to write into the RAM. All writes seem to fail, as indicated by the fault response, and that the error bits are set. 

Unfortunately, that doesn't really tell us why the writes fail.

Please try running the ram_download example in the probe-rs repo:

    cargo run --example ram_download -- -chip nRF9160_xxAA --address 0x20000200 --size 128

This should tell us if just accessing the RAM works.

macthestack, 2020-11-28:
This is what I'm getting:

```
╰─ cargo run --example ram_download -- --chip nRF9160_xxAA --address 0x20000200 --size 128
    Finished dev [unoptimized + debuginfo] target(s) in 0.55s
     Running `target/debug/examples/ram_download --chip nRF9160_xxAA --address 0x20000200 --size 128`
Wrote 512 bytes in 148.373249ms ( 3450.76 bytes/s)
Read  512 bytes in 183.925167ms ( 2783.74 bytes/s)
Verification failed!
Readback data differs at address 20000200: expected word 34489a0a, got word 00000000
```

macthestack, 2020-11-28:
is this helpful?

```
╰─ nrfjprog --ramwr 0x20000000  --val 0x1234
Parsing parameters.
Writing.
```

```
nrfjprog --memrd 0x20000000  --n 4
```

Tiwalun, 2020-11-28:
Yes, good to know that the RAM at this address seems to work normally with the `nrfjprog`. 

Can you run the `cargo run -p probe-rs-cli -- info`command? This should show some information about the available access port, which might help in figuring out what's going wrong.

If you have a logic analyzer, traces of both the nrfjprog, as well as the ram_download example would be very helpful, then we could compare them and see what the differences between them are. A small write size should be enough, no need for hundreds of bytes.

macthestack, 2020-11-28:
i have a saleae logic...all 6 pins or any ones in particular?

macthestack, 2020-11-28:
Here's the "info"

```
Available Access Ports:
Class1RomTable(
    ComponentId {
        component_address: 0xe00ff000,
        class: RomTable,
        peripheral_id: PeripheralID {
            REVAND: 0x000000,
            CMOD: No,
            REVISION: 0x000000,
            JEP106: Some(
                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
            ),
            PART: 0x0004c9,
            SIZE: 0x000001,
        },
    },
    RomTable {
        entries: [
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component: Class9RomTable(
                    ComponentId {
                        component_address: 0xe000e000,
                        class: CoreSightComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000000,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x000d21,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component: Class9RomTable(
                    ComponentId {
                        component_address: 0xe0001000,
                        class: CoreSightComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000000,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x000d21,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component: Class9RomTable(
                    ComponentId {
                        component_address: 0xe0002000,
[... 310 lines cut ...]
                        component_address: 0x8000b000,
                        class: CoreSightComponent,
                        peripheral_id: PeripheralID {
                            REVAND: 0x000000,
                            CMOD: No,
                            REVISION: 0x000003,
                            JEP106: Some(
                                JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                            ),
                            PART: 0x000908,
                            SIZE: 0x000001,
                        },
                    },
                ),
            },
        ],
        tpiu: None,
        dwt: None,
        itm: None,
    },
)
Unknown Type of access port
Unknown Type of access port
Unknown Type of access port
```

Tiwalun, 2020-11-28:
> i have a saleae logic...all 6 pins or any ones in particular?

The two SWD (SWDIO / SWCLK) would be enough.

macthestack, 2020-11-28:
I've attached 3 sal-files. I can export as something else if preferred.

I ran:
* `cargo run --example ram_download -- --chip nRF9160_xxAA --address 0x20000200 --size 1`
* `nrfjprog --ramwr 0x20000200  --val 0x1234`
* `nrfjprog --memrd 0x20000200  --n 4`

[Archive.zip](https://github.com/probe-rs/probe-rs/files/5610506/Archive.zip)

macthestack, 2020-11-28:
Apologies, I had a glitch-filter on which screwed up the SWCLK. Here's a new file.
[logic.zip](https://github.com/probe-rs/probe-rs/files/5610559/logic.zip)

macthestack, 2020-11-28:
@Yatekii is #168 related?

macthestack, 2020-11-28:
Both are M33 and it seems this could be the issue?

```
name	type	start_time	duration	value
SWD	v1frame	2.0633026	8.08e-06	Request  AccessPort Write CSW
SWD	v1frame	2.06331076	8.8e-07	Turnaround
SWD	v1frame	2.06331176	2.96e-06	ACK FAULT

```

Yatekii, 2020-11-28:
Ah yes, that would explain it ... M33 is v8 :/ We only implement v6 and v7. But v7 is very similar to v6, so I would assume making v8 work is similarly easy :)

macthestack, 2020-11-28:
I see. Is v8 on the roadmap? Do you know if anyone is working on this?

Yatekii, 2020-11-28:
It is not on the roadmap because I just completely forgot :D
As far as I know, nobody is working on this yet. I'll put it on the roadmap (https://hackmd.io/lwlYenbsSJS5XZQgTaGvxQ) which is very unordered at the moment :/

macthestack, 2020-11-28:
Great! Thank you. :)

closed 2020-12-03 by bors[bot]

Tiwalun, 2020-12-03:
@macthestack Could you please try again with the current master and check if your issue is now fixed? 

Feel free to reopen if you still have problems.

macthestack, 2020-12-03:
Rerunning the example gives me:

```
Wrote 512 bytes in 47.548317ms (10767.99 bytes/s)
Read  512 bytes in 79.984444ms ( 6401.24 bytes/s)
Verification successful.
```

Thanks everyone for your help!

commit e527946b (2020-12-03): Merge #465

465: Add support for HNONSEC bit in CSW register (MemoryAP) r=Yatekii a=Tiwalun

Add support for the HNONSEC bit, which is used in the CSW register of a MemoryAP.

Support for the HNONSEC bit is automatically detected, and it is set to 0, meaning secure access,
if it's supported.

This might fix #404, and might help with #310.
PR #465 (2020-12-03): Add support for HNONSEC bit in CSW register (MemoryAP)
Add support for the HNONSEC bit, which is used in the CSW register of a MemoryAP.

Support for the HNONSEC bit is automatically detected, and it is set to 0, meaning secure access,
if it's supported.

This might fix #404, and might help with #310.