probe-rs/probe-rs #414: STM32F103C8 on windows returns Error: The firmware on the probe is outdated

jaysonsantos, 2020-11-02:
**Describe the bug**
When running `probe-rs-cli.exe debug -c STM32F103C8`  it returns `Error: The firmware on the probe is outdated`
I am using a generic stlink (Arceli St Link V2 STLINK for STM8S Stm8l STM32 Cortex M0 CORTEX-M3 Swim JTAG SWD Interface Programmer ) with driver version `2.1.0.0`

**Expected behavior**
I guess it should print information about the probe instead of returning errors.

**Stacktrace**
```
probe-rs-cli.exe debug -c STM32F103C8
 DEBUG jaylink > libusb 1.0.23.11397
 DEBUG jaylink > libusb has capability API: true
 DEBUG jaylink > libusb has HID access: true
 DEBUG jaylink > libusb has hotplug support: false
 DEBUG jaylink > libusb can detach kernel driver: false
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 002: ID 045e:0291
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 002: ID 062a:5918
 DEBUG probe_rs::probe::daplink::tools > Device open error NoDevice
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 000: ID 1022:149c
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 001: ID 045e:028f
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 002: ID 045e:076d
 DEBUG probe_rs::probe::daplink::tools > Device open error NotFound
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 004: ID 0a12:0001
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 000: ID 1022:149c
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 001: ID 05e3:0610
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 005: ID 0483:3748
 DEBUG probe_rs::probe::daplink::tools > No device matches
 DEBUG probe_rs::probe::daplink::tools > Device Bus 003 Device 001: ID 1017:2010
 DEBUG probe_rs::probe::daplink::tools > Device open error NoDevice
 DEBUG probe_rs::probe::daplink::tools > Device Bus 001 Device 000: ID 1022:149c
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Device Bus 002 Device 003: ID 0bda:5411
 DEBUG probe_rs::probe::daplink::tools > Device open error NotSupported
 DEBUG probe_rs::probe::daplink::tools > Attempting to open 0483:3748 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 39, bNumInterfaces: 1, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 50, extra: None }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 0, bDeviceSubClass: 0, bDeviceProtocol: 0, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14152, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 TRACE probe_rs::probe::stlink                > Getting current mode of device...
 TRACE probe_rs::probe::stlink::usb_interface > Sending command [f5] to STLink, timeout: 1s
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
 TRACE probe_rs::probe::stlink::usb_interface > Sending command [f1] to STLink, timeout: 1s
 TRACE probe_rs::probe::stlink                > Getting current mode of device...
 TRACE probe_rs::probe::stlink::usb_interface > Sending command [f5] to STLink, timeout: 1s
 DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
Error: The firmware on the probe is outdated

Caused by:
    The firmware on the probe is outdated
```
**Desktop (please complete the following information):**
 - Windows 10

Let me know if I can provide more info on that, I am quite new on this and don't know what could help.

Disasm, 2020-11-02:
Maybe we should suggest a link to the firmware updater?

jaysonsantos, 2020-11-02:
@Disasm hi there, on my side I've installed the last one from st.com (https://www.st.com/en/development-tools/stsw-link009.html#overview&secondary=st-get-software)
Also, I just added a trace log to see what was the detected version on my stlink, here is the info:
```
STLink {
    device: STLinkUSBDevice {
        device_handle: "DeviceHandle<rusb::Context>",
        info: STLinkInfo {
            version_name: "V2",
            usb_pid: 14152,
            ep_out: 2,
            ep_in: 129,
            ep_swo: 131,
        },
    },
    hw_version: 2,
    jtag_version: 17,
    protocol: Swd,
    swd_speed_khz: 1800,
    jtag_speed_khz: 1120,
    swo_enabled: false,
    openend_aps: [],
}
```

Yatekii, 2020-11-02:
The minimal required JTAG version is 26, so I assume something with your updater/updater-process is broken :/ Because 17 seems veeeery outdated :/

Or, ofc, we have a bug reading the version, but I haven't heard of that yet ...

jaysonsantos, 2020-11-02:
Hey there, I just downloaded the firmware updater from [here](https://my.st.com/content/my_st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stsw-link007.license=1604289687555.product=STSW-LINK007.version=2.37.26.html), changed my device to DFU and it seems to be working fine now :)

closed 2020-11-02 by jaysonsantos