probe-rs/probe-rs #424: CMSIS-DAP v2 devices are not drained on attach

bcantrill, 2020-11-09:
On attach to a CMSIS-DAP device, any buffered data is drained to assure that we do not get out of sync with respect to requests and responses:

https://github.com/probe-rs/probe-rs/blob/30b01a5cd811061d92b11c54f46a68ee2d193eb7/probe-rs/src/probe/daplink/mod.rs#L81-L91

However, this is only done for CMSIS-DAP v1 devices; v2 devices have the same issue and must be similarly drained. Here is the fix that we're using for this, but there are obviously other ways to do this:

https://github.com/oxidecomputer/probe-rs/commit/bf97d3fb296c9100ee11767f4ad3b15afa5c36ca

Yatekii, 2020-11-09:
Thanks a lot for reporting this and especially the fix :)
Would you mind PRing the fix? Otherwise I can also pull your commit myself :)

bcantrill, 2020-11-09:
Yes, happy to!

closed 2021-05-23 by Tiwalun