probe-rs/probe-rs #433: Many probe+chip combinations only work at low speeds

jacobrosenthal, 2020-11-16:
As seen in many issues (https://github.com/probe-rs/probe-rs/issues/395) and in chat many probe+ chip combos dont work reliably above say speed 100. These chips function fine at low speeds so presumably theyre generally functional.

`RUST_LOG=probe_rs cargo flash --example ferris_img  --disable-progressbars > out`
https://gist.github.com/jacobrosenthal/b7de08cc3b675e8451cc26ecb02c2fa1
"Didn't receive any answer during batch processing" but the salae log shows a fault
[cargo flash cmsis samd fault.zip](https://github.com/probe-rs/probe-rs/files/5550172/cargo.flash.cmsis.samd.fault.zip)

Openocd succeeds seemingly at 400hz
[Uploading openocd cmsis samd.zip…]()

```
jacob@j:~/Downloads/atsamd/boards/pygamer (qspi-badge)$ openocd -f interface/cmsis-dap.cfg -c "cmsis_dap_vid_pid 0xc251 0xf001" -f target/atsame5x.cfg -c "init; halt; program ferris_img.bin reset exit 0x00040000; shutdown"
Open On-Chip Debugger 0.10.0+dev-01487-g475f42051-dirty (2020-11-08-13:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : atsame5.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for atsame5.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00000fc6 msp: 0x20012a48
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000574 msp: 0x20012ab8
** Programming Started **
Info : SAM MCU: SAMD51J19A (512KB Flash, 192KB RAM)
** Programming Finished **
** Resetting Target **
shutdown command invoked
jacob@j:~/Downloads/atsamd/boards/pygamer (qspi-badge)$ 
```

Ubuntu linux on a pygamer atsamd51j19a
b2bfa73c9edc47245659838cc61df209ba18dff4 cargo flash built on probe-rs 1581e113fb9eb5962dafc600d1734483b6b5fecf
openocd 0.10.0+dev-01487-g475f42051-dirty (2020-11-08-13:26)

jacobrosenthal, 2020-11-16:
Cmsis doesnt seem to be catching the fault, instead the error is being sent down the 0 response path probably due to some batching loging
https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/daplink/mod.rs#L232

I put logging in there to read the ctrl reg for sticky. I put the usual stick bit logging in there
```
       TRACE probe_rs::probe::daplink                             > Writing DAP register failed. Ctrl/Stat register value is: Ctrl { .0: 4026531936, csyspwrupack: true, csyspwrupreq: true, cdbgpwrupack: true, cdbgpwrupreq: true, cdbgrstack: false, c_dbg_rst_req: false, trn_cnt: 0, mask_lane: 0, w_data_err: false, read_ok: true, sticky_err: true, stick_cmp: false, trn_mode: 0, sticky_orun: false, orun_detect: false }
```
Looks like just a sticky error that could be cleared and retried?

jacobrosenthal, 2020-11-17:
A Jlink edu full size with same
cargo flash --speed 1000 fails
https://gist.github.com/jacobrosenthal/2b68004923a5796ed45cd962ca698987

[cargo flash jlink edu samd.zip](https://github.com/probe-rs/probe-rs/files/5550640/cargo.flash.jlink.edu.samd.zip)
but be warned, id didnt decode that last packet :(

jacobrosenthal, 2020-11-18:
Probe tracking https://docs.google.com/spreadsheets/d/1I08bEsEIZCnpnVwrYLkGp_juy_uMqMiLHWknFfPmP_o/edit#gid=0

Tiwalun, 2020-11-21:
I've tried to improve the error handling for the J-Link in #443, and with these changes I can use the maximum speed my J-Link supports (4 MHz) without problems. 

Without the changes, only slow frequencies (about 100 kHz) work reliably.

@jacobrosenthal @Yatekii Could you try and see if the changes in the PR improve things for you as well?

Yatekii, 2020-11-21:
@adamgreig maybe too :)

adamgreig, 2020-11-21:
I don't have a J-Link to test with sadly. Do you actually see the line reset happen often and then help? Is there a noticeable change in memory read speeds (or flash write speeds I guess) with the addition of the idle bits at the same clock? I'm sure having the much higher clock more than makes up for it though.

Tiwalun, 2020-11-21:
If I don't add the idle bits, I see a lot of line resets, and the total time for flashing goes up from 40 s to about 50 s, but it succeeds.
Without the line reset, it basically never succeeds, so the line reset definitely helps, but the idle bits seem to avoid the need for line resets.

For just memory read and write, I don't see any differences whether the line reset and idle bits are present or not. I think the J-Link implementation is just too slow right now for the idle bits to matter.

adamgreig, 2020-11-21:
Sounds good, then! I wonder if some idle bits could possibly help the cmsis-dap + atsamd51 issue... it seems to happen mid-batch though, so I'm not sure.

Yatekii, 2020-11-21:
Hmm if you do:

write
read

will it read the written value or read the value at the moment of read where the write might not have completed yet?

Tiwalun, 2020-11-21:
The actions are done in order, so I don't think that this is a problem. And if the write is not done yet, we get a WAIT response, from what I understand.

jacobrosenthal, 2020-11-29:
With both #450 and #462 we dont have any open reports about reduced probe speed being necessary. If you stumble across this in the future please report.

closed 2020-11-29 by jacobrosenthal