probe-rs/probe-rs #448: SWO streams on CMSIS-DAP v2 can become sluggish and corrupt

bcantrill, 2020-11-23:
I have been chasing an intermittent ITM issue on an LPC55 via an LPC-Link2 (LPC4322) running ARM's DAPLink (CMSIS-DAP v2).  As it turns out, this firmware really doesn't like packets larger than `wMaxPacketSize`.  probe-rs does a read of 1025 bytes; if the data available is less than `wMaxPacketSize` (which is 64 bytes for this firmware), this is considered a short read, and the DAPLink properly indicates a short read, resulting in the host returning the data to the caller (that is, to probe-rs, via rusb and libusb).  If, however, the data to be returned is *exactly* `wMaxPacketSize`, the DAPLink firmware will not indicate a short read -- but will also not indicate read completion.  In this case, the read will return -- but only after a read timeout.  (That is, the data will be returned when libusb indicates that the outstanding URB should be discarded.)  And because rusb eats the timeout when data is present (behavior which itself is arguably undesirable), and because probe-rs has a timeout of 100 ms (a timeout which itself is probably too large), the upshot of it all is that the SWO data rate drops to 640 bytes per second -- a paltry rate that (assuming ongoing ITM activity) ultimately results in dropped ITM data and a corrupted data stream.  

There is room for improvement in several different places here:

- Ideally, the DAPLink would be a little more tolerent of the outsized read -- and indicate completion under this condition.  But the USB spec is also very clear that reads should not exceed `wMaxPacketSize`, so unclear to what degree this is an issue.

- rusb probably shouldn't be eating the `LIBUSB_ERROR_TIMEOUT` -- and should probably be able to indicate a timeout-with-read as its own condition, or treat it as a timeout rather than a successful read.

- Assuming that reads greater than `wMaxPacketSize` can result in this kind of undefined behavior (that is, assuming that this isn't purely the fault of the DAPLink firmware), it would be great if some layer of software (rusb, libusb or the in-kernel USB stack) explicitly failed rather than persisting.

- probe-rs's 100 ms timeout is probably too large.

- probe-rs could/should propagate the overrun

But of course, the real issue is that probe-rs should issue the read of the appropriate size.  Fixing this resolves the issue; here's the diff, and a PR will be forthcoming:

```diff
diff --git a/probe-rs/src/probe/daplink/commands/mod.rs b/probe-rs/src/probe/daplink/commands/mod.rs
index 98e8c10..8be1427 100644
--- a/probe-rs/src/probe/daplink/commands/mod.rs
+++ b/probe-rs/src/probe/daplink/commands/mod.rs
@@ -192,12 +192,9 @@ pub(crate) fn send_command<Req: Request, Res: Response>(
     device: &mut std::sync::Mutex<DAPLinkDevice>,
     request: Req,
 ) -> Result<Res> {
-    // On CMSIS-DAP v2 USB HS devices, a single request might be up to 1024 bytes,
-    // plus we need one extra byte for the always-written HID report ID.
-    const BUFFER_LEN: usize = 1025;
+    // See comment below on the sizing of this buffer
+    let mut write_buffer = [0; 65];
 
-    // Write the command & request to the buffer.
-    let mut write_buffer = [0; BUFFER_LEN];
     write_buffer[1] = *Req::CATEGORY;
     let mut size = request.to_bytes(&mut write_buffer, 1 + 1)?;
     size += 2;
@@ -210,17 +207,33 @@ pub(crate) fn send_command<Req: Request, Res: Response>(
         // write exactly 64 (+1 for report ID) bytes for HID.
         // For v2 devices, we can write the precise request size.
         if let DAPLinkDevice::V1(_) = device {
-            size = 65;
+            size = write_buffer.len();
         }
 
         // Send buffer to the device.
         device.write(&write_buffer[..size])?;
         log::trace!("Send buffer: {:02X?}", &write_buffer[..size]);
 
+        // We need to be sure to read no more than wMaxPacketSize, as devices
+        // are within their rights to behave strangely on an outsized read:
+        // they can view a read of exactly wMaxPacketSize as neither a short
+        // read (it fills a packet) nor a completed one (it is less than the
+        // requested size), ultimately resulting in the correct data, but only
+        // after the host times out and cancels the outstanding read.  For HID
+        // devices, we don't have an easy way of determining the maximum packet
+        // size, so we just stick with the size calculation above -- but for
+        // V2 devices, we can use the correct value from the device descriptor.
+        let len: usize = match device {
+            DAPLinkDevice::V1(_) => size,
+            DAPLinkDevice::V2 { handle, .. } => {
+                handle.device().device_descriptor()?.max_packet_size().into()
+            }
+        };
+
         // Read back resonse.
-        let mut read_buffer = [0; BUFFER_LEN];
+        let mut read_buffer = vec![0u8; len];
         device.read(&mut read_buffer)?;
-        log::trace!("Receive buffer: {:02X?}", &read_buffer[..]);
+        log::trace!("Receive buffer: {:02X?}", &read_buffer);
 
         if read_buffer[0] == *Req::CATEGORY {
             Res::from_bytes(&read_buffer, 1)
```

adamgreig, 2020-11-23:
Thanks for reporting! I agree that we should try to perform the correct sized read, and that it should be 64 bytes for HID and wMaxPacketSize for v2 (or should it be CMSIS-DAP reported 'packet size'?).

Is there any reason to also change the write buffer length to 65 in your diff? Some probes genuinely want >64-byte bulk writes, and the actual write call is already truncated based on the actual contents or to 65 for HID.

Also, it looks like your diff is based on a somewhat older version of that file, though we've not changed this behaviour, but have at least updated the logging to not include trailing 0s in what gets logged, which was especially useful with a 1025-byte write buffer.

bcantrill, 2020-11-26:
Thanks for taking a look @adamgreig!  Good point on clamping of the buffer size: I had neglected that of course `TransferRequest` and `TransferBlockRequest` can be dynamically sized.  In addition to the issue that you noted, on CMSIS-DAP v1 devices that are USB HS (that is, have a `wMaxPacketSize` of 1024), my fix would do the wrong thing by undersizing the read buffer. Based on that, I have pared this fix back to:

```diff
diff --git a/probe-rs/src/probe/daplink/commands/mod.rs b/probe-rs/src/probe/daplink/commands/mod.rs
index 98e8c10..f326f6c 100644
--- a/probe-rs/src/probe/daplink/commands/mod.rs
+++ b/probe-rs/src/probe/daplink/commands/mod.rs
@@ -217,10 +217,26 @@ pub(crate) fn send_command<Req: Request, Res: Response>(
         device.write(&write_buffer[..size])?;
         log::trace!("Send buffer: {:02X?}", &write_buffer[..size]);
 
+        // We need to be sure to read no more than wMaxPacketSize, as devices
+        // are within their rights to behave strangely on an outsized read:
+        // they can view a read of exactly wMaxPacketSize as neither a short
+        // read (it fills a packet) nor a completed one (it is less than the
+        // requested size), ultimately resulting in the correct data, but only
+        // after the host times out and cancels the outstanding read.  For HID
+        // devices, we don't have an easy way of determining the maximum packet
+        // size, so we just stick with the buffer size -- but for V2 devices,
+        // we can use the correct value from the device descriptor.
+        let len: usize = match device {
+            DAPLinkDevice::V1(_) => BUFFER_LEN,
+            DAPLinkDevice::V2 { handle, .. } => {
+                handle.device().device_descriptor()?.max_packet_size().into()
+            }
+        };
+
         // Read back resonse.
-        let mut read_buffer = [0; BUFFER_LEN];
+        let mut read_buffer = vec![0u8; len];
         device.read(&mut read_buffer)?;
-        log::trace!("Receive buffer: {:02X?}", &read_buffer[..]);
+        log::trace!("Receive buffer: {:02X?}", &read_buffer);
 
         if read_buffer[0] == *Req::CATEGORY {
             Res::from_bytes(&read_buffer, 1)
```

On using the CMSIS-DAP reported packet size rather than `wMaxPacketSize`: I certainly agree (as it would also allow us to have the correct packet size on v1 devices), but I think it's going to require some refactoring of the code:  we either need to pass the packet size into every command, or we need `DAPLinkDevice` to become a struct which has `send_command` as a method and stores the packet size (which to me is a more intuitive way to structure it).   Happy to take a swing at that if that approach makes sense to you.

adamgreig, 2020-11-26:
> On using the CMSIS-DAP reported packet size rather than `wMaxPacketSize`: I certainly agree (as it would also allow us to have the correct packet size on v1 devices), but I think it's going to [...]

Ah, there is a chicken-and-egg issue with this, which is we need to read from the probe to find its reported size, which fails on HID if we don't already know the exact correct report size. This is responsible for #308 having to go to some measures to guess the report size for a few annoying probes which use 512 (despite being USB FS, so I understand this is strictly worse) and then pass that information everywhere.

adamgreig, 2020-11-28:
It turns out we should be using the `max_packet_size()` on the endpoint rather than the device handle, otherwise it returns the packet size for EP0 rather than the actual CMSIS-DAP endpoint (which can differ). @Disasm found this in https://github.com/adamgreig/ecpdap/pull/1; perhaps we should do a similar thing of storing the packet size in the probe struct.

commit 80c596fd (2021-04-21): CMSIS-DAP: Always use best USB transfer size, closes #448
closed 2021-04-21 by bors[bot]