probe-rs/probe-rs #453: Improved memory interface

Tiwalun, 2020-11-25:
## Current situation
The current interface offers `read_8`/`read_32` methods, and corresponding write methods. The `read_32` method only use 32 bit accesses to read memory, and requires an aligned access. However, the `read_8` does not only use 8-bit memory accesses, but also 32 bit accesses, both to improve speed, as well for compatibility reasons, if 8-bit reads are not available.

## Proposal
Introduce new, generic access methods, i.e. `read` and `write`, which don't guarantee a specific access width, but are optimized for speed and compatibility. The existing `read_8` and `read_32` methods will be changed, so that they only use 8-bit or 32-bit wide access, respectively. The same applies to the write methods.

Yatekii, 2020-11-25:
I genearally think this is a good change. The question is what type the read/write would map to then. Because read_8 and read_32 were never intended to guarantee a read size but to guarantee a data format. Generics don't work here because of Rusts weird rule that Trait objects cannot have generic methods :/
And I actually think we should definitely have both methods, one for reading an array of u32 and one for an array of u8's

Tiwalun, 2020-11-25:
> The question is what type the read/write would map to then.

I'm not sure if I understand you correctly, but I would just use `u8` as the data type for the new read/write methods.

Yatekii, 2020-11-25:
Hmm that would make sense since read_32 already does exactly the second part :) Good!

Yatekii, 2022-03-05:
I think this is resolved now. Please reopen if I am incorrect.

Yatekii, 2022-03-06:
I am working on this :)

commit 05d2c3ec (2022-12-09): Merge #1078

1078: Start standardizing memory ops r=Yatekii a=Yatekii

Fixes #453.
PR #1078 (2022-12-09): Start standardizing memory ops
Fixes #453.
closed 2022-12-09 by bors[bot]